Azure/Industrial-IoT #469: Can't browse using Engineering Tool 2.7.64

gillermaney, 2020-05-18:
Deployed using precanned script.  The engineering tool worked at first and I discovered the gateways, connected to the endpoints ( Wingat didn't have any endpoints), and subscribed to a variety of variables in the linux simulated OPC UA server.  The data is flowing into the data lake, but the engineering tool is no longer working.  Accessing any page results in the error message below.  This has been happening intermittently for over a week.

Error: {"StatusCode":500,"Message":"","Data":{},"Source":"Microsoft.Azure.IIoT.Abstractions","HResult":-2146233088,"Exception":"HttpTransientException"}

marcschier, 2020-05-19:
Hi @gillermaney, we see this during our load testing when the IoT Hub messaging quota is exceeded (for the day).   Could you check in your deployed IoT Hub whether you have reached your message limits and if so upgrade your Hub to an S2.   We are working on making the UX show this error rather than the cryptic message you are seeing.

marcschier, 2020-05-20:
Moving to 2.8 milestone.   If this is not due to quota exceeded, please let us know.

closed 2023-06-11 by marcschier