Azure/Industrial-IoT #470: Incorrect timestamping in data lake?

gillermaney, 2020-05-18:
Using canned deployment script, data in data lake seems to be incorrectly timestamped.  This shows up when using TSI as shown in attached screen shot.
[image]

cristipogacean, 2020-05-19:
@gillermaney, can you please detail a bit your statement? for example what are the "seen values" and "expected value"? please note all timestamps in the platform are UTC format.

gillermaney, 2020-05-19:
This error seems to occur when the IoT Hub messaging quota is exceeded.  The expected timestamp values are strictly periodic.  The seen timestamp values are "bursty" and seem to reflect the time that telemetry reaches the data lake and not the OPC UA source timestamp.

gillermaney, 2020-05-29:
What information do you need besides what I have provided?  Why would timestamps be incorrect if the telemetry data is being delayed in IoT Hub?

cristipogacean, 2020-06-03:
@gillermaney can you try the most recent preview version? we did fix some connectivity issues on the publisher lately and those issues might have caused the situation you experience with TSI.

closed 2021-03-20 by marcschier