Azure/Industrial-IoT #482: Twin not connecting to OPC UA server after upgrading

Bartor, 2020-05-22:
We had the stack deployed in the beginning of march using the deployment script provided in the repository, but the deployment stopped working abruptly last weekend. The microservices couldn't start, with the exception stating:
```
Container svchrfpi_0_2b279cec couldn't be started: Logs = 2020-05-22T08:15:28.574675662Z Unhandled exception. System.AggregateException: One or more errors occurred. (A keyvault uri was provided could not access keyvault at the address. If you want to read configuration from keyvault, make sure the keyvault is reachable, the required permissions are configured on keyvault and authentication provider information is available. Sign into Visual Studio or Azure CLI on this machine and try again.)
2020-05-22T08:15:28.574748666Z  ---> Microsoft.Azure.IIoT.Exceptions.InvalidConfigurationException: A keyvault uri was provided could not access keyvault at the address. If you want to read configuration from keyvault, make sure the keyvault is reachable, the required permissions are configured on keyvault and authentication provider information is available. Sign into Visual Studio or Azure CLI on this machine and try again.
2020-05-22T08:15:28.574762267Z  ---> Microsoft.Azure.KeyVault.Models.KeyVaultErrorException: Operation returned an invalid status code 'Unauthorized'
2020-05-22T08:15:28.574770867Z    at Microsoft.Azure.KeyVault.KeyVaultClient.GetSecretWithHttpMessagesAsync(String vaultBaseUrl, String secretName, String secretVersion, Dictionary`2 customHeaders, CancellationToken cancellationToken)
2020-05-22T08:15:28.574779468Z    at Microsoft.Azure.KeyVault.KeyVaultClientExtensions.GetSecretAsync(IKeyVaultClient operations, String vaultBaseUrl, String secretName, CancellationToken cancellationToken)
2020-05-22T08:15:28.574789568Z    at Microsoft.Extensions.Configuration.ConfigurationEx.KeyVaultConfigurationProvider.ValidateReadSecretAsync(String secretName) in D:\a\1\s\common\src\Microsoft.Azure.IIoT.Auth.ActiveDirectory\src\Extensions\ConfigurationEx.cs:line 225
2020-05-22T08:15:28.574799669Z    at Microsoft.Extensions.Configuration.ConfigurationEx.KeyVaultConfigurationProvider.CreateInstanceAsync(IConfigurationRoot configuration, Boolean allowInteractiveLogon, String keyVaultUrlVarName, Boolean lazyLoad) in D:\a\1\s\common\src\Microsoft.Azure.IIoT.Auth.ActiveDirectory\src\Extensions\ConfigurationEx.cs:line 190
2020-05-22T08:15:28.574809470Z    --- End of inner exception stack trace ---
2020-05-22T08:15:28.574818770Z    at Microsoft.Extensions.Configuration.ConfigurationEx.KeyVaultConfigurationProvider.CreateInstanceAsync(IConfigurationRoot configuration, Boolean allowInteractiveLogon, String keyVaultUrlVarName, Boolean lazyLoad) in D:\a\1\s\common\src\Microsoft.Azure.IIoT.Auth.ActiveDirectory\src\Extensions\ConfigurationEx.cs:line 193
2020-05-22T08:15:28.574828071Z    at Microsoft.Extensions.Configuration.ConfigurationEx.KeyVaultConfigurationProvider.CreateInstanceAsync(Boolean allowInteractiveLogon, Boolean singleton, IConfigurationRoot configuration, String keyVaultUrlVarName) in D:\a\1\s\common\src\Microsoft.Azure.IIoT.Auth.ActiveDirectory\src\Extensions\ConfigurationEx.cs:line 157
2020-05-22T08:15:28.574837271Z    --- End of inner exception stack trace ---
2020-05-22T08:15:28.574845072Z    at System.Threading.Tasks.Task`1.GetResultCore(Boolean waitCompletionNotification)
2020-05-22T08:15:28.574852872Z    at System.Threading.Tasks.Task`1.get_Result()
2020-05-22T08:15:28.574860673Z    at Microsoft.Extensions.Configuration.ConfigurationEx.AddFromKeyVault(IConfigurationBuilder builder, ConfigurationProviderPriority providerPriority, Boolean allowInteractiveLogon, Boolean singleton, String keyVaultUrlVarName) in D:\a\1\s\common\src\Microsoft.Azure.IIoT.Auth.ActiveDirectory\src\Extensions\ConfigurationEx.cs:line 42
2020-05-22T08:15:28.574879174Z    at Microsoft.Azure.IIoT.Services.All.Startup..ctor(IWebHostEnvironment env, IConfiguration configuration) in D:\a\1\s\services\src\Microsoft.Azure.IIoT.Services.All\src\Startup.cs:line 45
2020-05-22T08:15:28.574888574Z --- End of stack trace from previous location where exception was thrown ---
2020-05-22T08:15:28.574896575Z    at Microsoft.Extensions.DependencyInjection.ActivatorUtilities.ConstructorMatcher.CreateInstance(IServiceProvider provider)
2020-05-22T08:15:28.574905175Z    at Microsoft.Extensions.DependencyInjection.ActivatorUtilities.CreateInstance(IServiceProvider provider, Type instanceType, Object[] parameters)
2020-05-22T08:15:28.574913176Z    at Microsoft.AspNetCore.Hosting.GenericWebHostBuilder.UseStartup(Type startupType, HostBuilderContext context, IServiceCollection services)
2020-05-22T08:15:28.574921276Z    at Microsoft.AspNetCore.Hosting.GenericWebHostBuilder.<>c__DisplayClass12_0.b__0(HostBuilderContext context, IServiceCollection services)
2020-05-22T08:15:28.574929977Z    at Microsoft.Extensions.Hosting.HostBuilder.CreateServiceProvider()
2020-05-22T08:15:28.574937777Z    at Microsoft.Extensions.Hosting.HostBuilder.Build()
2020-05-22T08:15:28.574945978Z    at Microsoft.Azure.IIoT.Services.All.Program.Main(String[] args) in D:\a\1\s\services\src\Microsoft.Azure.IIoT.Services.All\src\Program.cs:line 26
```
which was surprising for us, as we haven't changed anything. We decided to redeploy it using the latest version of the repository and the deployment script. The microservices start just fine, but we have troubles connecting to the OPC UA server running on the IotEdge machine. The Twin module logs return an error:
```
[08:30:07 INF { name: "b7001faa-474a-45e0-9446-c82d92a29f69", sessionId: 9, url: "opc.tcp://172.17.0.1:26545/" }] Try to connect via opc.tcp://172.17.0.1:26545/ in 5000 ms... 
[08:30:07 WRN { name: "32b6fdb7-67ef-4dfe-b70d-c09dbbda77ce", sessionId: 6, url: "opc.tcp://172.17.0.1:26544/" }] Establishing unencrypted connection. 
[08:30:07 INF { name: "32b6fdb7-67ef-4dfe-b70d-c09dbbda77ce", sessionId: 6, url: "opc.tcp://172.17.0.1:26544/" }] Creating session via opc.tcp://172.17.0.1:26544/. 
[08:30:08 INF { name: "32b6fdb7-67ef-4dfe-b70d-c09dbbda77ce", sessionId: 6, url: "opc.tcp://172.17.0.1:26544/" }] Error establishing a connection: BadNotConnected creating session via opc.tcp://172.17.0.1:26544/ - retry.
```
but we're able to connect to the server on 172.17.0.1:26544 and browse it using UA Expert just fine. My colleague suggested that they may have been some change regarding the certification of the servers that we're unaware of, but that's only an unconfirmed guess. The server also does appear on the REST API Get list of endpoints endpoint of Registry microservice, but it's status is NotReachable.

marcschier, 2020-05-22:
Hi @Bartor, thanks for submitting this issue.    

Regarding the first issue, in 2.7.x we moved to managed service identity to access KeyVault instead of a SP + known password. Managed service identity (MIS) is a resource that must be deployed and as such you will unfortunately also need to re-run the deployment script on top of an existing deployment so that the Azure resources are "adjusted".   As we only really test upgrade from a previous release to a new release, any in between preview->preview updates (where the container is pulled again and again) might hit such issue.  

However, the second issue you mention is interesting.  One reason this happens could be this:  If the Twin and publisher are not using a volume mapping to persist their certificate store they on restart will recreate their certificate.   A properly configured server is supposed to reject the connection if a new unknown certificate is not in its trust list.   One might find out more on the server, e.g. looking at the server logs for the connection attempt.  

If that is not the issue here, would you mind sharing more of the Twin module log?

Bartor, 2020-05-25:
Hello,
sorry for replying so late, was out of office for the weekend. Anyway, unfortunately redeploying using the script on the existing resource groups yields multiple new errors and as an inexperienced Azure user I'm not quite able to resolve all of them well, and that's probably not a case to discuss here.

Regarding the second issue; I've checked the Twin service with `docker inspect` and `Config.Volumes` is set tu `null`. There is a bind mount of `/var/run/iotedge/workload.sock`, but except from that, there seem to be no volumes used by the module containers. We haven't changed anything in the Container Create Options when deploying module from Azure Portal, so I'm not sure  why isn't there anything there. 

I've also dumped the logs from freshly started Twin module (and replaced some ids with placeholder, although I'm not sure if they contain any confidential information). Here are the logs:

```
[06:35:46 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Connecting all clients to <NAME> using MqttOverTcp. 
[06:35:47 INF Root] Initiating prometheus at port 9701/metrics 
[06:35:47 INF Root] Prometheus metric server started. 
[06:35:47 WRN Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Bypassing certificate validation for client. 
[06:35:47 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Running in iotedge context. 
[06:35:48 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] 0: Module LenovoServer_twin reconnected due to Connection_Ok. 
[06:35:48 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Initialize device twin for LenovoServer - twin 
[06:35:48 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Applying initial desired state. 
[06:35:48 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Reporting currently initial state. 
[06:35:48 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Module Host started. 
[06:39:42 INF Root] Module exits... 
[06:39:42 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Stopping Module Host... 
[06:39:43 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Connecting all clients to <NAME> using MqttOverTcp. 
[06:39:44 INF Root] Initiating prometheus at port 9701/metrics 
[06:39:44 INF Root] Prometheus metric server started. 
[06:39:44 WRN Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Bypassing certificate validation for client. 
[06:39:44 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Running in iotedge context. 
[06:39:45 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] 0: Module LenovoServer_twin reconnected due to Connection_Ok. 
[06:39:45 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Initialize device twin for LenovoServer - twin 
[06:39:45 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Applying initial desired state. 
[06:39:45 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Reporting currently initial state. 
[06:39:45 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Module Host started. 
[06:49:09 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] New settings processed. 
[06:49:17 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] New settings processed. 
[06:51:32 INF Root] Module exits... 
[06:51:32 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Stopping Module Host... 
[06:51:49 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Connecting all clients to <NAME> using MqttOverTcp. 
[06:51:50 INF Root] Initiating prometheus at port 9701/metrics 
[06:51:50 INF Root] Prometheus metric server started. 
[06:51:50 WRN Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Bypassing certificate validation for client. 
[06:51:50 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Running in iotedge context. 
[06:51:51 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] 0: Module LenovoServer_twin disconnected due to Retry_Expired - now Disconnected... 
[06:52:04 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] 0: Module LenovoServer_twin reconnected due to Connection_Ok. 
[06:52:05 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Initialize device twin for LenovoServer - twin 
[06:52:05 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Applying initial desired state. 
[06:52:05 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Reporting currently initial state. 
[06:52:05 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Module Host started. 
[06:52:05 INF Root] Module reset... 
[06:52:05 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Stopping Module Host... 
[06:52:05 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Module Host stopped. 
[06:52:05 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] 1: Module LenovoServer_twin disconnected due to Client_Close - now Disabled... 
[06:52:05 INF Root] Stopped prometheus metric server 
[06:52:05 INF Microsoft.Azure.IIoT.OpcUa.Edge.Supervisor.Services.SupervisorServices] All twins stopped - supervisor exiting... 
[06:52:05 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Connecting all clients to <NAME> using MqttOverTcp. 
[06:52:05 INF Root] Initiating prometheus at port 9701/metrics 
[06:52:05 INF Root] Prometheus metric server started. 
[... cut ...]
[06:52:31 INF Microsoft.Azure.IIoT.Modules.OpcUa.Twin.ModuleProcess+TwinContainerFactory] Twin <DEVICE 3 ID> endpoint updated to opc.tcp://172.17.0.1:26544/. 
[06:52:31 INF Microsoft.Azure.IIoT.Modules.OpcUa.Twin.ModuleProcess+TwinContainerFactory] Reporting currently initial state. 
[06:52:31 INF Microsoft.Azure.IIoT.Modules.OpcUa.Twin.ModuleProcess+TwinContainerFactory] Module Host started. 
[06:52:31 INF Microsoft.Azure.IIoT.OpcUa.Edge.Supervisor.Services.SupervisorServices] <DEVICE 3 ID> twin started. 
[06:52:31 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] New settings processed. 
```

and that's pretty much all.

Bartor, 2020-05-28:
Hi once again,
we've tackled the problem with another approach. It seems that the Twin module had problem actually connecting to the server because of networking issues. We connected our OPC UA server with the Edge Module in the same Docker network and it seems to work fine now.

closed 2020-05-28 by Bartor

marcschier, 2020-05-28:
Thanks for the update @Bartor!