Azure/Industrial-IoT #513: Cap on Messages sent to IoT Hub in OPC Publisher

saurabhsadhu, 2020-05-29:
I am using the OPC Publisher module available at Azure Marketplace. I am connecting this module to KepServerEx which is configured to generate data on 10,000  nodes per second. So I should 
be getting 10,000 messages per second on the OPC Publisher. The publisher gets approximately 10,000 messages/second but it doesn't send more than 600-700 messages/second to the IoT Edge route defined. Is there any limitation on number of messages sent via an IoT Edge route? And how can I increase the number of messages sent to IoT Edge route.  

Following are my container create options for OPC Publisher module:

`
{
    "Hostname": "publisher",
    "Cmd": [
        "publisher",
        "--pf=/appdata/pn.json",
        "--di=60",
        "--ih=Mqtt",
        "--to",
        "--aa",
        "--si=0",
        "--ms=262144"
    ],
    "HostConfig": {
        "PortBindings": {
            "62222/tcp": [
                {
                    "HostPort": "62222"
                }
            ]
        },
        "Binds": [
            "/home/user1/iiotedge:/appdata"
        ],
        "ExtraHosts": [
            "localhost:127.0.0.1"
        ]
    }
}
`

And the route on iotedge is set as :

 `FROM /messages/* INTO BrokeredEndpoint("/modules/moduleone/inputs/opcmessage")`

**Screenshots**
[image]

marcschier, 2020-05-30:
Samples != messages.   Messages are packets sent to IoTHub containing multiple Samples.

saurabhsadhu, 2020-06-01:
> Samples != messages. Messages are packets sent to IoTHub containing multiple Samples.
But there is no such thing as Samples mentioned in the logs. Can you please explain what is Messages invoked on trigger, Messages sent to IoT Hub and SinkBlock input count?

saurabhsadhu, 2020-06-01:
I want a solution similar to the issue - [Throughput issues](https://github.com/Azure/azure-iot-sdk-java/issues/739). 
The solution in this issue is for DeviceClient , is there something similar for ModuleClient?

marcschier, 2020-06-02:
Messages invoked == monitored item messages received through opc ua client publish service set.  Messages sent to IoT Hub == IoT hub messages containing multiple aforementioned "messages" (which I called samples as they are typically sampled but not always)

saurabhsadhu, 2020-06-03:
Okay so what is meant by Sinkblock input count ? 
Also I am receiving only 100 messages each second on my receivng module, even though I see that OPCPublisher is sending 600 messages per second. How can improve the performance?

cristipogacean, 2020-06-03:
Would it be an option for you to enable batching mechanism of the publisher? this way the publisher is packing multiple the OPC samples in a single iot hub message as an array so the throughput will be optimized. The drawback is that you need to parse the batched array at the receiver side. 
The parameters are : 
--bs ='number of data-changes to be awaited for batching'  
--di= 'trigger batching interval in seconds'
Triggering is made either when bs or di are met.
E.g. --bs=100 --di=5

saurabhsadhu, 2020-06-05:
I set the batch size to 100 but I am still getting only one message and not an array. And according to the documentation, --di is the diagnostic interval and not batching interval.

cristipogacean, 2020-06-05:
sorry, my mistake, I meant --si, not --di

saurabhsadhu, 2020-06-05:
I am running Publisher with the following configuration:
```
   "publisher",
    "--pf=/appdata/pn.json",
    "--di=60",
    "--to",
    "--aa",
    "--ih=Mqtt",
    "--si=5",
    "--bs=100"
```
And I am getting this output:
[12:01:17 INF] 
   DIAGNOSTICS INFORMATION
   =======================
   # Messages invoked by the message trigger: 782390
   # Messages Sent to IoT Hub: 46841
   # Number of connection retries since last error: 0
   # EncodingBlock input/output count: 0/0
   # BatchBlock output count: 0
   # SinkBlock input count: 735548
   =======================
 (Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.DataFlowProcessingEngine)
[12:02:17 INF] 
   DIAGNOSTICS INFORMATION
   =======================
   # Messages invoked by the message trigger: 1188890
   # Messages Sent to IoT Hub: 70815
   # Number of connection retries since last error: 0
   # EncodingBlock input/output count: 0/0
   # BatchBlock output count: 0
   # SinkBlock input count: 1118074
   =======================
 (Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.DataFlowProcessingEngine)

Are these ecpected number? Can I improve these?

cristipogacean, 2020-06-05:
which version are you using ? the batching was only implemented starting with 2.7.105 version. Can you try out either 2.7.105, or the :preview tag?

saurabhsadhu, 2020-06-09:
I tried linux-amd64 tag and it's sending batched messages without --bs. I am just using --si and --ms. This solves my issue.
Thanks.

closed 2020-06-09 by saurabhsadhu