Azure/Industrial-IoT #516: Not able to get configuration from KeyVault  to run micro-services locally

payalgaikwad42, 2020-06-01:
Hi, 
I want to deploy the Micro-services locally on my machine. I am using following documentation given in this repository -
These documentation gives me steps to deploy minimum azure resources required to run micro-services locally. I have all the required azure resources running on my azure portal using following documentation -
https://github.com/Azure/Industrial-IoT/blob/master/docs/deploy/howto-deploy-local.md

I am facing issue when i ran the micro-services locally as mentioned in following link- 
https://github.com/Azure/Industrial-IoT/blob/master/docs/deploy/howto-run-microservices-locally.md
This is some issue regarding getting configuration from key vault it says access denied. forbidden errror. 

> [10:34:55 INF Microsoft.Azure.IIoT.Auth.Clients.Default.MsiAuthenticationClient] No managed service identity configured for this service.
> [10:34:58 DBG Microsoft.Azure.IIoT.Auth.Clients.Default.KeyVaultAuthentication+KeyVaultTokenSource] Try acquiring token for KeyVault using Microsoft.Azure.IIoT.Auth.Clients.Default.MsiAuthenticationClient.
> [10:34:58 DBG Microsoft.Azure.IIoT.Auth.Clients.Default.KeyVaultAuthentication+KeyVaultTokenSource] Try acquiring token for KeyVault using Microsoft.Azure.IIoT.Auth.Clients.Default.AppAuthenticationClient.
> [10:34:58 DBG Microsoft.Azure.IIoT.Auth.Clients.Default.KeyVaultAuthentication+KeyVaultTokenSource] Try acquiring token for KeyVault using Microsoft.Azure.IIoT.Auth.Clients.Default.DevAuthenticationClient.
> [10:35:05 INF Microsoft.Azure.IIoT.Auth.Clients.Default.DevAuthenticationClient] Successfully acquired token for KeyVault with AzureAD:077bb374-6b86-417c-8d8c-d17673df6734->https://vault.azure.net.
> [10:35:05 DBG Microsoft.Azure.IIoT.Auth.Clients.Default.KeyVaultAuthentication+KeyVaultTokenSource] Successfully acquired token for KeyVault using Microsoft.Azure.IIoT.Auth.Clients.Default.DevAuthenticationClient.
> Unhandled exception. System.AggregateException: One or more errors occurred. (A keyvault uri was provided could not access keyvault at the address. If you want to read configuration from keyvault, make sure the keyvault is reachable, the required permissions are configured on keyvault and authentication provider information is available. Sign into Visual Studio or Azure CLI on this machine and try again.)
>  ---> Microsoft.Azure.IIoT.Exceptions.InvalidConfigurationException: A keyvault uri was provided could not access keyvault at the address. If you want to read configuration from keyvault, make sure the keyvault is reachable, the required permissions are configured on keyvault and authentication provider information is available. Sign into Visual Studio or Azure CLI on this machine and try again.
>  ---> Microsoft.Azure.KeyVault.Models.KeyVaultErrorException: Operation returned an invalid status code 'Forbidden'
>    at Microsoft.Azure.KeyVault.KeyVaultClient.GetSecretWithHttpMessagesAsync(String vaultBaseUrl, String secretName, String secretVersion, Dictionary`2 customHeaders, CancellationToken cancellationToken)
>    at Microsoft.Azure.KeyVault.KeyVaultClientExtensions.GetSecretAsync(IKeyVaultClient operations, String vaultBaseUrl, String secretName, CancellationToken cancellationToken)
>    at Microsoft.Extensions.Configuration.ConfigurationEx.KeyVaultConfigurationProvider.ValidateReadSecretAsync(String secretName) in C:\HHI\RESTAPIOPC\Industrial-IoT\common\src\Microsoft.Azure.IIoT.Auth.ActiveDirectory\src\Extensions\ConfigurationEx.cs:line 232
>    at Microsoft.Extensions.Configuration.ConfigurationEx.KeyVaultConfigurationProvider.CreateInstanceAsync(IConfigurationRoot configuration, Boolean allowInteractiveLogon, String keyVaultUrlVarName, Boolean lazyLoad) in C:\HHI\RESTAPIOPC\Industrial-IoT\common\src\Microsoft.Azure.IIoT.Auth.ActiveDirectory\src\Extensions\ConfigurationEx.cs:line 197
>    --- End of inner exception stack trace ---
>    at Microsoft.Extensions.Configuration.ConfigurationEx.KeyVaultConfigurationProvider.CreateInstanceAsync(IConfigurationRoot configuration, Boolean allowInteractiveLogon, String keyVaultUrlVarName, Boolean lazyLoad) in C:\HHI\RESTAPIOPC\Industrial-IoT\common\src\Microsoft.Azure.IIoT.Auth.ActiveDirectory\src\Extensions\ConfigurationEx.cs:line 200
>    at Microsoft.Extensions.Configuration.ConfigurationEx.KeyVaultConfigurationProvider.CreateInstanceAsync(Boolean allowInteractiveLogon, Boolean singleton, IConfigurationRoot configuration, String keyVaultUrlVarName) in C:\HHI\RESTAPIOPC\Industrial-IoT\common\src\Microsoft.Azure.IIoT.Auth.ActiveDirectory\src\Extensions\ConfigurationEx.cs:line 164
>    --- End of inner exception stack trace ---
>    at System.Threading.Tasks.Task.ThrowIfExceptional(Boolean includeTaskCanceledExceptions)
>    at System.Threading.Tasks.Task`1.GetResultCore(Boolean waitCompletionNotification)
>    at System.Threading.Tasks.Task`1.get_Result()
>    at Microsoft.Extensions.Configuration.ConfigurationEx.AddFromKeyVault(IConfigurationBuilder builder, ConfigurationProviderPriority providerPriority, Boolean allowInteractiveLogon, Boolean singleton, String keyVaultUrlVarName) in C:\HHI\RESTAPIOPC\Industrial-IoT\common\src\Microsoft.Azure.IIoT.Auth.ActiveDirectory\src\Extensions\ConfigurationEx.cs:line 49
>    at Microsoft.Azure.IIoT.Services.All.Startup..ctor(IWebHostEnvironment env, IConfiguration configuration) in C:\HHI\RESTAPIOPC\Industrial-IoT\services\src\Microsoft.Azure.IIoT.Services.All\src\Startup.cs:line 45
> --- End of stack trace from previous location where exception was thrown ---
>    at Microsoft.Extensions.DependencyInjection.ActivatorUtilities.ConstructorMatcher.CreateInstance(IServiceProvider provider)
>    at Microsoft.Extensions.DependencyInjection.ActivatorUtilities.CreateInstance(IServiceProvider provider, Type instanceType, Object[] parameters)
>    at Microsoft.AspNetCore.Hosting.GenericWebHostBuilder.UseStartup(Type startupType, HostBuilderContext context, IServiceCollection services)
>    at Microsoft.AspNetCore.Hosting.GenericWebHostBuilder.<>c__DisplayClass12_0.<UseStartup>b__0(HostBuilderContext context, IServiceCollection services)
>    at Microsoft.Extensions.Hosting.HostBuilder.CreateServiceProvider()
>    at Microsoft.Extensions.Hosting.HostBuilder.Build()
>    at Microsoft.Azure.IIoT.Services.All.Program.Main(String[] args) in C:\HHI\RESTAPIOPC\Industrial-IoT\services\src\Microsoft.Azure.IIoT.Services.All\src\Program.cs:line 26
> C:\HHI\RESTAPIOPC\Industrial-IoT\services\src\Microsoft.Azure.IIoT.Services.All\src\bin\Debug\netcoreapp3.1\Microsoft.Azure.IIoT.Services.All.exe (process 3636) exited with code 0.
> To automatically close the console when debugging stops, enable Tools->Options->Debugging->Automatically close the console when debugging stops.
> Press any key to close this window . . .

I have checked my key vault settings. I have access policy defined which gives my manage identity permissions to access the key vault. 
Any help regarding this would be greatly appreciated. 
Thank you.

marcschier, 2020-06-02:
Did you run `az login` as described in the doc?   You can also use Azure Service Connections in the Visual Studio Options, if you have issues there, try "re-adding" your account as this seems to fix stale tokens for us.

marcschier, 2020-06-04:
Please reopen if this was not the issue.

closed 2020-06-04 by marcschier