Azure/Industrial-IoT #517: OPC Publisher continuously restarts after an error

toddysm, 2020-06-01:
**Describe the bug**
We see the OPC Publisher module continuously restarting with the following message in the logs:
```
2020-06-01 04:57:45.412 +00:00 [VRB] [Exit] "Informational": MqttIotHubAdapter#35236192 ProcessMessage MqttIotHubAdapter#0, PUBLISH 
2020-06-01 04:57:45.412 +00:00 [VRB] [Exit] "Informational": MqttIotHubAdapter#35236192 ChannelRead MqttIotHubAdapter#0 
2020-06-01 04:57:45.413 +00:00 [VRB] [Enter] "Informational": MqttIotHubAdapter#35236192 ChannelReadComplete (MqttIotHubAdapter#0) 
2020-06-01 04:57:45.413 +00:00 [VRB] [Exit] "Informational": MqttIotHubAdapter#35236192 ChannelReadComplete MqttIotHubAdapter#0 
2020-06-01 04:57:45.482 +00:00 [VRB] [Exit] "Informational": ErrorDelegatingHandler#63993496 ExecuteWithErrorHandlingAsync ErrorDelegatingHandler.ExecuteWithErrorHandlingAsync 
2020-06-01 04:57:45.486 +00:00 [VRB] [Exit] "Informational": RetryDelegatingHandler#39070558 SendTwinGetAsync CancellationToken#51484875 
2020-06-01 04:57:45.490 +00:00 [INF] Initialize device twin for at-iot-gw-home-bsroom-01 - at-iot-gw-home-bsroom-01-opc-publisher
2020-06-01 04:57:45.754 +00:00 [DBG] [Info] "Informational": MqttTransportHandler#46964992 OnTransportClosedGracefully OnTransportClosedGracefully 
2020-06-01 04:57:45.759 +00:00 [DBG] [Info] "Informational": RetryDelegatingHandler#39070558 HandleDisconnect Transport disconnected: closed by application. 
2020-06-01 04:57:45.760 +00:00 [VRB] [Enter] "Informational": InternalClient#45596481 OnConnectionStatusChanged (Disabled, Client_Close) 
2020-06-01 04:57:45.761 +00:00 [INF] 1: Module at-iot-gw-home-bsroom-01_at-iot-gw-home-bsroom-01-opc-publisher disconnected due to "Client_Close" - now "Disabled"...
2020-06-01 04:57:45.761 +00:00 [VRB] [Exit] "Informational": InternalClient#45596481 OnConnectionStatusChanged Disabled, Client_Close 
2020-06-01 04:57:45.770 +00:00 [VRB] [Enter] "Informational": MqttIotHubAdapter#35236192 WriteAsync (MqttIotHubAdapter#0, DisconnectPacket[Type=DISCONNECT, QualityOfService=AtMostOnce, Duplicate=False, 
Retain=False]) 
2020-06-01 04:57:45.771 +00:00 [VRB] [Exit] "Informational": MqttIotHubAdapter#35236192 WriteAsync MqttIotHubAdapter#0 
2020-06-01 04:57:45.782 +00:00 [ERR] Error during module execution - restarting!
System.InvalidCastException: Unable to cast object of type 'Newtonsoft.Json.Linq.JValue' to type 'System.String'.
   at Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost.StartAsync(String type, String siteId, String serviceInfo, IProcessControl reset) in D:\a\1\s\common\src\Microsoft.Azure.IIoT.Hub.Module.Fr
amework\src\Hosting\ModuleHost.cs:line 152
   at Microsoft.Azure.IIoT.Modules.OpcUa.Publisher.ModuleProcess.RunAsync() in D:\a\1\s\modules\src\Microsoft.Azure.IIoT.Modules.OpcUa.Publisher\src\ModuleProcess.cs:line 89
2020-06-01 04:57:45.801 +00:00 [VRB] [Enter] "Informational": MqttIotHubAdapter#35236192 ChannelInactive (MqttIotHubAdapter#0) 
2020-06-01 04:57:45.829 +00:00 [VRB] [Enter] "Informational": MqttIotHubAdapter#35236192 Shutdown (MqttIotHubAdapter#0) 
2020-06-01 04:57:45.832 +00:00 [VRB] [Enter] "Informational": MqttIotHubAdapter#35236192 CloseIotHubConnection (CloseIotHubConnection) 
2020-06-01 04:57:45.836 +00:00 [VRB] [Exit] "Informational": MqttIotHubAdapter#35236192 CloseIotHubConnection CloseIotHubConnection 
2020-06-01 04:57:45.837 +00:00 [VRB] [Exit] "Informational": MqttIotHubAdapter#35236192 Shutdown MqttIotHubAdapter#0 
2020-06-01 04:57:45.838 +00:00 [VRB] [Exit] "Informational": MqttIotHubAdapter#35236192 ChannelInactive MqttIotHubAdapter#0 
2020-06-01 04:57:45.868 +00:00 [INF] Connecting all clients to at-iot-gw-home-bsroom-01 using "MqttOverTcp".
2020-06-01 04:57:45.874 +00:00 [DBG] Starting Module Host...
2020-06-01 04:57:45.875 +00:00 [WRN] Bypassing certificate validation for client.
2020-06-01 04:57:45.875 +00:00 [INF] Running in iotedge context.
2020-06-01 04:57:45.906 +00:00 [DBG] [Info] "Informational": IotHubConnectionString#54135081 .ctor IAuthenticationMethod is AuthenticationWithTokenRefresh: ModuleAuthenticationWithHsm#5773521 
2020-06-01 04:57:45.907 +00:00 [VRB] [Associate] "Informational": IotHubConnectionString#54135081 TokenRefresher IotHubConnectionString#54135081 ModuleAuthenticationWithHsm#5773521 [{2}]<-->[{3}]
2020-06-01 04:57:45.907 +00:00 [VRB] [Enter] "Informational": InternalClient#63094882 InternalClient_ctor (Microsoft.Azure.Devices.Client.ITransportSettings[1], DeviceClientPipelineBuilder#49924125) 
```
Particularly the issue seems to be with the following cast error:
```
System.InvalidCastException: Unable to cast object of type 'Newtonsoft.Json.Linq.JValue' to type 'System.String'.
   at Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost.StartAsync(String type, String siteId, String serviceInfo, IProcessControl reset) in D:\a\1\s\common\src\Microsoft.Azure.IIoT.Hub.Module.Fr
amework\src\Hosting\ModuleHost.cs:line 152
   at Microsoft.Azure.IIoT.Modules.OpcUa.Publisher.ModuleProcess.RunAsync() in D:\a\1\s\modules\src\Microsoft.Azure.IIoT.Modules.OpcUa.Publisher\src\ModuleProcess.cs:line 89
```

This is related to the container configuration, however looking at the code only it is not clear that the issue is. Here is the Container Create Options that are set for the module in Azure Portal:
```
{
  "Hostname": "publisher",
  "Cmd": [
    "--ap=/appdata",
    "--pf=/appdata/pn.json",
    "--site=bsroom-01",
    "--di=60",
    "--logfile=/appdata/publisher.log",
    "--loglevel=verbose",
    "--fetchname=True",
    "--appcertstoretype=X509Store",
    "--aa",
    "--to",
    "--si=0",
    "--ms=0",
    "--ih=Amqp"
  ],
  "HostConfig": {
    "Binds": [
      "/appdata:/appdata"
    ]
  }
}
```

**Desktop (please complete the following information):**
 - OS: Raspbian Buster
 - IoT Edge Runtime Version: iotedge 1.0.9

toddysm, 2020-06-02:
@marcschier Thank you for your offline help. I did two things:
1. Tried to remove the `site` setting as you suggested but the `latest/2.6.146` continued to fail with the same error
2. Getting the latest version `2.7.107` solved the above issue but it seems it is not pulling any  data from the OPC server

Here are the logs when the module starts:
```
2020-06-02 20:15:50.810 +00:00 [INF] Connecting all clients to at-iot-gw-home-bsroom-01 using "MqttOverTcp".
2020-06-02 20:15:50.949 +00:00 [INF] Initiating prometheus at port 9702/metrics
2020-06-02 20:15:51.161 +00:00 [INF] Prometheus metric server started.
2020-06-02 20:15:51.209 +00:00 [INF] Starting module OpcPublisher version 2.7.107.2964.
2020-06-02 20:15:51.223 +00:00 [WRN] Bypassing certificate validation for client.
2020-06-02 20:15:51.239 +00:00 [INF] Running in iotedge context.
2020-06-02 20:15:52.977 +00:00 [INF] 0: Module at-iot-gw-home-bsroom-01_at-iot-gw-home-bsroom-01-opc-publisher reconnected due to "Connection_Ok".
2020-06-02 20:15:53.266 +00:00 [INF] Initialize device twin for at-iot-gw-home-bsroom-01 - at-iot-gw-home-bsroom-01-opc-publisher
2020-06-02 20:15:53.566 +00:00 [INF] Applying initial desired state.
2020-06-02 20:15:53.579 +00:00 [INF] Reporting currently initial state.
2020-06-02 20:15:53.675 +00:00 [INF] Module Host started.
2020-06-02 20:15:53.688 +00:00 [INF] Creating new worker...
2020-06-02 20:15:53.761 +00:00 [INF] File /appdata/pn.json has changed, reloading...
2020-06-02 20:15:53.853 +00:00 [INF] Read 1 items from published nodes file in "00:00:00.0893189"
2020-06-02 20:15:53.955 +00:00 [INF] Converted items to jobs in "00:00:00.1010797"
2020-06-02 20:15:54.224 +00:00 [INF] Starting worker 'StandalonePublisher_0'...
2020-06-02 20:15:54.236 +00:00 [INF] Worker WorkerId: StandalonePublisher_0 {}
2020-06-02 20:15:54.275 +00:00 [INF] Worker: StandalonePublisher_0, start processing new job: Standalone_at-iot-gw-home-bsroom-01_at-iot-gw-home-bsroom-01-opc-publisher, mode: "Active"
2020-06-02 20:15:54.680 +00:00 [WRN] Subscription not registered
2020-06-02 20:16:54.492 +00:00 [INF] 
   DIAGNOSTICS INFORMATION for Engine : opc.tcp://192.168.7.190:4840_92e098a7bcc7d553d6fdc2cf5fe5c0836e11da52
   # Ingestion StartTime - Duration   : "0001-01-01T00:00:00.0000000" - 0 seconds
   # Ingress DataChanges (from OPC)   : 0 - 0/second
   # Ingress ValueChanges (from OPC)  : 0 - 0/second
   # Ingress BatchBlock buffer size   : 0
   # Encoding Block input/output size : 0/0
   # Encoder Notifications processed  : 0
   # Encoder Notifications dropped    : 0
   # Encoder IoT Messages processed   : 0
   # Encoder avg Notifications/Message: 0
   # Encoder avg IoT Message body size: 0
   # Outgress Batch Block buffer size : 0
   # Outgress input buffer count      : 0
   # Outgress IoT message count       : 0 - 0/second
   # Connection retries               : 0
```

It seems that the module starts and reads the `pn.json` but it doesn't collect any information from it. The server is working because we are able to connect using UA Expert.

Occasionally, we see the following message in the logs:
```
2020-06-02 20:19:24.875 +00:00 [WRN] Failed to get or create session 5658e347c46f10c69d1d8a0137366f3156799b86 due to One or more errors occurred. (OPC UA configuration not valid).
```

How can we troubleshoot this new issue?

Also, why is the timestamp always the beginning of time?

cristipogacean, 2020-06-03:
Can you try removing the '--appcertstoretype=X509Store' from your configuration. I think this setting is only working on a windows based host system. The default is for the AppCertStoreType is Directory

toddysm, 2020-06-03:
@cristipogacean I tried that but it doesn't seem to work. I increased the debug level and here is the output:
```
2020-06-03 15:30:43.110 +00:00 [INF] Module Host started.
2020-06-03 15:30:43.122 +00:00 [INF] Creating new worker...
2020-06-03 15:30:43.197 +00:00 [INF] File /appdata/pn.json has changed, reloading...
2020-06-03 15:30:43.200 +00:00 [DBG] Reading published nodes file ("00:00:00.0000045"
2020-06-03 15:30:43.288 +00:00 [INF] Read 1 items from published nodes file in "00:00:00.0882693"
2020-06-03 15:30:43.391 +00:00 [INF] Converted items to jobs in "00:00:00.1013629"
2020-06-03 15:30:43.675 +00:00 [INF] Starting worker 'StandalonePublisher_0'...
2020-06-03 15:30:43.682 +00:00 [DBG] Sending heartbeat...
2020-06-03 15:30:43.686 +00:00 [INF] Worker WorkerId: StandalonePublisher_0 {}
2020-06-03 15:30:43.707 +00:00 [DBG] Worker starting...
2020-06-03 15:30:43.707 +00:00 [DBG] Try querying available job...
2020-06-03 15:30:43.726 +00:00 [INF] Worker: StandalonePublisher_0, start processing new job: Standalone_at-iot-gw-home-bsroom-01_at-iot-gw-home-bsroom-01-opc-publisher, mode: "Active"
2020-06-03 15:30:44.919 +00:00 [DBG] Sending job processor heartbeat...
2020-06-03 15:30:51.826 +00:00 [VRB] [Exit] "Informational": MqttIotHubAdapter#14347911 ScheduleCheckConnectTimeout MqttIotHubAdapter#0 
2020-06-03 15:30:58.932 +00:00 [DBG] Not found: Session not available - endpointUrl: opc.tcp://192.168.7.190:4840
2020-06-03 15:30:58.935 +00:00 [DBG] CheckAsync succeeded.
2020-06-03 15:31:03.933 +00:00 [DBG] Not found: Session not available - endpointUrl: opc.tcp://192.168.7.190:4840
2020-06-03 15:31:03.940 +00:00 [DBG] CheckAsync succeeded.
2020-06-03 15:31:07.589 +00:00 [DBG] Failed to connect on opc.tcp://192.168.7.190:4840: BadRequestTimeout - try again...
2020-06-03 15:31:07.592 +00:00 [WRN] Failed to get or create session 5658e347c46f10c69d1d8a0137366f3156799b86 due to One or more errors occurred. (BadRequestTimeout).
2020-06-03 15:31:07.594 +00:00 [DBG] Not found: Session/Subscription not available - endpointUrl: opc.tcp://192.168.7.190:4840
2020-06-03 15:31:08.935 +00:00 [DBG] Not found: Session not available - endpointUrl: opc.tcp://192.168.7.190:4840
2020-06-03 15:31:08.939 +00:00 [DBG] CheckAsync succeeded.
2020-06-03 15:31:13.938 +00:00 [DBG] Not found: Session not available - endpointUrl: opc.tcp://192.168.7.190:4840
2020-06-03 15:31:13.943 +00:00 [DBG] CheckAsync succeeded.
2020-06-03 15:31:14.903 +00:00 [DBG] Sending job processor heartbeat...
2020-06-03 15:31:18.957 +00:00 [DBG] Not found: Session not available - endpointUrl: opc.tcp://192.168.7.190:4840
2020-06-03 15:31:18.961 +00:00 [DBG] CheckAsync succeeded.
2020-06-03 15:31:22.777 +00:00 [DBG] Failed to connect on opc.tcp://192.168.7.190:4840: BadRequestTimeout - try again...
2020-06-03 15:31:22.783 +00:00 [WRN] Failed to get or create session 5658e347c46f10c69d1d8a0137366f3156799b86 due to One or more errors occurred. (BadRequestTimeout).
2020-06-03 15:31:22.788 +00:00 [DBG] Not found: Session/Subscription not available - endpointUrl: opc.tcp://192.168.7.190:4840
2020-06-03 15:31:39.092 +00:00 [DBG] Failed to connect on opc.tcp://192.168.7.190:4840: BadRequestTimeout - try again...
2020-06-03 15:31:39.098 +00:00 [WRN] Failed to get or create session 5658e347c46f10c69d1d8a0137366f3156799b86 due to One or more errors occurred. (BadRequestTimeout).
2020-06-03 15:31:39.102 +00:00 [DBG] Not found: Session not available - endpointUrl: opc.tcp://192.168.7.190:4840
2020-06-03 15:31:39.104 +00:00 [DBG] CheckAsync succeeded.
2020-06-03 15:31:43.965 +00:00 [DBG] Identity at-iot-gw-home-bsroom-01; at-iot-gw-home-bsroom-01-opc-publisher
2020-06-03 15:31:43.979 +00:00 [INF] 
   DIAGNOSTICS INFORMATION for Engine : opc.tcp://192.168.7.190:4840_bd8ffeb0828845b9cf378c19e3a3e04c013f1a25
   # Ingestion StartTime - Duration   : "0001-01-01T00:00:00.0000000" - 0 seconds
   # Ingress DataChanges (from OPC)   : 0 - 0/second
   # Ingress ValueChanges (from OPC)  : 0 - 0/second
   # Ingress BatchBlock buffer size   : 0
   # Encoding Block input/output size : 0/0
   # Encoder Notifications processed  : 0
   # Encoder Notifications dropped    : 0
   # Encoder IoT Messages processed   : 0
   # Encoder avg Notifications/Message: 0
   # Encoder avg IoT Message body size: 0
   # Outgress Batch Block buffer size : 0
   # Outgress input buffer count      : 0
   # Outgress IoT message count       : 0 - 0/second
   # Connection retries               : 0

2020-06-03 15:32:44.919 +00:00 [DBG] Sending job processor heartbeat...
2020-06-03 15:32:59.511 +00:00 [DBG] Failed to connect on opc.tcp://192.168.7.190:4840: BadRequestTimeout - try again...
2020-06-03 15:32:59.515 +00:00 [WRN] Failed to get or create session 5658e347c46f10c69d1d8a0137366f3156799b86 due to One or more errors occurred. (BadRequestTimeout).
2020-06-03 15:32:59.518 +00:00 [DBG] Not found: Session not available - endpointUrl: opc.tcp://192.168.7.190:4840
```

So, it seems that the publisher cannot connect to the OPC server. We checked the server and we are also able to connect to it using UA Expert. Also, the Publisher container is able to ping the IP address. So, right now we are stuck with figuring out this connectivity issue and why it is failing.

We also tried older versions like 2.5.1, 2.5.2, and 2.5.3 - the errors we see there are from the OPC-UA library:
```
2020-06-03 15:46:36.064 +00:00 [INF] Publisher is running. Press CTRL-C to quit.
[... cut ...]
   at Opc.Ua.Client.CoreClientUtils.SelectEndpoint(String discoveryUrl, Boolean useSecurity, Int32 operationTimeout)
   at OpcPublisher.OpcSession.ConnectSessionAsync(CancellationToken ct) in D:\a\1\s\modules\src\Microsoft.Azure.IIoT.Modules.OpcUa.Publisher\src\OpcSession.cs:line 431
2020-06-03 15:47:09.541 +00:00 [DBG] Update node configuration file, version: 000000B2
```

It seems to be a connectivity issue between the Publisher and the OPC server but the logs are not enough to understand how to fix that. Any ideas?

toddysm, 2020-06-03:
@cristipogacean We tried all of those. There are no firewalls between the host and the OPC server. We even did a telnet on the port 4840 from within the container and telnet can connect to the IP address on that port. We also restarted the server, which should disconnect the sessions. 

One of the issues we used to have in the past is the security because the server uses self signed cert. Hence, we use the `-aa` switch to accept all certs. Any other suggestions?

cristipogacean, 2020-06-03:
Does the server have an endpoint that allows security policy none? using security policy none, the app certificates are disregarded. 
What kind of server is this? any chance to look into the logs of the server?

toddysm, 2020-06-04:
@cristipogacean Would you mind explaining what this means: __using security policy none, the app certificates are disregarded__?
The OPC server is a proprietary device from a German manufacturer Rittal. When we connect to the server via UA Expert, there are several messages that show up that are related to (we assume) how the protocol is implemented and how the device is configured.

Here the sequence:
[image] (we can change the host name to the IP address or add a DNS entry for the server if needed.

Yes, there is an endpoint with security `None`:
[image]

Also, the certificate is untrusted because it is self signed:
[image]

For the latter two, we don't have a lot of wiggle room to change because this is done by the manufacturer and we have no way to update the endpoint or the cert. So, we are stuck with them.

Any suggestions?

toddysm, 2020-06-04:
Just FYI, the connectivity to the server works from within the container. Here are all the tests we did:
```
# env    
HOSTNAME=at-iot-gw-home-bsroom-01-pub
HOME=/root
OLDPWD=/app
DOTNET_RUNNING_IN_CONTAINER=true
IOTEDGE_DEVICEID=at-iot-gw-home-bsroom-01
IOTEDGE_APIVERSION=2019-11-05
IOTEDGE_WORKLOADURI=unix:///var/run/iotedge/workload.sock
TERM=xterm
PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
IOTEDGE_AUTHSCHEME=sasToken
IOTEDGE_GATEWAYHOSTNAME=at-iot-gw-home-bsroom-01
IOTEDGE_MODULEID=at-iot-gw-home-bsroom-01-opc-publisher
ASPNETCORE_URLS=http://+:80
PWD=/
IOTEDGE_IOTHUBHOSTNAME=iotgw-ioth-dev-usw2-sandbox.azure-devices.net
RuntimeLogLevel=Information
IOTEDGE_MODULEGENERATIONID=637263279964864624
# ping 192.168.7.190
PING 192.168.7.190 (192.168.7.190) 56(84) bytes of data.
64 bytes from 192.168.7.190: icmp_seq=1 ttl=63 time=2.52 ms
64 bytes from 192.168.7.190: icmp_seq=2 ttl=63 time=4.11 ms
64 bytes from 192.168.7.190: icmp_seq=3 ttl=63 time=6.37 ms
^C
--- 192.168.7.190 ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 2003ms
rtt min/avg/max/mdev = 2.526/4.339/6.379/1.582 ms
# telnet 192.168.7.190 4840
Trying 192.168.7.190...
Connected to 192.168.7.190.
Escape character is '^]'.
```
We don't think this is a connectivity issue but an issue how the Publisher tries to connect to the OPC server.

To answer your question about the server logs, the manufacturer doesn't expose those and we have no way to look at the device logs.

cristipogacean, 2020-06-04:
@toddysm thanks for the details! I might now have an idea what's going wrong. 
Apparently the server uses it's hostname when responding to the GetEndpoints service call from the publisher. then the publisher is trying to connect using the hostname instead of the ip address. This is probably not properly resolved by the DNS in the container's context and we have the network timeout ... The UA expert is either capable to resolve the hostname since is not running in a container context (I assume) or acts smarter and eventually forces the connection using the provided IP address after failing to connect using the server's retrieved endpoint url. if that's the case, the server's behavior is not exactly compliant, but we should be smarter in the publisher as well and handle the issue - I'll add a work item on this.

you can try pinging the funny hostname "CMCIII-PU-..." and see what happens. You can maybe add  force host's dns to resolve it properly somehow (e.g. by adding it in the C:\Windows\System32\drivers\etc\hosts on a windows box) 

btw, on the point: *using security policy none, the app certificates are disregarded?* I meant that when a client connects to a server using security policy none, both client and the server will allow the connection even if the pair's certificate is not trusted - typically the apps will log a warning, but will continue the communication.

hope this helps you moving further.

cristipogacean, 2020-06-04:
hi @toddysm, I'm running out of ideas. Can you share more logs, or sync for a remote debugging session?

toddysm, 2020-06-04:
@cristipogacean We don't know what the problem was but after several (in the number of 10s if not 100s) restarts of the container and the OPC server, the things connected. We are totally puzzled for the reason why is this happening. I think we can close this but the root cause is unclear.

One suggestion if you have any influence into the Opc.Ua libraries - the logging there seems to be not very detailed. It will be good to have more detailed logging in that library with detailed messages that will help with troubleshooting. Also, if there is any caching in the libraries, it will be good to know how to clean it because my assumption was that this was a cache problem from the initially wrong configured hostname.

As I mentioned, everything else worked more reliably and responded with understandable errors. We even tried Python `asyncua` and it worked out of the box.

toddysm, 2020-06-04:
@cristipogacean I did speak too soon though. Here is the situation:
We managed to get this working with v 2.5.3 but switching to 2.7.107 breaks the container and it start cycling continuously and we cannot get logs out of it. The configuration is exactly the same, the only thing we changed was the container tag from 2.5.3 -> 2.7.107.

cristipogacean, 2020-06-04:
Good to see you're unlocked! and thanks for the feedback. There is no caching or something similar in the modules.
Can you please share the configuration? there are some changes in the arguments and some were obsoleted in 2.7.
If the config is same as above can you try removing the following:
    "--appcertstoretype=X509Store"
maybe you can also add:
     "--PkiRootPath=/appdata/pki"  this will set the certificates root path to your mounted volume

toddysm, 2020-06-08:
Hi @cristipogacean, we already tried removing the
```
--appcertstoretype=X509Store
```
but it didn't help. Adding
```
--PkiRootPath=/appdata/pki
```
didn't help either - the container fails at start and continues restarting.

closed 2020-10-19 by dacolgit