Azure/Industrial-IoT #541: OPCTwin exception - unable to discover OPC UA server

henioStraszny, 2020-06-08:
**Describe the bug**
I'm unable to connect Kepware OPC Server. Each time I turn on discovery (scanner) in Connected Factory App (section browser) OPCTwin module reports UnauthorizedException.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to `Connected Factory - solution`
2. Click on `Browse`
3. Set `Scan On` for tested IoTEdge device.
4. Execute: `iotedge logs device`

**Expected behavior**
Kepware OPC Server endpoints should be visible in Connected Factory solution.

**Logs**
```
[06:07:12 INF] Twin stopped. ({ endpointId: "Discovery", product: "OpcTwin_2.6.146.17357" })
[06:22:11 INF] New settings processed. (Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost)
[06:22:23 INF] Module exits... (null)
[06:22:23 INF] Stopping Module Host... (Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost)
[06:22:24 INF] Connecting all clients to XXXXX using MqttOverTcp. (Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory)
[06:22:24 INF] Application certificate with XXXXXXXXXXXXXXXXXXXXXX, CN=Azure IIoT OPC Twin found in the certificate store (Microsoft.Azure.IIoT.OpcUa.Protocol.Services.ClientServices)
[06:22:24 INF] Setting new application certificate XXXXXXXXXXXXXXXXXXXXXX, CN=Azure IIoT OPC Twin... (Microsoft.Azure.IIoT.OpcUa.Protocol.Services.ClientServices)
[06:22:24 INF] Adding own certificate to configured certificate store (Microsoft.Azure.IIoT.OpcUa.Protocol.Services.ClientServices)
[06:22:24 WRN] Bypassing certificate validation for client. (Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory)
[06:22:24 INF] Running in iotedge context. (Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory)
[06:22:24 INF] 0: Module XXXXXXXXXXXXXXXX reconnected due to Connection_Ok. (Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory)
[06:22:24 INF] Initialize device twin for XXXXXXXXXXXXXXXXXXXXX - opctwin (Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost)
[06:22:24 INF] Applying initial desired state. (Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost)
[06:22:24 INF] Reporting currently initial state. (Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost)
[06:22:24 INF] Module Host started. (Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost)
[06:22:35 INF] Discovery twin attached to module. (Microsoft.Azure.IIoT.OpcUa.Edge.Supervisor.Services.SupervisorServices)
[06:22:35 INF] New settings processed. (Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost)
[06:22:35 INF] Connecting all clients to procek-linux using MqttOverTcp. (Microsoft.Azure.IIoT.Modules.OpcUa.Twin.ModuleProcess+TwinContainerFactory)
[06:22:35 INF] Starting twin host... ({ endpointId: "Discovery", product: "OpcTwin_2.6.146.17357" })
[06:22:35 WRN] Bypassing certificate validation for client. (Microsoft.Azure.IIoT.Modules.OpcUa.Twin.ModuleProcess+TwinContainerFactory)
[06:22:36 INF] 0: Device Discovery disconnected due to Bad_Credential - now Disconnected... (Microsoft.Azure.IIoT.Modules.OpcUa.Twin.ModuleProcess+TwinContainerFactory)
[06:22:36 INF] Twin not authorized using given secret - exit host... ({ endpointId: "Discovery", product: "OpcTwin_2.6.146.17357" })
Microsoft.Azure.Devices.Client.Exceptions.UnauthorizedException: CONNECT failed: RefusedNotAuthorized
   at Microsoft.Azure.Devices.Client.Transport.Mqtt.MqttTransportHandler.OpenAsyncInternal(CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.Transport.Mqtt.MqttTransportHandler.OpenAsync(CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.Transport.ProtocolRoutingDelegatingHandler.OpenAsync(CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.Transport.ErrorDelegatingHandler.<>c__DisplayClass23_0.<<ExecuteWithErrorHandlingAsync>b__0>d.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at Microsoft.Azure.Devices.Client.Transport.ErrorDelegatingHandler.ExecuteWithErrorHandlingAsync[T](Func`1 asyncOperation)
   at Microsoft.Azure.Devices.Client.Transport.RetryDelegatingHandler.<>c__DisplayClass33_0.<<OpenAsyncInternal>b__0>d.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at Microsoft.Azure.Devices.Client.Transport.RetryDelegatingHandler.EnsureOpenedAsync(CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.InternalClient.OpenAsync()
   at Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory.DeviceClientAdapter.CreateAsync(String product, IotHubConnectionStringBuilder cs, String deviceId, ITransportSettings transportSetting, TimeSpan timeout, IRetryPolicy retry, Action onConnectionLost, ILogger logger) in D:\a\1\s\common\src\Microsoft.Azure.IIoT.Hub.Module.Client\src\Default\IoTSdkFactory.cs:line 446
   at Microsoft.Azure.IIoT.Utils.Try.Options[T](Func`1[] options)
[06:22:36 INF] Twin stopped. ({ endpointId: "Discovery", product: "OpcTwin_2.6.146.17357" })
```

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04
 - Browser Firefox
 - Version 77.0.1

jleger-ptc, 2020-06-08:
If you open up the [OPC UA Configuration Manager](https://www.kepware.com/getattachment/de80e240-765e-451a-afce-640d413891c3/opc-ua-configuration-manager-manual.pdf) for the Kepware server, can you confirm that the OPC Twin client certificate is trusted?

Client certificates are not auto-accepted by default, so you will need to grant it yourself. 

Similarly, if/when you attempt to wire telemetry up via the OPC Publisher, you will need to trust that certificate as well

henioStraszny, 2020-06-08:
And how can I export this certificate so then later I can register it on
OPC server?

jleger-ptc, 2020-06-08:
I believe the [guidance from the IIoT team](https://github.com/Azure/Industrial-IoT/issues/377#issuecomment-622984338) is to generate your own certificate and mount them into the Twin/Publisher containers.

Without provisioning the cert yourself, if/when the container is redeployed (upgrade, etc), a new self signed certificate will be generated on startup and you will need to go through this manual process again to establish the trust relationship between client/server

henioStraszny, 2020-06-10:
Thanks, for your response, I was trying to feed in the certificate, I've setup the mounted directory however it looks like OPC-Twin ignores that certificate, do you know what should be directory structure so application would be able to properly use certificate?

rowenwu, 2020-06-24:
Hi @henioStraszny, I'm the PM working on the roadmap for the team. Would you like to schedule some time to chat so I can better understand your use case and make sure that our roadmap meets your requirements? If so, send me an email rowwu@microsoft.com

erlendp, 2020-07-15:
Getting exactly the same error here. I don't think it's due to a connection failure to the OPC UA server, as the same error occurs whether or not there is an OPC server on the network. The error mentions an incorrect secret as the cause of authentication failure, which I'm guessing relates to the connection between the micro-service and twin module. I can confirm that the device connection string matches the connection string reported in the IoT Hub.

Steps to reproduce:

- Run One-click Connected Factory Installer.
- Create VM and run the IoT Edge Gateway Installer, providing connection string to IoT Hub created by Connected Factory Installer.
- Deploy OPC Twin and Publisher modules via Azure portal in IoT Hub.
- Edge device now appears in IoT Hub & when browsing in the Connected Factory app.
- As soon as discovery is enabled, an identical error to the one reported above occurs.

Have tried running edge device on both Ubuntu and Windows VMs, with the same results.

erlendp, 2020-07-17:
FYI - turns out the Connected Factory installer was installing old versions of the micro services. Once I switching these to match the OPC modules, everything started to work correctly.

hansgschossmann, 2020-10-02:
@henioStraszny have you verified that you are using the latest versions as @erlendp did?

hansgschossmann, 2021-06-15:
Pls ignore this comment. This is just a test comment.

closed 2023-01-16 by marcschier