Azure/Industrial-IoT #554: Duplicate published nodes are created for different credentials

amccollough-ptc, 2020-06-11:
**Describe the bug**
Once published, a node subscription can only be removed by exactly matching the credentials used to start publishing the node to begin with. This includes cases where credentials are entered incorrectly. All nodes created with those credentials will be permanently 'stuck' unless the original credentials can be recovered.

**To Reproduce**
Steps to reproduce the behavior:
1. Use the REST interface to `start` publishing a node with node id "exampleid" and a set of username/password credentials in the message body. e.g.
```
{
    "item": 
    {
        "nodeId" : "exampleid"
    },
    "header": {
        "elevation": {
            "type": "UserName",
            "value": {
                "user": "username1",
                "password": "password1"
            }
        }
    }
}
```
2. Get currently published nodes and confirm that "exampleid" is in the list
3. Use the REST interface to `stop` publishing the node but change the username and password in the message body. e.g.
```
{
    "item": 
    {
        "nodeId" : "exampleid"
    },
    "header": {
        "elevation": {
            "type": "UserName",
            "value": {
                "user": "username2",
                "password": "password2"
            }
        }
    }
}
```
4. Once again, check the list of published nodes and observe that the node is still being published.
5. Taking this a step further, you can also create many publishing entries for the **same** node with different sets of credentials, and they must each be cleaned up using only the original credentials.

**Expected behavior**
I would expect that unpublishing a node would cause the entry in the published nodes to be cleaned up regardless of which credentials are used. I would also only expect a single entry in the published nodes list for each node.

**Version**
I'm using a newly deployed 2.7.105 stack, deployed with the HELM charts.

amccollough-ptc, 2020-06-12:
I did some more testing and discovered some additional strange behavior.

If I start publishing a node with **invalid** user/password, then start publishing the any other node on that endpoint with **valid** user/password. The publisher module will produce an endless loop of errors in the log, and no subscription will be made to the UA Server for any of the valid nodes. I can go back and delete the bad entry and after ~10 seconds the errors will stop and a subscription will finally be created.

However, if I add the node entries in the opposite order: **Valid** first, then **invalid**, there are no issues at all. The subscription is made.

So it appears that when there are multiple entries for an endpoint, the publisher will only use the credentials of whichever was added first, even if the credentials are different. This is a problem for the user if they accidentally use invalid credentials to add a single node, that entire endpoint will be unable to transmit any data, even if valid credentials are used for other nodes, until the invalid node is removed.

marcschier, 2020-06-24:
The code removing or updating a node matches the credential in addition to the endpoint Url.   I will change it to only match the endpoint Url and also to remove not just the first, but *all* possible variables under the same endpoint url with the same node id.  This will support calling publish twice, whereby the second publish overrides the first ones credentials and publishing interval.

marcschier, 2020-06-24:
#573

closed 2020-06-25 by marcschier