Azure/Industrial-IoT #559: Is it possible to run the publisher as a standalone module?

AndreasDanfoss, 2020-06-16:
We want to run a small and scalable environment. But we only need the twin and publisher module to communicate with a robot. 

With the legacy mode, we can configure the publisher on the machine with the modules, and the twin module is recovering a C2D method request for reading values.

So is it possible or what is the minimum amount of resources to run the publisher? Where is it documented? I can see in the code the C2D methods are removed in the new version.

As I read it the publisher is getting is configuration from the CosmosDB?

marcschier, 2020-06-24:
We are currently planning a fully contained Edge hosting option for the entire platform.   Tentative for 3.0. 
  
At this point, publisher can either be configured via publishednodes.json file or via the API (and thus CosmosDB).

rowenwu, 2020-06-24:
hi Andreas, I'm the PM helping to coordinate our roadmap, and I'd like to better understand your requirements so we can capture them and align them with our roadmap (potentially for 3.0, as Marc called out). If you would like to hop on a call to chat, send me an email at rowwu@microsoft.com

marcschier, 2020-06-25:
Duplicate: #464

barnstee, 2020-09-17:
Standalone mode is supported from V2.7 onwards and documented in our [Ops manual](https://github.com/Azure/Industrial-IoT/blob/release/2.7.180/IIoT%20Platform%20Overview%20and%20Operations%20manual%20-%202.7.183.pdf), please try our latest version (2.7.183).

closed 2020-09-17 by barnstee