Azure/Industrial-IoT #574: Disvocery after restarting a OPC UA Server causes new IoT Devices in the IoT Hub

Asthmatiker, 2020-06-24:
**Is your feature request related to a problem? Please describe.**
I have the following setup: 
A server running a OPC UA server (running in docker container) and the Azure IoT Edge Modul.
The OPC UA server is running with a hardcoded IP and Port.
The OPC UA endpoint is activated and connected to Azure and everything is working fine.

When restarting the server, the OPC UA server will automatically start too.
After the restart, the activated endpoint is shown as disconnected.
After a dicovery over the engineering tool a new endpoint for the same OPC UA server is found. 
The old one (disconnected) is still listed.
The activation of the "new" is working just fine again.
But now there are two IoT Devices in the IoT Hub which are basicly the same (one is never used again).
Repeat the restart process to "create" more IoT Devices of the same OPC UA endpoint.

**Describe the solution you'd like**
The Azure IoT Edge Module should detect, that the "new" discovered OPC UA server is the same as the already present one and just reconnect to them.

**Additional context**
I am not 100% sure if the OPC UA server is configured wrong and that causes the issue. How does the Discovery Module decides when a OPC UA endpoint is a new one.

marcschier, 2020-06-24:
Hi @Asthmatiker - The application and endpoint identities (represented by the device Ids starting "uas"/"uac" and "uat") are created as follows: 

- Application Id: "uas" or "uac" + Hash(<IoT Edge Device or Site Id>, <Application Type>, <Application Uri>)
- Endpoint Id: "uat" + Hash (<Application Id>, <Endpoint Url reported by GetEndpoint>, SecurityMode, SecurityPolicyString>

It is possible that the Application Uri or Endpoint Url changes (e.g. the host part of the Uri if it is a container without a fixed host name assigned to the container and the server using the GetHostName call to get its hostname, which changes per restart).

marcschier, 2020-07-01:
Would be great to see what the issue was here, but closing for now.   If the server properties were stable, and it wasn't the issue, please re-open.

closed 2020-07-01 by marcschier