Azure/Industrial-IoT #586: Troubleshooting help for Job Orchestrator connectivity problems

pcameron-ptc, 2020-06-29:
**Describe the bug**
We recently encountered a problem when deploying the Industrial IoT microservices to a "managed" AKS cluter. The problem turned out to be that the edge device hosting the publisher IP was not whitelisted, so that it could not communicate with the job scheduler endpoint. 

Here is the publisher module log when the publisher is running at the "info" log level:
[image]

Here is the publisher module log when the publisher is running at the "debug" log level:
[image]

Whitelisting the edge device IP allowed the publisher to connect to the job scheduler service and ultimately deliver

**Expected behavior**
It is non-obvious from the info level log that there is any problem at all. An explicit warning or error would help users diagnose this problem without having to redeploy the module at a more verbose log level.

marcschier, 2020-07-01:
Thanks for opening @pcameron-ptc - we will address this with better warning/error in 2.8.

barnstee, 2020-10-06:
We switched the log level from debug to info for this error in the 2.7.199 release.

closed 2020-10-06 by barnstee