Azure/Industrial-IoT #588: Publisher sends double message

alonsors-809, 2020-06-30:
Hi,
I am currently facing an issue with the opc publisher located on an IoT edge device that is sending data with almost the same timestamp. I am not sure if it is because I am using the version 2.5.1 (“mcr.microsoft.com/iotedge/opc-publisher:2.5.1”) or is the configuration of the node but it seems that if the value changes during the sampling or publishing intervals, the publisher get both values. I need that every 5 minutes the opc publisher send the value of the tag, and it doesn’t matter if the value change or not but only one message of each 5 minutes.

Example:

"OpcNodes": [
      {
        "Id": "ns=1;s=tag01
        "OpcSamplingInterval": 300000,
 "OpcPublishingInterval": 300000,
        "HeartbeatInterval": 300
      },
]

Results:
[image]
 
Does these three intervals (OpcSamplingInterval, OpcPublishingInterval, HeartbeatInterval) that are configured on the node must be different?

If anyone knows how to avoid receiving this duplicate, would be grate.

cristipogacean, 2020-06-30:
Hi, would it be possible for you to try the preview (2.7.xxx) version and check if you see the same behavior?

alonsors-809, 2020-06-30:
I will try upgrade the version, is there any version for windows (10 and server) that is recomended and stable enough for production?

cristipogacean, 2020-06-30:
We're close to release 2.7.x so preview is stable enough. the windows 10 is fully supported.

alonsors-809, 2020-07-01:
I switch to version 2.7.7, then 2.7.1 and 2.7.0 but the publisher from a list of 10 nodes, it only sends one. On the logs it shows that the publisher identifys correctly all the nodes but just send data from the last node. Also, from that same node it sends several message with the same timestamp. I am testing on a windows server.

cristipogacean, 2020-07-01:
sorry for misleading note  I meant :preview tag respectively 2.7.161 tag.

alonsors-809, 2020-07-02:
Hi, 
I switched to version 2.7.161 and the monitored nodes are behaving the same as version 2.5.1 of the publisher (it is good to have a more updated stable version) but the node is now sending the node repeated with the **same** timestamp. So, I am suspecting it is something with the node configuration and I have a couple of questions:
1.	How does the HeartBeat is correctly configured?
2.	Does the opc sampling, publishing and heartbeat intervals should have different times? If yes, then how much difference?
3.	Also, if you have more ideas to don’t get that repeated row of the same node and timestamp?

What I am trying to do is that every x minutes, the publisher sends data of the node even if the node hasn’t changed.

cristipogacean, 2020-07-02:
Hi,

I'm not sure what kind of values make sense for your use-case, so I will explain how this settings work and hopefully we can tune them together accordingly.
 
In OPC UA, the publishing mechanism is designed to save bandwidth so only the value changes are propagated from the server to the client(publisher in our case). On the monitored item creation the actual value is sent to the client, but afterwards just the  changes in value or status. The changes are are determined considering the sampling interval, so basically the server checks the data source periodically with the sampling interval period, checks if the value has changed since sample and if so it queues into the monitored item queue. Please note that if the source value changes and is reverted within the sample cycle, this change will not be sampled therefore not propagated to the client. The publishing interval refers to the data exchange between the client and the server and is the cycle time when the data queued by the server in the monitored item queues is actually transported to the client. Now the heartbeat interval is something specific to our publisher implementation and is meant to notify the cloud processor that communication is ok also in case of static or rarely changing monitored items. So if the publisher detects that no actual value change was received from the server it will resend the last known value change. the resending is done keeping the original timestamps as well since they are the server timestamps. having this in mind, for dynamic nodes the heartbeat should be configured as 10 or more factor the publishing interval. For static-ish nodes I would say 2 factor the sampling interval should be more relevant. The sampling and publishing setting depends on how fast the data is required. Using the same value would be a good practice for not super sensitive use-cases.

So in your case I would say something like x/2 sampling and publishing and x heartbeat, or if you are really keen on having values exactly at x minutes, publishing interval = heartbeat = x, but I would use x/2 for sampling so that we don't miss or get too delayed value-changes.

Hope this helps

alonsors-809, 2020-07-03:
Hi,
Thanks for the explanation, it was useful to clarify some doubts I had. I have tested in two environments (one publisher in windows and the other one in linux) with the version 2.7.161 of the publisher. For windows I notice that for some reason it starts sending the telemetry after a couple of hours. 

Then I notice that with this configuration:
        "OpcSamplingInterval":300000,
        "OpcPublishingInterval": 600000,
        "HeartbeatInterval": 600
The timestamp difference of the telemetry where 3 minutes, shouldn’t be 5 minutes?

Also, if in the configuration file I put a second node with different intervals from the first, it only sends one of the nodes and with repeated data, it is related?

cristipogacean, 2020-07-07:
Hi,

can you please detail a bit the issue on windows server 2019? this is a totally abnormal situation. Do you use the publisher in standalone mode (using publishednodes.json config file) or orchestrated mode? could it be that there are connectivity issues between the host and the IoTHub?

On the described configuration, the timestamp difference of 3 minutes is strange as well. I will try to replicate this.

Would you be able to share a sample of the configuration file where you experience the third issue?

alonsors-809, 2020-07-07:
Hi,

Do you use the publisher in standalone mode (using publishednodes.json config file) or orchestrated mode? 
A/ standalone, I use a publishednodes.json config file

could it be that there are connectivity issues between the host and the IoTHub?
A/ seems difficult because the module of the publisher and the server are on the same machine.

Would you be able to share a sample of the configuration file where you experience the third issue?
A/ Sure, in order to help you reproduce the steps, both nodes I am monitoring are generating exactly each minute a new value on the server. These two nodes are simulated.

Case 1 (two nodes configured with different interval):
"OpcNodes": [
      {
        "Id": "ns=02.;s=test0",
        "OpcSamplingInterval":30000,
        "OpcPublishingInterval": 60000,
        "HeartbeatInterval": 70
      },
      {
        "Id": "ns=02;s=test1",
        "OpcSamplingInterval":90000,
        "OpcPublishingInterval": 108000,
        "HeartbeatInterval": 108
      }
]
Result:

Message 1:
  {
      "TagName": "test0",
      "Timestamp": "2020-07-07T22:02:57.3514539Z",
      "TagValue": "5",
    },
  {
      "TagName": " test0",
      "Timestamp": "2020-07-07T22:02:57.3514539Z",
      "TagValue": "5",
    }
Message 2:
    {
      "TagName": " test0",
      "Timestamp": "2020-07-07T22:03:57.3516326Z",
      "TagValue": "9",
    },
    {
      "TagName": " test0",
      "Timestamp": "2020-07-07T22:03:57.3516326Z",
      "TagValue": "9",
    }

Example of message:
 
[image]

For a more agile and quicker communication, just let me know.

dacolgit, 2020-10-14:
Hi,

I tried to reproduce your issue, but I cannot see the double message.
Could you please try using this server: [https://github.com/Azure-Samples/iot-edge-opc-plc](url) ?

Just run it with the following command.
`docker run --rm -it -p 50000:50000 -p 8080:8080 --name opcplc mcr.microsoft.com/iotedge/opc-plc:latest --pn=50000 --autoaccept --nospikes --nodips --nopostrend --nonegtrend --nodatavalues --sph --sn=25 --sr=60 --st=uint --fn=5 --fr=1 --ft=uint`

The server like in your description generates a new value each minute.
If you need to change the port number change the --pn parameter.
In your publishednodes.json just use the followings node id: ns=2;s=SlowUInt1, ns=2;s=SlowUInt2.

You could also update to the latest publisher version.

Let me know if you still have issue.

dacolgit, 2020-10-19:
Reopen if you still have the issue.

closed 2020-10-19 by dacolgit