Azure/Industrial-IoT #592: UADP NetworkMessage sytntax

mpostol, 2020-07-01:
**Is your feature request related to a problem? Please describe.**

I need the definition of the Opc.Ua.PubSub.NetworkMessage syntax for the current implementation published on the master branch.

cristipogacean, 2020-07-02:
Hi Mariusz,
You can find the Pub/Sub message encoding logic in this file: https://github.com/Azure/Industrial-IoT/blob/master/components/opc-ua/src/Microsoft.Azure.IIoT.OpcUa.Protocol/src/Stack/PubSub/NetworkMessage.cs
We don't have the schema documented since this implementation is rather specific to our solution and I believe it is not completely spec compliant. We are working right now on making it compliant so I would kindly ask for your patience until it's done. I'll get back to you as soon as we have a compliant preview.

mpostol, 2020-07-02:
Hi @cristipogacean,

Thanks for your answer. I know where it is, but my concern is if there is any compliant implementation you know. My implementation also needs to be upgraded mpostol/OPC-UA-OOI#208 and it looks similar to my implementation compliant with a draft proposal of the specification. Now I am working on a road map and priorities of work on my side so any information that I can use for harmonization of our work is welcome.

cristipogacean, 2020-07-02:
I believe our friends at Softing and Unified Automation have compliant implementations in their SDKs and trial/demo apps. I only briefly tested Softings PubSub in the .net toolkit. The issue with UADP is that there are a lot of flags and knobs to setup the messages and some require state-full decoding. It's pretty hard to test and support everything.

mpostol, 2020-07-02:
I have collected the results of my investigation [here](https://github.com/mpostol/OPC-UA-OOI/issues/208#issuecomment-608511722), I suppose no one is compliant. 
The syntax defined by the spec is really very complex and the native language text-based. I have worked out a file that contains Augmented BNF definition](https://github.com/mpostol/OPC-UA-OOI/blob/master/Networking/SemanticData/MessageHandling/NetworkMessage.abnf) of the NetworkMessage.  I hope we could use it to automatically generate a parser.

cristipogacean, 2020-09-17:
closing for now, please reopen if needed

closed 2020-09-17 by cristipogacean