Azure/Industrial-IoT #593: Kubernetes version error when deploying to WestEurope

BeameX, 2020-07-02:
When trying the deploy the production ([kubernetes version](https://azure.github.io/Industrial-IoT/deploy/howto-deploy-aks.html)) to WestEurope region i get this error:

"statusMessage": "{\"code\":\"AgentPoolK8sVersionNotSupported\",\"message\":\"Version 1.15.7 is not supported in this region. Please use [az aks get-versions] command to get the supported version list in this region. For more information, please check https://aka.ms/supported-version-list\"}",

It turns out that this version is not supported in this region or as it seems any region anymore.

az aks get-versions --location westeurope --output table
KubernetesVersion    Upgrades
-------------------  -------------------------------------------------
1.18.2(preview)      None available
1.18.1(preview)      1.18.2(preview)
1.17.5(preview)      1.18.1(preview), 1.18.2(preview)
1.17.4(preview)      1.17.5(preview), 1.18.1(preview), 1.18.2(preview)
1.16.9               1.17.4(preview), 1.17.5(preview)
1.16.8               1.16.9, 1.17.4(preview), 1.17.5(preview)
1.15.11              1.16.8, 1.16.9
1.15.10              1.15.11, 1.16.8, 1.16.9
1.14.8               1.15.10, 1.15.11
1.14.7               1.14.8, 1.15.10, 1.15.11

or 

az aks get-versions --location eastus --output table
KubernetesVersion    Upgrades
-------------------  -------------------------------------------------
1.18.2(preview)      None available
1.18.1(preview)      1.18.2(preview)
1.17.5(preview)      1.18.1(preview), 1.18.2(preview)
1.17.4(preview)      1.17.5(preview), 1.18.1(preview), 1.18.2(preview)
1.16.9               1.17.4(preview), 1.17.5(preview)
1.16.8               1.16.9, 1.17.4(preview), 1.17.5(preview)
1.15.11              1.16.8, 1.16.9
1.15.10              1.15.11, 1.16.8, 1.16.9
1.14.8               1.15.10, 1.15.11
1.14.7               1.14.8, 1.15.10, 1.15.11

or 

az aks get-versions --location southeastasia --output table
KubernetesVersion    Upgrades
-------------------  -------------------------------------------------
1.18.2(preview)      None available
1.18.1(preview)      1.18.2(preview)
1.17.5(preview)      1.18.1(preview), 1.18.2(preview)
1.17.4(preview)      1.17.5(preview), 1.18.1(preview), 1.18.2(preview)
1.16.9               1.17.4(preview), 1.17.5(preview)
1.16.8               1.16.9, 1.17.4(preview), 1.17.5(preview)
1.15.11              1.16.8, 1.16.9
1.15.10              1.15.11, 1.16.8, 1.16.9
1.14.8               1.15.10, 1.15.11
1.14.7               1.14.8, 1.15.10, 1.15.11

So i think you need to modify the version used and as a feature request make some kind of logic that always deploys on the newest version.

karok2m, 2020-07-03:
Thank you for reporting this bug. I will upgrade Kubernetes version to `1.16.9` in #596.

karok2m, 2020-07-03:
@BeameX, code changes are merged into `master`. Can you please validate that it works fine for you ?

On a side note, you can find compiled binaries of `Microsoft.Azure.IIoT.Deployment` (`2.7.166` version) at the following links:
* [linux-x64](https://azureiiot.blob.core.windows.net/binaries/master/2.7.166/linux-x64/Microsoft.Azure.IIoT.Deployment)
* [osx-x64](https://azureiiot.blob.core.windows.net/binaries/master/2.7.166/osx-x64/Microsoft.Azure.IIoT.Deployment)
* [win-x64](https://azureiiot.blob.core.windows.net/binaries/master/2.7.166/win-x64/Microsoft.Azure.IIoT.Deployment.exe)

closed 2020-07-08 by karok2m

karok2m, 2020-07-15:
UPDATE: By [Release 2020-06-29](https://github.com/Azure/AKS/releases/tag/2020-06-29) of AKS Kubernetes versions <1.16.10 and <1.17.7 were removed due to severe bugs and security issues flagged upstream. Because of this `Microsoft.Azure.IIoT.Deployment` has been updated to use `1.16.10` version of Kubernetes.

Latest binaries, version `2.8.5`:

* [linux-x64](https://azureiiot.blob.core.windows.net/binaries/master/2.8.5/linux-x64/Microsoft.Azure.IIoT.Deployment)
* [osx-x64](https://azureiiot.blob.core.windows.net/binaries/master/2.8.5/osx-x64/Microsoft.Azure.IIoT.Deployment)
* [win-x64](https://azureiiot.blob.core.windows.net/binaries/master/2.8.5/win-x64/Microsoft.Azure.IIoT.Deployment.exe)