Azure/Industrial-IoT #603: Receive All Node values from OPC Server

mandar-opc, 2020-07-06:
I am using an OPC UA Publisher to read nodes from OPC Server. 

Sampling Interval is 300ms

I need to know is there any existing feature which will allow me to receive updated & not updated tags values.

cristipogacean, 2020-07-07:
The heartbeat mechanism of the publisher helps on publishing static or relatively static nodes. the heartbeat setting can be configured per node.

saurabhsadhu, 2020-07-07:
I am also trying to achieve this. Even if the node is generating data at a greater interval than sampling interval, I want to get the value of nodes at every sampling interval, even if they don't change. How should I configure heartbeat in pn.json to achieve this?

cristipogacean, 2020-07-07:
There are 2 options: 
1) add the heartbeat interval to the node defined in the config file, E.g. 
```
...
			{
				"Id": "ns=2;s=PositiveTrendData",
				"OpcSamplingInterval": 1000,
				"OpcPublishingInterval": 1000,
				"HeartbeatInterval": 30
			},			
			{
				"Id": "ns=2;s=NegativeTrendData",
				"OpcSamplingInterval": 1000,
				"OpcPublishingInterval": 1000,
				"HeartbeatInterval": 30
			}
...
```

2) use --hb=30 command line argument for the publisher to specify the default heartbeat interval. This will apply the heartbeat for all nodes that have no heartbeat configured.

Note: that the heartbeat interval is in seconds while the sampling and publishing intervals are in milliseconds

saurabhsadhu, 2020-07-07:
Yes this is the exact feature I want. But can I set this heartbeat interval in milliseconds?

cristipogacean, 2020-07-07:
1. is not possible
2. is possible using --DefaultHeartbeatInterval=00:00:00.100 as argument or  DefaultHeartbeatInterval=00:00:00.100 as environment variable

saurabhsadhu, 2020-07-07:
Setting the DefaultHeartbeatInterval as argument or env variable doesn't seem to work. Do I need to use a specific version of OPCPublisher

cristipogacean, 2020-07-07:
yes, :2.7.167 or :preview tag

saurabhsadhu, 2020-07-08:
It seems the latest tag is not able to connect to the OPC UA Server. I am getting the following warning and no messages. I am able to connect to the server using the `linux-amd64` tag. Am I missing something?

```
[05:38:46 WRN Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Failed to create session '5f5eb3672da238625099fe14cd1d152bc706536e' due to One or more errors occurred. (Unexpected error processing response.) 
```

cristipogacean, 2020-07-08:
I cannot manage to replicate this issue, our setup works fine with :latest tag (2.7.170). Could you share some details on the setup?

saurabhsadhu, 2020-07-09:
I am using OPCPublisher as an IoTEdge module and following are the container create options:
```
{
  "Hostname": "OPCPublisher",
  "Cmd": [
    "publisher",
    "--pf=/appdata/pn.json",
    "--aa",
    "--to",
    "--ih=Mqtt_Tcp_Only",
    "--si=0",
    "--ms=512",
    "--di=60",
    "--oi=300",
    "--mq=163840",
    "--DefaultHeartBeatInterval=00:00:00.400"
  ],
  "HostConfig": {
    "Binds": [
      "/home/user1/iiotedge:/appdata"
    ],
    "PortBindings": {
      "62222/tcp": [
        {
          "HostPort": "62222"
        }
      ]
    },
    "ExtraHosts": [
      "localhost:127.0.0.1"
    ]
  }
}
```

I can connect to OPCUA server using "linux-amd64" tag of the OPCPublisher, but for tag 2.7.170, it's failing to create session with OPC UA. server. Following are the logs: 
```
[11:04:50 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Azure IIoT 5f5eb3672da238625099fe14cd1d152bc706536e took 00:00:00.5085431. 
[11:04:50 WRN Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Failed to create session '5f5eb3672da238625099fe14cd1d152bc706536e' due to One or more errors occurred. (Unexpected error processing response.) 
[11:05:00 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Creating session '5f5eb3672da238625099fe14cd1d152bc706536e' for endpoint 'opc.tcp://<ip-address:port>'... 
[11:05:00 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Azure IIoT 5f5eb3672da238625099fe14cd1d152bc706536e took 00:00:00.5054768. 
[11:05:00 WRN Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Failed to create session '5f5eb3672da238625099fe14cd1d152bc706536e' due to One or more errors occurred. (Unexpected error processing response.) 
[11:05:10 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Creating session '5f5eb3672da238625099fe14cd1d152bc706536e' for endpoint 'opc.tcp://<ip-address:port>'... 
```

markusstuhler, 2020-09-25:
Hi @saurabhsadhu, we're currently working on a fix to be able to specify the Heartbeat Interval for each node in the pn.json file.

However, the cli argument "--DefaultHeartBeatInterval" should work. Could you please try to increase the interval (for testing purposes) to a higher value (like 5 seconds)?

Thanks!
Markus

markusstuhler, 2020-10-20:
I will close that issue as I didn't hear back from you. Please reopen if the issue persists.

Thanks!
Markus

closed 2020-10-20 by markusstuhler