Azure/Industrial-IoT #605: Polling mechanism instead of PubSub in OPCPublisher

saurabhsadhu, 2020-07-07:
**Problem**
I am trying to organize data coming through OPCpublisher according to timestamp. OPCPublisher works on PubSub model and hence it doesn't bring the data of nodes whose value have not been updated. For example if I have configured 100 nodes for OPCPublisher and all of them are sampled at 800ms interval. Now all nodes except 2 have changed their values at the next sampling interval. OPCPublisher doesn't get the data of those 2 nodes who haven't changed the values, because event wasn't detected.

I think polling mechanism instead of PubSub will resolve the issue. Is there any way to achieve this in OPCPublisher?

Thanks

cristipogacean, 2020-07-07:
Polling is not possible with the publisher. you can configure the heartbeat interval for the 2 static nodes.

cristipogacean, 2020-07-07:
I think that for your use-case the best option would be the using the key frame mechanism from the OPC UA PubSub. This feature is on roadmap.

FMauroy, 2020-07-24:
Hi all,

we also had the same problem, and ended up modifying the code of publisher 2.5.3 as a temporary measure. Maybe not that temporary as the newer version of publisher doesn't support message shaping anymore, and this is something we rely heavily on...

So, what I changed is in OpcSession.cs, after the creation of a new OpcUaMonitoredItem, I add a filter of type DataChangeFilter, with a trigger value of "StatusValueTimestamp". This way, publisher is also notified when the timestamp changes, and we have the events we expect. With error checks, it's just 5 lines to add. So far so good, data volume is expectedly more important, but we have other components that rely on a complete stream of events, and a heartbeat of 1 sec was not sufficient.

Team, do you see any side-effect of using this approach? Appart from the extra events that are generated of course.

HTH
fred

markusstuhler, 2020-09-22:
Hi @FMauroy, did you try to configure the heartbeat interval like @cristipogacean proposed in his first response? That should give you values from the static nodes even when their value didn't change.

I will close this issue for now. Please reopen if the issue cannot be solved with that approach.

Thanks!

FMauroy, 2020-09-22:
Of course not, the heartbeat is unreliable. In many cases, we had values in the future, that's why we had to implement it this way.

markusstuhler, 2020-09-24:
Could you please provide the code you wrote so that we can have a look at it to answer your question concerning side effects?

Thanks!
Markus

FMauroy, 2020-10-19:
Hi, here is the code snippet I added (sorry, didn't see your reply sooner):

`
    if (UseDataChangeTrigger)

        try

        {

           var filter = monitoredItem.MonitoredItem.Filter as DataChangeFilter ?? new DataChangeFilter();

            filter.Trigger = DataChangeTrigger.StatusValueTimestamp;

            monitoredItem.MonitoredItem.Filter = filter;

        }

        catch (Exception ex)

        {

            Logger.Warning(ex, "*** Filter hack error");

        }
`

closed 2023-06-11 by marcschier