Azure/Industrial-IoT #623: Cannot see TSI Environment

terrymandin, 2020-07-17:
**Describe the bug**
I have deployed the IIoT Platform, and have enabled some OPC UA endpoints to send data.  I see the data coming into the IoT Hub.  When I go to TSI I get a message that says "There are no environments available".

**To Reproduce**
Steps to reproduce the behavior:
1. Install the Industrial IoT Platform
2. Enable some OPC UA variables through the UI
3. Go to TSI in the portal
4. Click on "Go to TSI Explorer"
5. Get message "There are no environments available" (see attached)

**Expected behavior**
Ability to view OPC data in TSI

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser: chrome

cristipogacean, 2020-07-17:
I think you've missed assigning yourself data access rights. Please have a look here: https://github.com/Azure/Industrial-IoT/blob/master/docs/tutorials/tut-timeseriesinsights.md

terrymandin, 2020-08-12:
Thanks @cristipogacean !  I missed that in the documentation.

closed 2020-08-12 by terrymandin