Azure/Industrial-IoT #626: TSI: Mismatch of Time Series ID Case

nanowireUK, 2020-07-21:
**Describe the bug**
The Time Series ID Properties has a case mismatch with the data that is being sent from the publisher. This causes all data to appear under the 'null' entry in TSI.

In TSI the property is defined as publisherId, dataSetWriterId, nodeId

The data is arriving PublisherId, DataSetWriterId, NodeId

This is on a fresh install of 2.170 release. 

**To Reproduce**
Steps to reproduce the behavior:
1. Deploy the 2.170 Release
1. Publish data
1. Go to TSI to look at data

**Expected behavior**
The attributes are recognised as the property ids

**Screenshots**
[image]

**Additional context**
Add any other context about the problem here.

cristipogacean, 2020-07-21:
thank you for bring this up! for now the only solution is create a new TSI environment having the parameters you have correctly described above.

tomas-gottwald, 2020-07-28:
Further to this there seems to be another version in the docs: https://github.com/Azure/Industrial-IoT/blob/master/docs/tutorials/tut-timeseriesinsights.md where it says that a data point "is uniquely identified by the EndpointId, SubscriptionId and NodeId" but 'SubscriptionId' field is not available in the incoming JSON. Should it be "PublisherId" or "DataSetWriterId" instead? I am also confused why "EndpointId", "PublisherId" and "DataSetWriterId" have the same value like so:

`"ExtensionFields": {
		"EndpointId": "uatd8e83ad2edbbcf3f04cef0e01847e3d401cdd406",
		"PublisherId": "uatd8e83ad2edbbcf3f04cef0e01847e3d401cdd406",
		"DataSetWriterId": "uatd8e83ad2edbbcf3f04cef0e01847e3d401cdd406"
	}`

marcschier, 2021-02-26:
We have removed TSI integration and related services from master branch and thus 2.8.

closed 2021-02-26 by marcschier