Azure/Industrial-IoT #628: Cannot sign in into Engineering Tool after a successful deployment

vslepakov, 2020-07-22:
**Describe the bug**
IIoT platform is successfully deployed using deploy.ps1. When trying to sign in into Engineering Tool I see this error:
[image]

luiscantero, 2020-07-24:
Further information that I gathered:
* @vslepakov first generated the following aad.json file with this command: `./aad-register.ps1 -Name <AppName> -Output aad.json`
[image]
* After that he executed: `./deploy.ps1 -aadConfig aad.json`

luiscantero, 2020-07-28:
Please see our recommended solutions when you [don't have rights to grant admin consent](https://github.com/Azure/Industrial-IoT/blob/master/docs/deploy/howto-deploy-all-in-one.md#azure-active-directory-registration).

closed 2020-07-28 by luiscantero