Azure/Industrial-IoT #632: Admin access required on internal subscription

minettes, 2020-07-24:
Hi there, 
I deployed the industrial IOT platform onto my internal microsoft subscription. The deployment is successful, but when I launch the app it requests permissions to sign in and  read my profile. 
[image]
If I accept I get an error saying that I need admin approval. 
[image]

luiscantero, 2020-07-24:
You need to follow the process in your organization to grant admin consent to the app. Either that or use another AAD tenant for which you are an admin.

closed 2020-07-24 by luiscantero

luiscantero, 2020-07-28:
Please see our recommended solutions when you [don't have rights to grant admin consent](https://github.com/Azure/Industrial-IoT/blob/master/docs/deploy/howto-deploy-all-in-one.md#azure-active-directory-registration).