Azure/Industrial-IoT #647: OPC Publisher ignores "autoaccept" option

javafrog, 2020-07-29:
**Describe the bug**
We are trying to connect to an OPCUA server, but are facing error messages "Certificate chain validation incomplete.". We want to trust the server, so we are using the -aa option as described here: https://docs.microsoft.com/en-us/azure/iot-accelerators/howto-opc-publisher-run, so I would not expect the error.

Create secured session for endpoint URI 'opc.tcp://192.168.0.1:4840' with timeout of 50000 ms.","[09:21:22 ERR] Session creation to endpoint 'opc.tcp://192.168.0.1:4840' failed 81 time(s). Please verify if server is up and Publisher configuration is correct.","One or more errors occurred. (Certificate chain validation incomplete.\r\nSubjectName: CN=bdg.showcase.netz, OU=BU DIS, O=Corp, L=Dresden, S=Saxony, C=DE\r\nIssuerName: CN=festo-subca.showcase.netz, OU=BU DIS, O=Corp, L=Dresden, S=Saxony, C=DE)","   at System.Threading.Tasks.Task`1.GetResultCore(Boolean waitCompletionNotification)\n   at System.Threading.Tasks.Task`1.get_Result()\n   at OpcPublisher.OpcSession.ConnectSessionAsync(CancellationToken ct) in D:\\a\\1\\s\\modules\\src\\Microsoft.Azure.IIoT.Modules.OpcUa.Publisher\\src\\OpcSession.cs:line 444"

A connection to a different test server (that offers anonymous access) works fine with the same client.

**To Reproduce**
- use image `mcr.microsoft.com/iotedge/opc-publisher:latest`
- Create an IoT Edge deployment with the following settings:
`{
  "Hostname": "publisher",
  "Cmd": [
    "--pf=./publishednodes.json",
    "--aa"
  ],
  "HostConfig": {
    "Binds": [
      "/iiotedge:/appdata"
    ]
  }
}
`

**Expected behavior**
I would expect the client to skip the certificate validation completely and open the connection.

luiscantero, 2020-07-30:
@javafrog what version of the IIoT platform are you using?
Could you please try again with the latest one?

javafrog, 2020-07-30:
Hi @luiscantero thanks for the reply! We did try a few versions (2.5.3, 2.6.146 and 2.7.170), but the problem persists across all of them. At the very moment, we are using `latest`, as described above.

luiscantero, 2020-07-31:
@javafrog can you connect to the server with another client, e.g. UaExpert?
I will try to repro the issue today.

javafrog, 2020-07-31:
@luiscantero Yes, there is already another solution in place that is connected to the server.
We also tried writing a small python script ourselves, that works, too.

luiscantero, 2020-07-31:
@javafrog I've been unable to repro so far. The reason why I asked about the version is because the file OpcSession.cs from the error message belongs to an old version, the current one doesn't have this file. Please also try with the tag ":preview".

javafrog, 2020-07-31:
@luiscantero Thanks for the pointer, we will look into that!

javafrog, 2020-07-31:
Okay, with the :preview version, we run into another issue. Seemingly, the Publisher can't read the user/pw configuration anymore

```
[13:30:53 INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.LegacyJobOrchestrator] File /appdata/pn.json has changed, reloading...

[13:30:53 INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Models.PublishedNodesJobConverter] Read 1 items from published nodes file in 00:00:00.1944641

[13:30:53 WRN Microsoft.Azure.IIoT.Http.Default.HttpClient] POST to https://publisher/modules/OPCPublisher/genid/637317979264485651/decrypt?api-version=2019-11-05 returned InternalServerError (took 00:00:00.2746035).

[13:30:54 ERR Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Models.PublishedNodesJobConverter] failed to convert the published nodes.

System.AggregateException: One or more errors occurred. ({"message":"Could not decrypt\n\tcaused by: An error occurred in the key store.\n\tcaused by: HSM failure\n\tcaused by: HSM API failure occurred: 428"})

 ---> Microsoft.Azure.IIoT.Exceptions.ResourceInvalidStateException: {"message":"Could not decrypt\n\tcaused by: An error occurred in the key store.\n\tcaused by: HSM failure\n\tcaused by: HSM API failure occurred: 428"}

   at Microsoft.Azure.IIoT.Utils.Retry.DelayOrThrow(ILogger logger, Func`2 cont, Func`3 policy, Int32 maxRetry, Int32 k, Exception ex, CancellationToken ct) in D:\a\1\s\common\src\Microsoft.Azure.IIoT.Core\src\Utils\Retry.cs:line 723

   at Microsoft.Azure.IIoT.Utils.Retry.Do[T](ILogger logger, CancellationToken ct, Func`1 work, Func`2 cont, Func`3 policy, Int32 maxRetry) in D:\a\1\s\common\src\Microsoft.Azure.IIoT.Core\src\Utils\Retry.cs:line 115

   at Microsoft.Azure.IIoT.Module.Framework.Hosting.EdgeletClient.DecryptAsync(String initializationVector, Byte[] ciphertext, CancellationToken ct) in D:\a\1\s\common\src\Microsoft.Azure.IIoT.Hub.Module.Framework\src\Clients\EdgeletClient.cs:line 99

   at Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Models.PublishedNodesJobConverter.ToUserNamePasswordCredentialAsync(PublishedNodesEntryModel entry) in D:\a\1\s\components\opc-ua\src\Microsoft.Azure.IIoT.OpcUa.Edge.Publisher\src\Storage\PublishedNodesJobConverter.cs:line 258

   --- End of inner exception stack trace ---
```

Unfortunately, we haven't found any way to set it besides via the IoT Hub direct methods, which have been removed in the latest version. In the documentation, there is no other way mentioned.

luiscantero, 2020-07-31:
I will investigate further next week. Please also check the iotedge logs for HSM failures: `sudo journalctl -u iotedge`

luiscantero, 2020-08-10:
@javafrog where you able to get more logs about the HSM failures?

javafrog, 2020-08-12:
@luiscantero, sorry, but I was on vacation. I'll try and get those logs in the next days.

luiscantero, 2020-09-03:
Closing due to no response

closed 2020-09-03 by luiscantero