Azure/Industrial-IoT #648: Publisher Data Persistence

erlendp, 2020-07-30:
**Describe the bug**
Currently it seems that if the publisher module is restarted, any queued packets will be lost.

**To Reproduce**
Steps to reproduce the behavior:
1. Ensure the publisher module is running and set to publish one or more nodes.
2. Disable connectivity between the gateway and IoT Hub.
3. Wait for several packets to accumulate in the queue.
4. Restart the publisher module - i.e. `iotedge restart publisher`.
5. Inspecting the raw packets in TSI reveals that these packets were lost.

**Expected behavior**
Any queued packets are persisted to disk, and picked up and sent once connectivity is restored.

**Versions**
 - OS: Windows Server 1809
- Modules: 2.7.170

erlendp, 2020-08-02:
@luiscantero Is it safe to assume this is intended behavior? We were hoping that setting up mapped docker volumes was for the purpose of preserving these buffers, but given the added feature request label, it looks like that's not the case?

luiscantero, 2020-08-03:
@erlendp The mapped volume is used for the publishfile and/or certificates, currently not to persist the queue in case of lost connectivity. I have created a feature request in our system.

closed 2020-08-03 by luiscantero