Azure/Industrial-IoT #653: Unable to execute aad-register.ps1 using Azure Cloud Shell

gregeva, 2020-08-04:
**Describe the bug**
I am using Azure Cloud Shell as a manner of having the most stable basis for PowerShell and Azure CLI components as this comes loaded with both (according to the Azure CLI documentation).

Attempting to run the deployment scripts fails stating that you should be sure to be running the shell as Administrator, and to ensure that the AzureAD CmdLet is installed.  I have been unable to find anything around runas/elevate in Azure Cloud Shell as exists in Windows, and AzureAD CmdLet is installed yet has no effect in resolution.

The aad-register.ps1 script first logs in with my user account  (despite already being logged in with Connect-AzAccount), and then attempts to run, failing with the following error.

> An error occurred: The expression after '&' in a pipeline element produced an object that was not valid.  It must result in a command name, a script block, or a CommandInfo object.

**To Reproduce**
Steps to reproduce the behavior:
1. Open an Azure Cloud Shell from the Azure Portal
2. Clone the Azure Industrial IoT repository 'git clone https://github.com/Azure/Industrial-IoT'
3. Change to the deployment scripts directory 'cd Industrial-IoT/deploy/scripts'
4. Execute the Azure AD application registration script './aad-register.ps1 -Name application_name -Output aad.json'

**Expected behavior**
I'd guess that this should be obvious, but I expect to be able to run the deployment scripts to get the Azure Industrial IoT components provisioned and configured without encountering blocking errors such as this.

**Screenshots**
[image]

[image]

[image]

luiscantero, 2020-08-05:
@gregeva Could you please try the version from this [branch](https://github.com/Azure/Industrial-IoT/tree/lcantero/deploycloudshell). It could be that you will be prompted to login twice. There is an issue in the cloud shell related to connect-azuread that leads to the script not working and having to login twice.

Otherwise, using deploy.ps1 to deploy from the cloud shell is not really recommended as it takes over 30 min to complete, and the cloud shell will end the session after 20 min - so the last step of registering the reply_url might fail.

luiscantero, 2020-08-06:
PR #660 adds support for the Cloud Shell

closed 2020-08-06 by luiscantero

gregeva, 2020-08-25:
Hello @luiscantero 

I tried using the Cloud Shell as so many people had issues across various environments with variance (versions, OSes, etc.)  The intention was to have a stable fixed approach.  I understand that this is not ideal due to timeout but was testing.

I think that the underlying issue is with the  './aad-register.ps1 -Name application_name -Output aad.json' command which I was never able to get working on Windows either.  It seems that this is related to the wrong context being used (default account context versus actual logged in context).  Shall I create a different issue for this?

luiscantero, 2020-09-03:
@gregeva did you try the latest version and is still not working?
I also added a parameter TenantId that you can use to select a particular AAD tenant. Please try this as well and let me know the results.

gregeva, 2020-10-14:
Hi @luiscantero  - sorry but I've moved on and won't be going back to troubleshoot this.