Azure/Industrial-IoT #669: OPC Publisher Module - Issue with multiple endpoints in pn.json

oliverchristen, 2020-08-12:
**Describe the bug**
When I configure multiple endpoints in the OPC Publisher configuration file `pn.json` the behaviour is weird and random seemingly. Sometimes, only the first subscription job is executed but twice. And sometimes both subscription jobs are executed correctly.
When the first behaviour occurs, sometimes it helps to switch the order of the endpoint configurations in `pn.json` and restart the module. Sometimes it also helps to remove the second endpoint configuration, restart the module, add it again and restart the module again.

**To Reproduce**
Steps to reproduce the behavior:
1. Configure multiple endpoints in `pn.json`:
```
[
  {
    "EndpointUrl": "opc.tcp://<OPC-UA-Server1>:49580",
    "OpcNodes": [
      {
        "Id": "ns=2;s=NodeId1",
        "OpcPublishingInterval": 20000,
        "OpcSamplingInterval": 20000,
        "DisplayName": "NodeId1"
      }
    ]
  },
  {
    "EndpointUrl": "opc.tcp://<OPC-UA-Server2>:49580",
    "OpcNodes": [
      {
        "Id": "ns=2;s=NodeId1",
        "OpcPublishingInterval": 5000,
        "OpcSamplingInterval": 5000,
        "DisplayName": "NodeId1"
      }
    ]
  }
]
```
2. See logs `iotedge logs OPCPublisher`
3. Sometimes the first endpoint configuration is executed twice:
```
[09:19:09 INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.DataFlowProcessingEngine] 
  DIAGNOSTICS INFORMATION for          : opc.tcp://<OPC-UA-Server1>:49580_4f305e5cae5a136d38ea46deae0d53b2bde57121
  # Ingestion duration                 :    00:17:52:32 (dd:hh:mm:ss)
  # Ingress DataChanges (from OPC)     :              1 (0/s)
  # Ingress ValueChanges (from OPC)    :              2 (0/s)
  # Ingress BatchBlock buffer size     :              0
  # Encoding Block input/output size   :              0 | 0
  # Encoder Notifications processed    :              2
  # Encoder Notifications dropped      :              0
  # Encoder IoT Messages processed     :              1
  # Encoder avg Notifications/Message  :              2
  # Encoder avg IoT Message body size  :         10,521 (4%)
  # Encoder avg IoT Chunk (4 KB) usage :            2.6
  # Estimated IoT Chunks (4 KB) per day:              4
  # Outgress Batch Block buffer size   :              0
  # Outgress input buffer count        :              0
  # Outgress IoT message count         :              1 (0/s)
  # Connection retries                 :              0
  
[09:19:09 INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.DataFlowProcessingEngine] 
  DIAGNOSTICS INFORMATION for          : opc.tcp://<OPC-UA-Server1>:49580_4f305e5cae5a136d38ea46deae0d53b2bde57121
  # Ingestion duration                 :    00:17:52:32 (dd:hh:mm:ss)
  # Ingress DataChanges (from OPC)     :              1 (0/s)
  # Ingress ValueChanges (from OPC)    :              2 (0/s)
  # Ingress BatchBlock buffer size     :              0
  # Encoding Block input/output size   :              0 | 0
  # Encoder Notifications processed    :              2
  # Encoder Notifications dropped      :              0
  # Encoder IoT Messages processed     :              1
  # Encoder avg Notifications/Message  :              2
  # Encoder avg IoT Message body size  :         10,521 (4%)
  # Encoder avg IoT Chunk (4 KB) usage :            2.6
  # Estimated IoT Chunks (4 KB) per day:              4
  # Outgress Batch Block buffer size   :              0
  # Outgress input buffer count        :              0
  # Outgress IoT message count         :              1 (0/s)
  # Connection retries                 :              0
```

And sometimes both endpoint configurations are executed correctly:
```
[09:19:09 INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.DataFlowProcessingEngine] 
  DIAGNOSTICS INFORMATION for          : opc.tcp://<OPC-UA-Server1>:49580_4f305e5cae5a136d38ea46deae0d53b2bde57121
  # Ingestion duration                 :    00:17:52:32 (dd:hh:mm:ss)
  # Ingress DataChanges (from OPC)     :              1 (0/s)
  # Ingress ValueChanges (from OPC)    :              2 (0/s)
  # Ingress BatchBlock buffer size     :              0
  # Encoding Block input/output size   :              0 | 0
  # Encoder Notifications processed    :              2
  # Encoder Notifications dropped      :              0
  # Encoder IoT Messages processed     :              1
  # Encoder avg Notifications/Message  :              2
  # Encoder avg IoT Message body size  :         10,521 (4%)
  # Encoder avg IoT Chunk (4 KB) usage :            2.6
  # Estimated IoT Chunks (4 KB) per day:              4
  # Outgress Batch Block buffer size   :              0
  # Outgress input buffer count        :              0
  # Outgress IoT message count         :              1 (0/s)
  # Connection retries                 :              0
 
[09:19:09 INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.DataFlowProcessingEngine] 
  DIAGNOSTICS INFORMATION for          : opc.tcp://<OPC-UA-Server2>:49580_33921d4bbb3237f118d4da4eaa0e2d39717e7a9f
  # Ingestion duration                 :    00:17:52:17 (dd:hh:mm:ss)
  # Ingress DataChanges (from OPC)     :          3,217 (0.05/s)
  # Ingress ValueChanges (from OPC)    :         44,995 (0.7/s)
  # Ingress BatchBlock buffer size     :              0
  # Encoding Block input/output size   :              0 | 0
  # Encoder Notifications processed    :         44,995
  # Encoder Notifications dropped      :              0
  # Encoder IoT Messages processed     :          3,216
  # Encoder avg Notifications/Message  :             14
  # Encoder avg IoT Message body size  :          3,454 (1%)
  # Encoder avg IoT Chunk (4 KB) usage :            0.8
  # Estimated IoT Chunks (4 KB) per day:          4,319
  # Outgress Batch Block buffer size   :              0
  # Outgress input buffer count        :              0
  # Outgress IoT message count         :          3,216 (0.05/s)
  # Connection retries                 :              0
```

**Expected behavior**
Both subscription jobs are executed according endpoint configurations.

**Versions**
 - OS: Ubuntu Linux 18.4
 - Edge Runtime: 1.0.9.3
 - OPC Publisher Module: 2.7.170

oliverchristen, 2020-08-20:
I'm still struggling with the configuration of multiple endpoints. I found out that when I repeatedly restart the module without changing the pn.json file the OPCPublisher only works as expected in 1 of about 5 times. There seems to be a randomness in this part of the code. I tried to analyze the source code but didn't find anything so far.

erlendp, 2020-08-24:
I believe this issue is also happening when there are multiple dataset writers, which can happen if different publishing intervals have been specified.

Thanks for creating a PR - hopefully it gets up-streamed quickly!

luiscantero, 2020-09-03:
I have tested the publisher hundreds of times with multiple servers and have honestly never ran into this issue. Thanks for the PR.

karok2m, 2020-09-08:
As mentioned in the discussion under #679 PR, the issue does not seem to be resolved.

oliverchristen, 2020-09-09:
The behaviour of the publisher is the same as before: Randomly the first endpoint configuration is executed twice. In my case in ~4 out of 5 times after a restart of the module. No matter how I configure the various settings. Our module configuration looks like this:
```json
"OPCPublisher": {
            "settings": {
              "image": "mcr.microsoft.com/iotedge/opc-publisher:2.8.42",
              "createOptions": {
                "Hostname": "publisher",
                "Cmd": [
                  "publisher",
                  "--pf=/appdata/pn.json",
                  "--di=20",
                  "--to",
                  "--aa",
                  "--si=20",
                  "--ms=0",
                  "--oi=20000",
                  "--op=20000"
                ],
                "HostConfig": {
                  "PortBindings": {
                    "62222/tcp": [
                      {
                        "HostPort": "62222"
                      }
                    ]
                  },
                  "Binds": [
                    "x509certstores:/root/.dotnet/corefx/cryptography/x509stores",
                    "/etc/opcpublisher:/appdata"
                  ],
                  "ExtraHosts": [
                    "localhost:127.0.0.1"
                  ]
                }
              }
            },
            "type": "docker",
            "version": "1.0",
            "status": "running",
            "restartPolicy": "always"
          }
```

luiscantero, 2020-09-09:
@oliverchristen just to be clear: your local repo that includes the changes in your PR #679 does fix the issue, but opc-publisher:2.8.42 does not?

oliverchristen, 2020-09-09:
@luiscantero no that's not true. I did not manage to create a module with my local repo. So I was not sure whether my pull request fixes the issue or not.

luiscantero, 2020-09-09:
@oliverchristen This is how you can build your own module and push to your own ACR repo (from Windows):
cd tools\scripts
powershell -command "& '.\docker-source.ps1' ..\modules\src\Microsoft.Azure.IIoT.Modules.OpcUa.Publisher\src"
az acr login --name <YOUR_ACR_NAME>
docker build -f ..\..\modules\src\Microsoft.Azure.IIoT.Modules.OpcUa.Publisher\src\bin\publish\Release\linux-x64\Dockerfile.linux-amd64 -t <YOUR_ACR_NAME>.azurecr.io/iotedge/opc-publisher:latest ..\..\modules\src\Microsoft.Azure.IIoT.Modules.OpcUa.Publisher\src\bin\publish\Release\linux-x64\
docker push <YOUR_ACR_NAME>.azurecr.io/iotedge/opc-publisher:latest

Then on the IoT Hub Page -> Iot Edge just enter your ACR's URL, username/password and image.

oliverchristen, 2020-09-09:
@luiscantero thanks for this hint.

commit dfa45b5f (2020-09-03): multiple endpoints (#679)

* Pass cancellation token to Task.Run() as well in order to avoid wrapper.Processing.IsCompleted is false

* Fix misspellings

* Use a concurrent dictionary because there can be multiple threads which access the dictionary

Fixes #669
PR #716 (2020-09-16): Fix multiple endpoints concurrency issue
No I managed to build the OPC Publisher edge module locally with the help of @luiscantero. I also managed to solve the issue #669 with that locally built module. So this PR now really fixes #669. Hope to get it merged soon so @erlendp can profit from it as well.
closed 2020-09-16 by karok2m