Azure/Industrial-IoT #678: Autmomatic Deployment of a specific Release fails

logg-sar, 2020-08-19:
**Describe the bug**
I have to deploy a predifined version on devices with some limitations and I have two issues with this.
One of the limitations is that git isn´t installed and not installable.

After reading [https://github.com/Azure/Industrial-IoT/issues/456](url) I´ve cloned the repository on a developer pc and want to use this.

However - after calling deploy.ps1 it stops because git isn´t installed.
**Outcommenting Line 1152 in deploy.ps1 (Select-RepositoryAndBranch) helped**, but it IMHO it shouldnt be so.

The second issue - there are a lot of parameters wich are documented. With this given parameters I can run this script with one command-line. Except the .env-file.
During the deployment I get the Question "_Save environment as $ENVVARS for local development? [y/n]_" - This is not scriptable. At least I couldnt find any parameter for this

**To Reproduce**
Steps to reproduce the behavior:
1a. Git clone to a directory
2a. Transfer to another pc
3a. run deploy.ps1 (with parameter)
4a. error "This script requires *git* to be installed..." occures - even when its not really needed

Steps to reproduce the behavior:
1b. Run deploy.ps1 with the correct parameter
2b. get a prompt for saving environment - which is a manual entry

**Expected behavior**
Script should run accept all paramater and need no additional manual entries

luiscantero, 2020-09-07:
@logg-sar I will take a look at this in the next few days, sorry for the long delay.

luiscantero, 2020-09-08:
@logg-sar Reg. 2b, when you pass a context for authentication, that sets the interactive mode to false and it will not ask to save the environment, see: $script:interactive = !$script:context
Does this help?

luiscantero, 2020-09-08:
Reg. 4a, the script needs the branch name because it uses ARM templates that are located in the online repo. It defaults to the currently selected branch, which should also exist online. Passing -branchName master to the script could bypass the need for git. Would that help?

luiscantero, 2020-09-11:
Closing due to no response.

closed 2020-09-11 by luiscantero