Azure/Industrial-IoT #683: Helm installation instructions install old non-working version (2.16.3)

gregeva, 2020-08-25:
Following the [instructions to add AKS to the Azure Industrial IoT microservices deployment](https://nam04.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2FAzure%2FIndustrial-IoT%2Fblob%2Fmaster%2Fdocs%2Fdeploy%2Fhowto-add-aks-to-ps1.md&data=02%7C01%7Cgeva%40ptc.com%7Cb5ad0b67e9cc405b373708d848d6eac4%7Cb9921086ff774d0d828acb3381f678e2%7C0%7C0%7C637339435181977438&sdata=82l57nQe2dhUY1qNlGgjLfvsTLqlQgR1VOP16DFFKvg%3D&reserved=0), you are directed to install kubectl and Helm.  The Helm installation command 'az acr helm install' installs the wrong version of Helm and causes the following installation steps to fail.

Our experience has shown that a [manual installation of Helm](https://helm.sh/docs/intro/install/) is required in order to get the latest version which works with the Helm Charts provided for this AKS deployment.

karok2m, 2020-09-02:
Thank for reporting this @gregeva. I will update the doc to link to official Helm documentation for installation instructions. I will also keep Azure CLI based installation as an alternative approach, adding a client version to it. The updated command will be:

```bash
 az acr helm install-cli --client-version "3.2.0"
```

Regarding the deprecation of `az acr helm` command group: there is an ongoing discussion about this, and there is a possibility that the whole command group will be moved under `az aks`. You can find more about this [here](https://github.com/Azure/AKS/issues/1331).

closed 2020-09-02 by karok2m

gregeva, 2020-10-14:
You're welcome @karok2m - thanks for improving the documentation to address this.  It was a brick wall for a lot of our people trying to use AIIOT.