Azure/Industrial-IoT #685: Duplicate entries for dataSetWriter in job on CosmosDB

martin-weber, 2020-08-26:
## Description of the Bug

When unregistering an asset and running a new discovery, the and register the same endpoint again, it results in a second dataSetWriter in the job in Cosmos DB with the same dataSetWriterId, when one of the alternative `connection.endpoint.alternativeUrls` change between the discoveries.

## Steps to reproduce

I used the free Pro-SYS OPC-UA Simulation Server to reproduce this issue. These are the steps:

* Start OPC-UA Simulation server. And disable some of the binded addresses in the Tab Endpoints -> Bind Addresses -> Custom
*  In the Engineering Tool run a discovery and turn on one of the asset/endpoint
* Turn on publishing of an endpoint item
* Wait for the job to run
* check cosmos DB entry for the Job entry. It contains now one dataSetWriter for the given endpoint. In my example with the following dataSetWriterId:
  ```json
  "dataSetWriterId": "uat442d2f329e1ba4e37320a299ae18564d8885982d",
  ```
  and the following connection:
  ```json
                                "connection": {
                                    "endpoint": {
                                        "url": "opc.tcp://192.168.1.146:53530/OPCUA/marweber",
                                        "alternativeUrls": [
                                            "opc.tcp://172.23.112.1:53530/OPCUA/marweber",
                                            "opc.tcp://host.docker.internal:53530/OPCUA/marweber"
                                        ],
                                        "securityMode": "None",
                                        "securityPolicy": "http://opcfoundation.org/UA/SecurityPolicy#None",
                                        "certificate": "XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX"
                                    }
                                },
  ```
* Turn off discover
* Unregister the asset 
* **Questions:**
  * The CosmosDB entry for the job still remains. **Is this expected?**
  * If running the module form visual studio, the debug output still seams to read from the server, as it prints messages like this:  `NotificationReceived: ClientHandle=175, Value=-86.66667` **Is this expected, or should it stop?**

* In the OPC-UA Simulation Server in the Tab Endpoints -> Bind Addresses -> Custom activate one of the previously deactivated addresses. Then apply and restart the OPC-UA server
* In the engineering portal start a new discovery.
* Add the same endpoint again.
* Check the CosmosDB entry. A new data set writer with the same dataSetWriterId is created
  ```json
   "dataSetWriterId": "uat442d2f329e1ba4e37320a299ae18564d8885982d" 
   ```
   but with the new connection, even though the connection.endpoitn.url is still the same
  ```json
                                "connection": {
                                    "endpoint": {
                                        "url": "opc.tcp://192.168.1.146:53530/OPCUA/marweber",
                                        "alternativeUrls": [
                                            "opc.tcp://172.23.112.1:53530/OPCUA/marweber",
                                            "opc.tcp://host.docker.internal:53530/OPCUA/marweber",
                                            "opc.tcp://192.168.64.1:53530/OPCUA/marweber",
                                            "opc.tcp://100.X.X.X:53530/OPCUA/marweber"
                                        ],
                                        "securityMode": "None",
                                        "securityPolicy": "http://opcfoundation.org/UA/SecurityPolicy#None",
                                        "certificate": "XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX"
                                    }
                                },
  ```
* When turning on the asset again and navigation to the corresponding endpoint control in the engineering portal, the item is already / still turned on and shows updating data.

**Questions**:
  * The Visual studio output  prints still the `NotificationReceived: ClientHandle=175, Value=-86.66667` messages even when no endpoint is turned on since the discovery of the endpoint. Shouldn't this stop when the asset is unregistered?
 

## Final Job entry in CosmosDB

This is the final entry in CosmosDB after the 
[... 164 lines cut ...]
            "_etag": null,
            "ClassType": "ProcessingStatus",
            "JobId": "uat442d2f329e1ba4e37320a299ae18564d8885982d",
            "LastKnownHeartbeat": "2020-08-25T15:32:34.5510992Z",
            "LastKnownState": null,
            "ProcessMode": "Active"
        },
        "marweber-dev_publisher_2": {
            "id": null,
            "_etag": null,
            "ClassType": "ProcessingStatus",
            "JobId": "uat442d2f329e1ba4e37320a299ae18564d8885982d",
            "LastKnownHeartbeat": "2020-08-26T07:29:38.0799652Z",
            "LastKnownState": null,
            "ProcessMode": "Active"
        }
    },
    "Updated": "2020-08-26T07:29:38.0799707Z",
    "Created": "2020-08-25T08:23:20.4575204Z",
    "_rid": "dag+AIS5XlsyAAAAAAAAAA==",
    "_self": "dbs/dag+AA==/colls/dag+AIS5Xls=/docs/dag+AIS5XlsyAAAAAAAAAA==/",
    "_attachments": "attachments/",
    "_ts": 1598426978
}
```

marcschier, 2021-03-20:
#774

karok2m, 2021-04-12:
Issue of duplicate `DataSetWriterId`s in a job definition has been fixed by #689. The fix is present in `2.7.199` and `2.7.206` versions of the platform.

closed 2021-07-14 by marcschier