Azure/Industrial-IoT #699: OPC publisher - Could not get logs for module publisher

aagamb12, 2020-09-07:
Dear all, 
I have a problem with OPC publisher to sending logs into IoTHub. I have succesfully deployed OPC publisher into my raspberry PI (already configured as IoTEdge device) and everything seems to be properly configured in Azure OPC module container. When i run the command "iotedge list" you can see that the 3 services are running well but when i want to see the publisher OPC logs i do not get and en error is shown.  See error shown below (could not get logs for module OPC publisher)

[image]

Here below is my my OPC container config file in Azure. 
[image]

my pn.json file located in my raspberry as shown below.
[image]

i know there are OPC tags coming in becuase i am running an UA OPC client in my raspeberry which is succesfully collecting OPC tags from another OPC server (located on the same VLAN) and i can see tag coming into the OPC UA client running constantly. see image below. 

[image]

Any ideas how to troubleshoot this error? 
Thanks

luiscantero, 2020-09-07:
@aagamb12 please try `sudo iotedge logs OPCPublisher -f`. The syntax is `iotedge logs <CONTAINER_NAME>`.

aagamb12, 2020-09-07:
Luis,Tthat was exactly the issue! i had another issue with the container as well but now everything seems to be working. Thank you so much,
[image]

Do you know how i can see the json OPC telemetry being sent into my IoThub? the direct messages (GetInfo, etc ) does not work. I know the OPC trafffic is coming in successfully. see below 

[image]

luiscantero, 2020-09-07:
@aagamb12 I'm glad I was able to help.
BTW, you can also use docker commands directly, such as:
sudo docker ps
sudo docker logs OPCPublisher --tail 200
etc.

closed 2020-09-07 by luiscantero

luiscantero, 2020-09-08:
@aagamb12 I just noticed your last question. To view the telemetry, you can use "Azure IoT Explorer" or the Azure IoT Tools extension for VS Code.