Azure/Industrial-IoT #702: The value of deployment parameter 'keyVaultPrincipalId' is null

nspyke, 2020-09-08:
**Describe the bug**
While running the minimal deployment with the following command;
```
$context = Get-AzContext
./deploy.ps1 -type minimum -version latest -resourceGroupName rg-iiot-prd -subscriptionId xxxx -tenantId xxxx -applicationName xxx-xxx -context $context
```
 i get this output & error;
```
Created new resource group rg-iiot-prd in australiaeast.
Deployment will use 'master' branch in 'https://github.com/Azure/Industrial-IoT'.
Start time: 09:47

Registering client and services AAD applications in your tenant...
Importing module AzureAD
Using passed context (Account xxxx, Tenant 0000-0000xxxxx)
Client and services AAD applications registered...

Starting deployment...
09:48:04 - Error: Code=InvalidDeploymentParameterValue; Message=The value of deployment parameter 'keyVaultPrincipalId' is null. Please specify the value or use the parameter reference. See https://aka.ms/resource-manager-parameter-files for details.

Deployment failed.
Removing resource group rg-iiot-prd...
True
New-AzResourceGroupDeployment: C:\xxxxx\Industrial-IoT\deploy\scripts\deploy.ps1:925
Line |
 925 |  … eployment = New-AzResourceGroupDeployment -ResourceGroupName $resourc …
     |                ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
     | 09:48:04 - Error: Code=InvalidDeploymentParameterValue; Message=The value of deployment parameter 'keyVaultPrincipalId' is null. Please specify the value or use the parameter reference. See
     | https://aka.ms/resource-manager-parameter-files for details.

```

I am not a Global Admin, but I am owner on the subscription. But it looks like it was able to create the AAD apps just fine?

**To Reproduce**
Steps to reproduce the behavior:
Run the command above

**Expected behavior**
The deployment to work as explained in readme

**Desktop (please complete the following information):**
 - OS: Windows
 - Version 10

nspyke, 2020-09-08:
Tried again with the Cloud App Administrator role in AAD and it failed again with the same error

nspyke, 2020-09-08:
After being granted Cloud App Admin role in AAD, I tried again and received the same error.
I tried to run the following;
```
./deploy/scripts/aad-register.ps1 -Name data-platform-prd -ReplyUrl https://ci-data-platform-prd.azurewebsites.net/ -tenantId xxxxxxxx -Output aad-prd.json
```
and all that I get in the output file aad-prd.json is `null`

nspyke, 2020-09-09:
Tried to run the deploy script again on a AAD tenant where I am Global Administrator and it still fails with the same error.

luiscantero, 2020-09-09:
@nspyke could you please try deploying and/or registering directly from the Azure Cloud Shell?

nspyke, 2020-09-09:
I was able to get this to work by dropping all the additional parameters apart from `-type minimum` and then switching from the master branch to the latest release branch. This is a work around though. I will try using the Cloud Shell when I get to work tomorrow.

luiscantero, 2020-09-09:
I just deployed using master and all your minimum parameters and it was successful:
[image]

luiscantero, 2020-09-09:
@nspyke Are you able to create/modify app registrations manually in the AAD tenants that you are using?

nspyke, 2020-09-09:
> 
> 
> @nspyke could you please try deploying and/or registering directly from the Azure Cloud Shell?

deploy.ps1 works on the master branch, in Azure Cloud Shell, only using the parameter -type minimum.

> 
> 
> @nspyke Are you able to create/modify app registrations manually in the AAD tenants that you are using?

Yes, I have the Cloud Application Administrator role.

I was having this issue while Powershell 7, and I'm thinking that its possible that that may have something to do with it.
Now that I've reverted back to Powershell 5, I'm not encountering the issue.

luiscantero, 2020-09-11:
Closing for now. We can reopen if necessary.

closed 2020-09-11 by luiscantero

vjrantal, 2020-10-01:
As reference that I had this same issue but it got fixed by deleting existing powershell module installations and letting the deploy script install the dependencies.

I wonder could it be related to how the module dependencies are checked in file `aad-register.ps1` and whether there would be any scenario where it detects a module installation but the version is old or incompatible with the deployment?

luiscantero, 2020-10-02:
@vjrantal One important module used by `aad-register.ps1` is AzureAD, you can check if it's installed like this:
`Get-Module -ListAvailable -Name "AzureAD"`

The script also includes two fallbacks for running it in the Azure Cloud Shell. I've noticed that if you start a cmd command prompt from VS2019, then start `powershell` and run the above command, the module might not be found. If you start a PS directly (blue window), the module does get found. So the module path may also affect the execution: `$env:psmodulepath -split ";"`

Here are other useful commands:
```
Get-Module -ListAvailable -Name "AzureAD"
Install-Module -Name "AzureAD"
Update-Module -Name "AzureAD"
```