Azure/Industrial-IoT #703: Running the Microsoft.Azure.IIoT.App sample Blazor app

apittier, 2020-09-09:
Hi, 

I'd like to try out the Industrial-IoT\samples\src\Microsoft.Azure.IIoT.App Blazor app. 

When attempting to debug it from Visual Studio it fails with the error below. 

Are there any guidelines on how to get the needed environment setup?
Is there any demo version of this app already deployed to take a look at it?

```
Microsoft.Azure.IIoT.Exceptions.InvalidConfigurationException
  HResult=0x80131500
  Message=Storage configuration is missing in your configuration for dataprotection to store all keys across all instances.
  Source=Microsoft.Azure.IIoT.AspNetCore
  StackTrace:
   at Microsoft.Azure.IIoT.AspNetCore.Auth.DataProtectionBuilderEx.AddAzureBlobKeyStorage(IDataProtectionBuilder builder, IConfiguration configuration) in C:\work\Industrial-IoT\common\src\Microsoft.Azure.IIoT.AspNetCore\src\Auth\Extensions\DataProtectionBuilderEx.cs:line 86
   at Microsoft.Azure.IIoT.AspNetCore.Auth.DataProtectionBuilderEx.AddAzureDataProtection(IServiceCollection services, IConfiguration configuration) in C:\work\Industrial-IoT\common\src\Microsoft.Azure.IIoT.AspNetCore\src\Auth\Extensions\DataProtectionBuilderEx.cs:line 43
   at Microsoft.Azure.IIoT.App.Startup.ConfigureServices(IServiceCollection services) in C:\work\Industrial-IoT\samples\src\Microsoft.Azure.IIoT.App\src\Startup.cs:line 152
   at System.RuntimeMethodHandle.InvokeMethod(Object target, Object[] arguments, Signature sig, Boolean constructor, Boolean wrapExceptions)
   at System.Reflection.RuntimeMethodInfo.Invoke(Object obj, BindingFlags invokeAttr, Binder binder, Object[] parameters, CultureInfo culture)
   at Microsoft.AspNetCore.Hosting.ConfigureServicesBuilder.InvokeCore(Object instance, IServiceCollection services)
   at Microsoft.AspNetCore.Hosting.ConfigureServicesBuilder.<>c__DisplayClass9_0.<Invoke>g__Startup|0(IServiceCollection serviceCollection)
   at Microsoft.AspNetCore.Hosting.ConfigureServicesBuilder.Invoke(Object instance, IServiceCollection services)
   at Microsoft.AspNetCore.Hosting.ConfigureServicesBuilder.<>c__DisplayClass8_0.<Build>b__0(IServiceCollection services)
   at Microsoft.AspNetCore.Hosting.GenericWebHostBuilder.UseStartup(Type startupType, HostBuilderContext context, IServiceCollection services)
   at Microsoft.AspNetCore.Hosting.GenericWebHostBuilder.<>c__DisplayClass12_0.<UseStartup>b__0(HostBuilderContext context, IServiceCollection services)
   at Microsoft.Extensions.Hosting.HostBuilder.CreateServiceProvider()
   at Microsoft.Extensions.Hosting.HostBuilder.Build()
   at Microsoft.Azure.IIoT.App.Program.Main(String[] args) in C:\work\Industrial-IoT\samples\src\Microsoft.Azure.IIoT.App\src\Program.cs:line 36

  This exception was originally thrown at this call stack:
    Microsoft.Azure.IIoT.AspNetCore.Auth.DataProtectionBuilderEx.AddAzureBlobKeyStorage(Microsoft.AspNetCore.DataProtection.IDataProtectionBuilder, Microsoft.Extensions.Configuration.IConfiguration) in DataProtectionBuilderEx.cs
    Microsoft.Azure.IIoT.AspNetCore.Auth.DataProtectionBuilderEx.AddAzureDataProtection(Microsoft.Extensions.DependencyInjection.IServiceCollection, Microsoft.Extensions.Configuration.IConfiguration) in DataProtectionBuilderEx.cs
    Microsoft.Azure.IIoT.App.Startup.ConfigureServices(Microsoft.Extensions.DependencyInjection.IServiceCollection) in Startup.cs
    [External Code]
    Microsoft.Azure.IIoT.App.Program.Main(string[]) in Program.cs	
```

Thanks.
Amparo

luiscantero, 2020-09-09:
@apittier You need to [deploy the platform](https://github.com/Azure/Industrial-IoT/blob/master/docs/deploy/howto-deploy-all-in-one.md#running-the-script) to the cloud. When deployment finishes, you will be asked to save the environment. This will save a .env file to the root of your repository and will enable you to debug the application locally.
I hope this helps.

apittier, 2020-09-09:
@luiscantero many thanks for the quick response!

closed 2020-09-10 by luiscantero