Azure/Industrial-IoT #707: Deployment Unsuccessful without RBAC's Key Vault

NooraKubati, 2020-09-10:
**Describe the bug**
Unable to deploy the platform. This is due an issue with the deployment of the KeyVault, for security purposes I require it to have RBAC (Role Based Access Control). Without this the deployment will not run in my organisation. 
However my team and I are struggling to find the configuration options for this in the cloned repo. Appreciate this may be a simple fix / pointer in the right direction but would be gladly appreciated. 

Script ran in powershell: 
`./deploy.ps1 -aadConfig aad.json -branchName poc/pubsub_binary -type app`

**To Reproduce**
Steps to reproduce the behavior:
1. Try to deploy on a managed subscription that only allows RBAC key vaults

**Expected behavior**
The deployment to be successful. 

**Error Message** 
`"properties": {
        "statusCode": "BadRequest",
        "serviceRequestId": null,
        "statusMessage": "{\"error\":{\"code\":\"InvalidTemplateDeployment\",\"message\":\"The template deployment failed because of policy violation. Please see details for more information.\",\"details\":[{\"code\":\"AppendPoliciesUndefinedFields\",\"message\":\"Found policies that refer to an undefined field property for api-version '2016-10-01'. 
		Fields: 'Microsoft.Keyvault/vaults/accessPolicies[*],Microsoft.Keyvault/vaults/accessPolicies[*],Microsoft.Keyvault/vaults/accessPolicies[*],Microsoft.Keyvault/vaults/accessPolicies[*],Microsoft.Keyvault/vaults/accessPolicies[*],Microsoft.Keyvault/vaults/accessPolicies[*],Microsoft.Keyvault/vaults/accessPolicies[*],Microsoft.Keyvault/vaults/accessPolicies[*]'.
		Policy identifiers: '[{\\\"policyAssignment\\\":{\\\"name\\\":\\\"Policy Initiative to enforce RBAC in KeyVault to meet IRM requirements\\\"`

**Additional context**
Add any other context about the problem here.

luiscantero, 2020-09-11:
@NooraKubati I updated the KV apiVersion in the file azuredeploy.minimum.json. Could you please clone and deploy using branch `lcantero/updatekvapi`, then post any errors that you might get here.

cristipogacean, 2020-09-11:
Thanks @luiscantero can you please do the same on the poc/pubsub_binary branch?

NooraKubati, 2020-09-11:
@luiscantero Hi Luis thanks for your response. I have tried deploying from that branch and hit the same error again: 
`Deployment will use 'lcantero/updatekvapi' branch in 'https://github.com/Azure/Industrial-IoT'.

Please specify a name for your application.
>: #####
Start time: 11:21
Using preview images from mcr.microsoft.com.
Starting deployment...
11:21:53 - Error: Code=InvalidTemplateDeployment; Message=The template deployment failed because of policy violation. Please see details for more information.`

and further detailed in the activity log: 

`"properties": {
        "statusCode": "BadRequest",
        "serviceRequestId": null,
        "statusMessage": "{\"error\":{\"code\":\"InvalidTemplateDeployment\",\"message\":\"The template deployment failed because of policy violation. Please see details for more information.\",\"details\":[{\"code\":\"AppendPoliciesUndefinedFields\",\"message\":\"Found policies that refer to an undefined field property for api-version '2019-09-01'. Fields: 'Microsoft.Keyvault/vaults/accessPolicies[*],Microsoft.Keyvault/vaults/accessPolicies[*],Microsoft.Keyvault/vaults/accessPolicies[*],Microsoft.Keyvault/vaults/accessPolicies[*],Microsoft.Keyvault/vaults/accessPolicies[*],Microsoft.Keyvault/vaults/accessPolicies[*],Microsoft.Keyvault/vaults/accessPolicies[*],Microsoft.Keyvault/vaults/accessPolicies[*]'. Policy identifiers: '[{\\\"policyAssignment\\\":{\\\"name\\\":\\\"Policy Initiative to enforce RBAC in KeyVault to meet IRM requirements\\\"`

I am running this deployment under a managed subscription but after also creating a new tenant and AAD global admin, as it is a POC. It is possible this set up is also contributing to the failures.

luiscantero, 2020-09-14:
@NooraKubati We are investigating and will get back ASAP.

luiscantero, 2020-09-15:
@NooraKubati Please pull to update your `lcantero/updatekvapi` repo, then try to deploy again.
If it succeeds, you will need to perform the following manual steps to complete the deployment:
1. Go to the resource group, find the resource of type Managed Identity and copy its name
2. Select the Key vault, click on Access control (IAM) -> Add -> Add role assignment
Role: Key Vault Administrator
Select: <Managed Identity Name>
Save
3. Select the App service (for the Web app, not the services) -> Overview -> Restart
Allow up to ~5 min for the web app to be ready and refresh the page a few times

NooraKubati, 2020-09-15:
@luiscantero Hi Luis, I have tried your suggestions from scratch however the deployment is still hitting the same error unfortunately. 
Please review my attempts below: 

```
PS C:\WINDOWS\system32> cd "C:\Users\Noora.Kubati"
PS C:\Users\Noora.Kubati> git clone --single-branch --branch lcantero/updatekvapi  https://github.com/Azure/Industrial-IoT
Cloning into 'Industrial-IoT'...
remote: Enumerating objects: 21, done.
remote: Counting objects: 100% (21/21), done.
remote: Compressing objects: 100% (16/16), done.
remote: Total 42824 (delta 7), reused 13 (delta 5), pack-reused 42803
Receiving objects: 100% (42824/42824), 41.19 MiB | 7.88 MiB/s, done.
Resolving deltas: 100% (29102/29102), done.
Updating files: 100% (3102/3102), done.
PS C:\Users\Noora.Kubati\\Industrial-IoT> git branch
PS C:\Users\Noora.Kubati\Industrial-IoT> cd "C:\Users\Noora.Kubati\Industrial-IoT\deploy\scripts"
PS C:\Users\Noora.Kubati\Industrial-IoT\deploy\scripts> git branch
* lcantero/updatekvapi
PS C:\Users\Noora.Kubati\Industrial-IoT\deploy\scripts> ./aad-register.ps1 -TenantId YYYYYYYYYYYYYYYYYYYY -Output aad.json -Name doorsiot1009 -ReplyUrl https://doorsiot1009.azurewebsites.net/
Importing module AzureAD
Save credentials in .user file? [y/n]: y

Overwrite File?
Overwrite existing file at 'C:\Users\Noora.Kubati\Industrial-IoT\.user'?
[Y] Yes  [N] No  [S] Suspend  [?] Help (default is "Y"): y
Using context (Account Noora.Kubati@XXX.com, Tenant XXXXXXXXXXXXXXXXXXXXXXX)
Selected Tenant 'DoorsIOT.onmicrosoft.com' as authority.
'doorsiot1009-service' updated with required resource access, app roles and known applications.
User Noora.Kubati_shell.com#EXT#@DoorsIOT.onmicrosoft.com already has all app roles assigned.
Admin consent granted to native client application.
'doorsiot1009-client' updated with required resource access.
Registering https://doorsiot1009.azurewebsites.net/ as reply URL ...
Admin consent granted to web application.
'doorsiot1009-web' updated with required resource access.
PS C:\Users\Noora.Kubati\Industrial-IoT\deploy\scripts> ./deploy.ps1 -aadConfig aad.json -branchName lcantero/updatekvapi -type app
Signing in ...

Please choose a subscription from this list (using its index):

Index Subscription                                    Id
----- ------------                                    --
    1 XXX YYYYYYYYYYYYYYYYYYYYYYYYYYYYYY
    2 XXX YYYYYYYYYYYYYYYYYYYYYYYYYYYYYY
    3 XXX YYYYYYYYYYYYYYYYYYYYYYYYYYYYYY

>: 3
Azure subscription XXX (YYYYYYYYYYYYYYYYYYYYYYYYYYYYYY) selected.

Please provide a name for the resource group.
>: XXX-IOT
Using existing resource group XXX-IOT...
Deployment will use 'lcantero/updatekvapi' branch in 'https://github.com/Azure/Industrial-IoT'.

Please specify a name for your application.
>: doorsiot1009
Start time: 14:39
Using preview images from mcr.microsoft.com.
Starting deployment...
14:39:53 - Error: Code=InvalidTemplateDeployment; Message=The template deployment failed because of policy violation. Please see details for more information.

Deployment failed.
Try again? [y/n]: n
New-AzResourceGroupDeployment : 14:39:53 - Error: Code=InvalidTemplateDeployment; Message=The template deployment failed because of policy violation. Please see details for more information.
At C:\Users\Noora.Kubati\Industrial-IoT\deploy\scripts\deploy.ps1:925 char:27
+ ... eployment = New-AzResourceGroupDeployment -ResourceGroupName $resourc ...
+                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : NotSpecified: (:) [New-AzResourceGroupDeployment], Exception
    + FullyQualifiedErrorId : Microsoft.Azure.Commands.ResourceManager.Cmdlets.Implementation.NewAzureResourceGroupDeploymentCmdlet
```
 
 
 
 
 in the portal: 
 
` "statusMessage": "{\"error\":{\"code\":\"InvalidTemplateDeployment\",\"message\":\"The template deployment failed because of policy violation. Please see details for more information.\",\"details\":[{\"code\":\"AppendPoliciesUndefinedFields\",\"message\":\"Found policies that refer to an undefined field property for api-version '2016-10-01'. Fields: 'Microsoft.Keyvault/vaults/accessPolicies[*],Microsoft.Keyvault/vaults/accessPolicies[*],Microsoft.Keyvault/vaults/accessPolicies[*],Microsoft.Keyvault/vaults/accessPolicies[*],Microsoft.Keyvault/vaults/accessPolicies[*],Microsoft.Keyvault/vaults/accessPolicies[*],Microsoft.Keyvault/vaults/accessPolicies[*],Microsoft.Keyvault/vaults/accessPolicies[*]'. Policy identifiers: '[{\\\"policyAssignment\\\":{\\\"name\\\":\\\"Policy Initiative to enforce RBAC in KeyVault to meet IRM requirements\\\",\\\"id\\\":\\\"<censored>"},\\\"policyDefinition\\\":{\\\"name\\\":\\\"Enforce RBAC in KeyVault to meet IRM requirements\\\",\\\"id\\\":\\\"/<censored>"},\\\"policySetDefinition\\\":{\\\"name\\\":\\\"Apply RBAC policies to Key vault\\\"`

Internally we have removed all the chunks of code that refer to the key vault in the files: azuredeploy.minimum.json, azuredeploy.json and deploy.ps1 
This has led to all the components being created successfully and the only failure point is the resource "infrastructure.minimum"
Please see below: 
[image]

The error within this is the same as we have been experiencing (Enforce RBAC in KeyVault to meet IRM requirements). 

For our use case we may be able to make this work as all the components are there bar the secrets for the vault so we will attempt to manually add these in. 

Hope this helps your work.

luiscantero, 2020-09-16:
@NooraKubati Could you please contact your IT and ask about more details regarding the policy? (the error is not standard, so I cannot check myself).
In the repo that we are using, we set the enableRbacAuthorization=true, which results in a Key vault with this setting:
[image]

luiscantero, 2020-09-16:
@NooraKubati I just updated the template again and removed all access policies, which are not needed when using RBAC. Could you please try to deploy again? Thanks.

NooraKubati, 2020-09-16:
@luiscantero Hi Luis, unfortunately this did not succeed either. The key vault was made correctly with the required RBAC access, however this has not gotten rid of the error and there were no secrets/keys to be found. 

Have just had a long call with some of your colleagues and I think this issue will take a lot of investigating so I am to happy to continue working on that together privately and maybe this issue can be closed for now. We can update later on with a solution if there is one to be found.

luiscantero, 2020-09-16:
We will continue offline

closed 2020-09-16 by luiscantero