Azure/Industrial-IoT #710: OPCPublisher startup delays

mscott012, 2020-09-13:
**Describe the bug**
Later versions of OPCpublisher seeing large delay in subscription items.
Comparison done between V107 and V183 versions

**To Reproduce**
Steps to reproduce the behavior:
1. Install V107 on opcpublisher module to iotedge
2. Start Opcpublisher and watch logs
3. Repeat with V183 and compare logs looking at times between subscripting each 1000 items

**Expected behavior**
Expect that later versions would have same performance or better than previous versions

**logs**

[image]

Between “Added 1000 monitored items to subscription” messages in 183, we are seeing a delay of around 15-20 seconds.
[23:18:52 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] Added 1000 monitored items to subscription opc.tcp://10.10.53.99:49320_9cf2a794cb9f29ae854266c5f35eb149576cc904 
[23:18:53 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] Now monitoring 1000 nodes in subscription opc.tcp://10.10.53.99:49320_9cf2a794cb9f29ae854266c5f35eb149576cc904 
[23:19:07 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] Added 1000 monitored items to subscription opc.tcp://10.10.53.99:49320_092c30715110b994cdfc006b6c59f89f78835187 
[23:19:07 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] Now monitoring 1000 nodes in subscription opc.tcp://10.10.53.99:49320_092c30715110b994cdfc006b6c59f89f78835187

In 107 we are seeing a delay of  ~ 1 second.
[23:11:47 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] Added 1000 monitored items in subscription 48
[23:11:47 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] Now monitoring 1000 nodes in subscription 48
[23:11:47 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] Added 1000 monitored items in subscription 49
[23:11:48 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] Now monitoring 1000 nodes in subscription 49

[opcpublisher-startup-107.log](https://github.com/Azure/Industrial-IoT/files/5215016/opcpublisher-startup-107.log)
[opcpublisher-startup-183.log](https://github.com/Azure/Industrial-IoT/files/5215017/opcpublisher-startup-183.log)

**IOTedge Details:**
 - OS: Ubuntu 18.04.4 LTS (GNU/Linux 4.15.0-112-generic x86_64)
 - iotedge 1.0.9.4

root@nzshstiotedg01m:~# iotedge list
NAME                STATUS           DESCRIPTION      CONFIG
edgeAgent           running          Up 5 days        mcr.microsoft.com/azureiotedge-agent:1.0.9.4
edgeHub             running          Up 5 days        mcr.microsoft.com/azureiotedge-hub:1.0.9.4
opcpublisher        running          Up 3 days        mcr.microsoft.com/iotedge/opc-publisher:2.7.183

barnstee, 2020-09-25:
@mscott012 , I tired to repro this with 2.7.170 and 2.7.183 and couldn't, not in standalone mode or in orchestrated mode. Here are my results (tested with Kepware):

Orchestrated mode, 2.7.170:
[10:32:40 INF Root] Starting module OpcPublisher version 2.7.170.10628.
[10:32:40 INF Root] Initiating prometheus at port 9702/metrics
[10:32:40 INF Root] Prometheus metric server started.
[10:32:40 WRN Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Bypassing certificate validation for client.
[10:32:40 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Running in iotedge context.
[10:32:40 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] 0: Module iotedge-27192_publisher reconnected due to Connection_Ok.
[10:32:41 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Initialize device twin for iotedge-27192 - publisher
[10:32:41 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Applying initial desired state.
[10:32:41 INF Microsoft.Azure.IIoT.Modules.OpcUa.Publisher.Controller.ConfigurationSettingsController] Agent configuration updated.
[10:32:41 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Reporting currently initial state.
[10:32:41 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Module Host started.
[10:32:41 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.WorkerSupervisor] Creating new worker...
[10:32:41 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.WorkerSupervisor] Creating new worker...
[10:32:41 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.WorkerSupervisor] Creating new worker...
[10:32:41 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.WorkerSupervisor] Creating new worker...
[10:32:41 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.WorkerSupervisor] Creating new worker...
[10:32:41 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.WorkerSupervisor] Starting worker 'iotedge-27192_publisher_0'...
[10:32:41 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker] Worker iotedge-27192_publisher_0: {"Type": "publisher", "SiteId": "iotedge-27192", "DeviceId": "iotedge-27192", "ModuleId": "publisher"}
[10:32:41 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.WorkerSupervisor] Starting worker 'iotedge-27192_publisher_1'...
[10:32:41 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker] Worker iotedge-27192_publisher_1: {"Type": "publisher", "SiteId": "iotedge-27192", "DeviceId": "iotedge-27192", "ModuleId": "publisher"}
[10:32:41 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.WorkerSupervisor] Starting worker 'iotedge-27192_publisher_2'...
[10:32:41 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker] Worker iotedge-27192_publisher_2: {"Type": "publisher", "SiteId": "iotedge-27192", "DeviceId": "iotedge-27192", "ModuleId": "publisher"}
[10:32:41 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.WorkerSupervisor] Starting worker 'iotedge-27192_publisher_3'...
[10:32:41 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker] Worker iotedge-27192_publisher_3: {"Type": "publisher", "SiteId": "iotedge-27192", "DeviceId": "iotedge-27192", "ModuleId": "publisher"}
[10:32:41 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.WorkerSupervisor] Starting worker 'iotedge-27192_publisher_4'...
[10:32:41 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker] Worker iotedge-27192_publisher_4: {"Type": "publisher", "SiteId": "iotedge-27192", "DeviceId": "iotedge-27192", "ModuleId": "publisher"}
[10:32:44 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker] Worker: iotedge-27192_publisher_0 processing job: uataa8aebaa7379c902b9a727e1141fdecf3354068f, mode: Active
[10:32:44 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Connecting all clients to iotedge-27192 using MqttOverTcp.
[10:32:44 WRN Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Bypassing certificate validation for client.
[10:32:44 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] 0: Device DataSetWriterV2_uataa8aebaa7379c902b9a727e1141fdecf3354068f reconnected due to Connection_Ok.
[10:32:44 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Subscription '54366053-4fcc-492d-a809-ba2047bde800' registered/updated in session 'c18c6d5a63b7f7a80d5391d348d4d71a81a05b71' in state Init
[10:32:44 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Subscription '54366053-4fcc-492d-a809-ba2047bde800' registered/updated in session 'c18c6d5a63b7f7a80d5391d348d4d71a81a05b71' in state Init
[10:32:44 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Subscription 'e183c9c0-0e7a-4fca-a2fb-ad8ac70d5d92' registered/updated in session 'c18c6d5a63b7f7a80d5391d348d4d71a81a05b71' in state Init
[10:32:44 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Subscription 'e183c9c0-0e7a-4fca-a2fb-ad8ac70d5d92' registered/updated in session 'c18c6d5a63b7f7a80d5391d348d4d71a81a05b71' in state Init
[10:32:44 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Subscription 'd0d192d9-3a85-414f-8d2a-b21e2dcebefe' registered/updated in session 'c18c6d5a63b7f7a80d5391d348d4d71a81a05b71' in state Init
[10:32:44 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Subscription 'd0d192d9-3a85-414f-8d2a-b21e2dcebefe' registered/updated in session 'c18c6d5a63b7f7a80d5391d348d4d71a81a05b71' in state Init
[... cut ...]
2020-09-24 13:23:40.559 +00:00 [INF] Set Monitoring to "Reporting" for 1000 nodes in subscription opc.tcp://172.16.6.4:49320_34e9f86ef746ede8f6007170142faec85f5afeb6
2020-09-24 13:23:40.572 +00:00 [INF] Set Monitoring to "Reporting" for 1000 nodes in subscription opc.tcp://172.16.6.4:49320_ab5e328afacc4c49a2da52524338e74442339265
2020-09-24 13:23:40.582 +00:00 [INF] Set Monitoring to "Reporting" for 1000 nodes in subscription opc.tcp://172.16.6.4:49320_2036ee377daddd7da401060c860be1c04539f805
=> 8 seconds startup until publishing

jmbrunskill, 2020-09-28:
Could this be due to having invalid tags in our publishednodes.json?
Maybe the error handling is causing a wait state?

We have errors like this in our logs
[23:20:24 WRN Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] Failed to read display name for 'ns=2;s=Sim.CH1.NZ010_SIM_CH1_TAG13972_JMB1468\31c4323-c63-e0a01' due to 'BadNodeIdUnknown' 
[23:20:24 WRN Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] Failed to read display name for 'ns=2;s=Test2NZ010_test_jimmy_pV\2b5659d-c63-fda02' due to 'BadNodeIdUnknown' 

The subscriptions delay seems to start before we see any of these BadNodeIdUnknown errors though.

(This we are running in standalone mode, with a publishenodes.json configuration file)

barnstee, 2020-10-08:
I don't think the BadNodeIdUnknown has an impact. I just tried again with Kepware and 100K published nodes and still can't repro, but I did try our latest release (2.7.199) compared with your original test (2.7.107):

2020-10-08 10:39:10.994 +00:00 [INF] Starting module OpcPublisher version 2.7.199.64639.
2020-10-08 10:39:10.996 +00:00 [INF] Initiating prometheus at port 9702/metrics
2020-10-08 10:39:11.017 +00:00 [INF] Prometheus metric server started.
2020-10-08 10:39:11.021 +00:00 [WRN] Bypassing certificate validation for client.
2020-10-08 10:39:11.026 +00:00 [INF] Running in iotedge context.
2020-10-08 10:39:11.523 +00:00 [INF] 0: Module erichb2_OpcPublisher reconnected due to "Connection_Ok".
2020-10-08 10:39:11.634 +00:00 [INF] Initialize device twin for erichb2 - OpcPublisher
2020-10-08 10:39:11.704 +00:00 [INF] Applying initial desired state.
2020-10-08 10:39:11.709 +00:00 [INF] Reporting currently initial state.
2020-10-08 10:39:11.762 +00:00 [INF] Module Host started.
2020-10-08 10:39:11.766 +00:00 [INF] Creating new worker...
2020-10-08 10:39:12.582 +00:00 [INF] File /mount/published_nodes.json has changed, reloading...
2020-10-08 10:39:14.061 +00:00 [INF] Read 1 items from published nodes file in "00:00:01.4753124"
=> 3s startup

2020-10-08 10:42:59.795 +00:00 [INF] Converted items to jobs in "00:03:45.7316059"
=> 4 min to parse the published nodes file

2020-10-08 10:43:01.590 +00:00 [INF] Starting worker 'StandalonePublisher_0'...
2020-10-08 10:43:01.594 +00:00 [INF] Worker StandalonePublisher_0: {}
2020-10-08 10:43:01.606 +00:00 [INF] Worker: StandalonePublisher_0 processing job: Standalone_erichb2_OpcPublisher, mode: "Active"
2020-10-08 10:43:02.904 +00:00 [INF] Subscription 'opc.tcp://172.16.6.4:49320_a704a85dd691b0df30263db40c6b85aeda57654e' registered/updated in session 'e23512f2f7cb22d59e422b94c470ba97c4d01d53' in state "Init"
2020-10-08 10:43:02.912 +00:00 [INF] Subscription 'opc.tcp://172.16.6.4:49320_a704a85dd691b0df30263db40c6b85aeda57654e' registered/updated in session 'e23512f2f7cb22d59e422b94c470ba97c4d01d53' in state "Init"
2020-10-08 10:43:02.913 +00:00 [INF] Subscription 'opc.tcp://172.16.6.4:49320_5695f5305033786c87c06f63846b142c7710f2e4' registered/updated in session 'e23512f2f7cb22d59e422b94c470ba97c4d01d53' in state "Init"
2020-10-08 10:43:02.914 +00:00 [INF] Subscription 'opc.tcp://172.16.6.4:49320_5695f5305033786c87c06f63846b142c7710f2e4' registered/updated in session 'e23512f2f7cb22d59e422b94c470ba97c4d01d53' in state "Init"
2020-10-08 10:43:02.915 +00:00 [INF] Subscription 'opc.tcp://172.16.6.4:49320_36ab12a6a1a09e6cac89bace3f27ed862725ad36' registered/updated in session 'e23512f2f7cb22d59e422b94c470ba97c4d01d53' in state "Init"
2020-10-08 10:43:02.916 +00:00 [INF] Subscription 'opc.tcp://172.16.6.4:49320_36ab12a6a1a09e6cac89bace3f27ed862725ad36' registered/updated in session 'e23512f2f7cb22d59e422b94c470ba97c4d01d53' in state "Init"
2020-10-08 10:43:02.917 +00:00 [INF] Subscription 'opc.tcp://172.16.6.4:49320_a2e3be4d48c2c534d489d56caf58a82ad8e66750' registered/updated in session 'e23512f2f7cb22d59e422b94c470ba97c4d01d53' in state "Init"
2020-10-08 10:43:02.918 +00:00 [INF] Subscription 'opc.tcp://172.16.6.4:49320_a2e3be4d48c2c534d489d56caf58a82ad8e66750' registered/updated in session 'e23512f2f7cb22d59e422b94c470ba97c4d01d53' in state "Init"
2020-10-08 10:43:02.919 +00:00 [INF] Subscription 'opc.tcp://172.16.6.4:49320_d6a3419e051d064369c60f1918fb971c29837fe2' registered/updated in session 'e23512f2f7cb22d59e422b94c470ba97c4d01d53' in state "Init"
2020-10-08 10:43:02.919 +00:00 [INF] Subscription 'opc.tcp://172.16.6.4:49320_d6a3419e051d064369c60f1918fb971c29837fe2' registered/updated in session 'e23512f2f7cb22d59e422b94c470ba97c4d01d53' in state "Init"
2020-10-08 10:43:02.920 +00:00 [INF] Subscription 'opc.tcp://172.16.6.4:49320_b27aa86c65e3c01ac6d4fffcf4f5249008c5fc44' registered/updated in session 'e23512f2f7cb22d59e422b94c470ba97c4d01d53' in state "Init"
2020-10-08 10:43:02.922 +00:00 [INF] Subscription 'opc.tcp://172.16.6.4:49320_b27aa86c65e3c01ac6d4fffcf4f5249008c5fc44' registered/updated in session 'e23512f2f7cb22d59e422b94c470ba97c4d01d53' in state "Init"
2020-10-08 10:43:02.923 +00:00 [INF] Subscription 'opc.tcp://172.16.6.4:49320_b1d9b19fab874e87d6109286fa45d70a88affd99' registered/updated in session 'e23512f2f7cb22d59e422b94c470ba97c4d01d53' in state "Init"
2020-10-08 10:43:02.924 +00:00 [INF] Subscription 'opc.tcp://172.16.6.4:49320_b1d9b19fab874e87d6109286fa45d70a88affd99' registered/updated in session 'e23512f2f7cb22d59e422b94c470ba97c4d01d53' in state "Init"
2020-10-08 10:43:02.925 +00:00 [INF] Subscription 'opc.tcp://172.16.6.4:49320_e2a3173ab5a03fc5ab69cd4037bff62fc619af08' registered/updated in session 'e23512f2f7cb22d59e422b94c470ba97c4d01d53' in state "Init"
2020-10-08 10:43:02.926 +00:00 [INF] Subscription 'opc.tcp://172.16.6.4:49320_e2a3173ab5a03fc5ab69cd4037bff62fc619af08' registered/updated in session 'e23512f2f7cb22d59e422b94c470ba97c4d01d53' in state "Init"
2020-10-08 10:43:02.927 +00:00 [INF] Subscription 'opc.tcp://172.16.6.4:49320_c16bc19374b08b4e3e2780cb7c1c1f12f1237c54' registered/updated in session 'e23512f2f7cb22d59e422b94c470ba97c4d01d53' in state "Init"
2020-10-08 10:43:02.928 +00:00 [INF] Subscription 'opc.tcp://172.16.6.4:49320_c16bc19374b08b4e3e2780cb7c1c1f12f1237c54' registered/updated in session 'e23512f2f7cb22d59e422b94c470ba97c4d01d53' in state "Init"
[... cut ...]
2020-10-08 11:06:13.897 +00:00 [INF] Added 1000 monitored items in subscription 198
2020-10-08 11:06:13.960 +00:00 [INF] Now monitoring 1000 nodes in subscription 198
2020-10-08 11:06:13.965 +00:00 [INF] Added 1000 monitored items in subscription 199
2020-10-08 11:06:14.032 +00:00 [INF] Now monitoring 1000 nodes in subscription 199
2020-10-08 11:06:14.038 +00:00 [INF] Added 1000 monitored items in subscription 200
2020-10-08 11:06:14.102 +00:00 [INF] Now monitoring 1000 nodes in subscription 200
=> 11s to create the subscriptions

barnstee, 2020-10-08:
Can you try to repo again on your side with 2.7.199, please? Do you see this regression consistently?

barnstee, 2020-10-19:
Closing this issue due to inactivity. If you still see this issue, please reopen with additional details (full debug log).

barnstee, 2020-11-18:
Reopening. James, thanks for the detailed logs. From these, I can see that at first, we are in-line with what you expect, i.e. we are adding about 500 monitored items per second, but after about 5 seconds, the time it takes to add 1000 notes is increased by 2 seconds every time. This is due to an in-memory list getting longer and longer. The good news is that we are currently planning OPC Publisher version 3.0 and having a linear/predictable performance is one of its requirements.
So please stay on version 107 for now if this is a blocking issue for you and then try out version 3.0 when we release it early next year. Sound good?

jmbrunskill, 2020-11-19:
Thanks @barnstee,
We look forward to trying out version 3.0

jmbrunskill, 2021-02-02:
@barnstee any update on when this is likely to be resolved? I see you are using a new repo now for this project?

dacolgit, 2021-04-20:
Hi @jmbrunskill,  I started working on this issue, could you provide the publishednodes.json file you are using?
Thanks, Daniele

jmbrunskill, 2021-04-21:
Thanks @dacolgit,

Sorry I ran out of time for this today but will get you something tomorrow NZ time.

James

jmbrunskill, 2021-04-22:
Hi @dacolgit

In the attached zip is a JSON file for configuring a Thingworx Kepware server with approx 20,000 tags.
I've also provided a publishednodes.json file for the same configuration.

The thingworx trial version can be downloaded from this website - https://www.ptc.com/en/products/thingworx/thingworx-kepware-server

In my testing just I'm seeing this performance difference between versions and this configuration.

```
[06:54:51 INF Root] Starting module OpcPublisher version 2.7.206.28401.
[06:59:12 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] Set Monitoring to Reporting for 1000 nodes in subscription opc.tcp://10.10.53.99:49320_b2fd0a6b6cc703611884ed0047cab80cd9354c68 

~4 minutes

[07:09:42 INF Root] Starting module OpcPublisher version 2.7.107.2964. 
[07:10:32 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] Set Monitoring to Reporting for 5 nodes in subscription 179 

~1 minute
```

This configuration has around 20,000 nodes, our biggest install has around 90,000 which makes this performance difference even bigger.

Logs from my testing are also in the attached zip.

Let me know if you need anything else.

[OpcPublisherExample.zip](https://github.com/Azure/Industrial-IoT/files/6356357/OpcPublisherExample.zip)

dacolgit, 2021-04-22:
Thanks @jmbrunskill

dacolgit, 2021-04-23:
Hi @jmbrunskill,

I made a lot of test. Here are my results.
I run all the test with the publisher running as IoTEdge module and as a standalone application.

I first used the configuration file you provide for KepWare server with 20000 tags, but always with good startup time.

I then added further 20000 tags to your configuration for a total of 40000 tags. Here are my results.

```
publisher as IoTEdge module

[15:07:43 INF Root] Starting module OpcPublisher version 2.7.206.28401.
[15:08:05 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices>>+SubscriptionWrapper] Set Monitoring to Reporting for 1000 nodes in subscription opc.tcp://192.168.100.20:49320_bb050195c5624bb070485f8caa86a900eaa4b4e3

Publisher running outside iotEdge context 

[16:50:10 INF Root] Starting module OpcPublisher version 2.7.206.28401.
[16:50:54 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] Set Monitoring to Reporting for 1000 nodes in subscription opc.tcp://192.168.100.20:49320_f4c8b40fc37f69eb5ec8b362666ac6f61536c47b
```

As you can see I cannot reproduce your issue.
My setup is the following:
- Publisher running as standalone application on my PC
- Publisher running as IoTEdge Module on a Linux VM (Eflow). The Vm is configured to use just one core and 2048 MB of Ram
- KepWare Server version 6.8.796.0 running on my PC

For my test Publisher and server are running on the same machine.

What exactly is your test setup?
Thanks,
Daniele

Logs from my testing are attached.
[logs.zip](https://github.com/Azure/Industrial-IoT/files/6366658/logs.zip)

jmbrunskill, 2021-06-10:
We found that the parameter --fd=true was causing this delay.

This parameter enables fetching OpcNode DisplayName which isn't required for our use case.

By disabling this parameter, I'm seeing OPCPublisher subscription time down to around 30seconds where it was taking 6-7 minutes with --fd=true

closed 2021-06-10 by dacolgit