Azure/Industrial-IoT #748: Question: releases and opc-publisher version numbers

apittier, 2020-10-02:
Hello, 

I'd like the confirm the correspondence between the Industrial-IoT releases and the opc-publisher edge module docker image versions.

Does _"the latest release - version 2.7.199 of the Industrial IoT Platform"_ correspond to the docker image _"mcr.microsoft.com/iotedge/opc-publisher:2.7.199"_ ?

(It confuses me that the latest available version of the opc-publisher seems to be: 2.8.9).
https://mcr.microsoft.com/v2/iotedge/opc-publisher/tags/list

Thanks,
Amparo

koepalex, 2020-10-05:
HI Amparo,

there are currently two versions under development 2.7.x and 2.8.x. 2.7.x is currently the released one. Version 2.8.x is under development and not official released, you can check [Releases Page](https://github.com/Azure/Industrial-IoT/releases).

So you should use the 2.7.199 image.

regards,
Alex

apittier, 2020-10-05:
Perfect. Many thanks @koepalex.

closed 2020-10-05 by apittier