Azure/Industrial-IoT #752: OPCPublisher delays before first message is sent to IoTHub

githubjanson, 2020-10-06:
Setup:

- Windows VM running IoTEdge with OPCPublisher module
- Windows VM with OPC UA simulation server with 2 nodes changing values every 5 s
- OPCPublisher is run with command flag --si=5, i.e., it should send messages every 5 s to IoTHub

When I start the OPC UA simulation, I can immediately see in the OPCPublisher diagnostics log that it registers the value changes. But there is a delay of about 2m30s until it sends the first message to IoTHub, and this message includes all the value changes up until then. After that, it sends messages every 5 s as it should.

I don't understand this behaviour. Any thoughts on this?

Thanks.

koepalex, 2020-10-07:
Which version of the OPC Publisher are you using?

barnstee, 2020-10-20:
Closed due to inactivity. If you still experience this issue, please reopen.

closed 2020-10-20 by barnstee