Azure/Industrial-IoT #757: Not able to use  CA certificate which is not created by OPC Publisher module.

SandeepThombare, 2020-10-12:
I am not able use CA certificate which is not created by OPC publisher module it means I don’t want to use certificate created by OPC publisher module during start-up.
I tried to copy valid. Der and .Pfx files to Own folder in /PKI directory so that module can see there is certificate in folder but still OPC publisher creating its own certificate. 

**To Reproduce**
Steps to reproduce the behaviour:
Run OPC publisher module with version 2.7.199 and use valid certificate. der and .pfx. instead of certificate created by OPC publisher module at the start-up.
**Expected behavior**
The same certificate should be used to established the trust between OPC publisher module and OPC UA server.
**Screenshots**
 
Container create option: -

{
"Hostname": "publisher",
"Cmd": [
"--pf=/appdata/pn.json",
"--ap=/pki/own",
"--tp=/pki/trusted",
"--rp=/pki/rejected",
"--lf=/pki/publisher.log",
"--mm=PubSub",
"--kt=4",
"--fm=true",
"--bs=15",
"--si=30",
"--sc=1",
"--ll=debug"
],
"HostConfig": {
"Binds": [
"/home/Final:/appdata",
"/home/edge3:/pki"
]
}
}
[image]

cristipogacean, 2020-10-12:
you need to make sure that the custom certificate (CA signed) contains the subject recognized by the publisher "CN = Microsoft.Azure.IIoT"
[image]

This certificate will be picked up by the publisher and no self signed certificate will be generated anymore

SandeepThombare, 2020-10-13:
Thanks cristipogacean for your reply.
I have created self signed certificate using openssl and as per your comment given the "CN = Microsoft.Azure.IIoT" and copied to the folder \own.  In this test, OPC publisher didn't use the  custom certificate . OPC publisher module have created and used.
please guide if I miss anything.

cristipogacean, 2020-10-16:
@SandeepThombare, you are right, the CN is not enough, the entire subject must match the expectation of the OPC Publisher. 

So there are 2 options:
1) either you create a certificate that has the following subject pattern:  "CN=Microsoft.Azure.IIoT, C=DE, S=Bav, O=Microsoft, DC=localhost", where localhost should be replaced with the hostname of the gateway running the iotedge with the publisher.

2) or you configure the publisher to use the subject in your custom certificate using a construct in the container create options of your publisher:
{
    "User": "ContainerAdministrator",
    "Hostname": "publisher",
    "Cmd": [
        "--pf=/mount/publishednodes.json",
        "--aa",
        "--tm",
        "PkiRootPath=/mount/pki",
        **"ApplicationName=test",
        "ApplicationCertificateSubjectName=CN=test, C=TestCountry, S=TestState, O=TestOrg, DC=localhost"**
    ],
    "HostConfig": {
        "Mounts": [
            {
                "Type": "bind",
                "Source": "C:\\mount",
                "Target": "C:\\mount"
            }
        ]
    }
}

where the ApplicationName and the CN= are the same. in the ApplicationCertificateSubjectName you need to apply the complete subject from your certificate. The DC=localhost will be replaced with the hostname of the edge device hosting the publisher. if the DC is different than localhost it will be used as specified. 
if your certificate is CA signed, you need to copy the CA's public certificate in the trusted store of the publisher. 

btw, the very same applies to twin and discovery modules.

Hope this helps you further.

SandeepThombare, 2020-10-22:
Thanks cristipogacean for your reply. I have tried both options that you mentioned in your reply.
1.I have created the self signed certificate with same subject details as Module does(CN = Microsoft.Azure.IIoT,C = DE,S = Bav,O = Microsoft,DC = virtualbox). 

Observation--> Publisher Module has moved this certificate to rejected folder and create another certificate. 

2. As per second option, I have configured the publisher with custom certificate subject details (CN= selfcert,C=DE,S=Bav,O=microsoft1).

Observation--> Publisher module has created one more certificate with subject name same as custom self signed certificate(CN= selfcert,C=DE,S=Bav,O=microsoft1) and that has been sent to KEPServer.

Could you please check and let me know if anything needs to be improved to use self signed certificate instead of certificate created by Publisher module.

Thanks a lot in advance.

cristipogacean, 2020-10-23:
@SandeepThombare, I did the same and works for me, so probably we're generating the certificate differently. Would you be able to generate and upload a certificate for me? or give me some more details about how you're generating your certificate?

SandeepThombare, 2020-10-26:
Below steps I have used to create the self signed certificate,

Certificate creation steps:-
1. **Create RSA private key**
openssl genrsa -des3 -out server.pass.key 2048
openssl rsa -in server.pass.key -out server.key

2. **Create Certificate Signing request(CSR) utilizing the RSA private key**
openssl req -new -key server.key -out server.csr -subj /CN=Microsoft.Azure.IIoT/C=DE/ST=Bav/O=Microsoft/DC=hostname

3. **Generate file cert.ext with below details**
authorityKeyIdentifier= keyid,issuer
basicConstraints=CA:FALSE
keyUsage=digitalSignature, nonRepudiation, keyEncipherment, dataEncipherment ,certificateSigning
extendedKeyUsage=serverAuth,clientAuth
subjectKeyIdentifier=hash
subjectAltName=@alt_names

[alt_names]
URI.1 = urn:localhost:Microsoft.Azure.IIoT:microsoft,
DNS.1 = Localhost,
IP.1 = xx.xx.xx.xx"

Note:- In above file, getting error if i mentioned certificateSigning in keyUsage.
       Replace localhost with hostname of the edge device in [alt_names] section above.

4. **Create SSL certificate using CSR created in last step**
openssl x509 -req -sha256 -extfile cert.ext -days 365 -in server.csr -signkey server.key -out server.der

5. **Create pfx file using server.key and server.der**
openssl pkcs12 -export -name server -in server.der -inkey server.key -out server.pfx

Above steps are generating two files 
1) server.der
2) server.pfx

Copied generated files as below,
1. Copy .der file to pki/own/certs and pki/trusted/certs
2..pfx file to pki/own/private 

Please let me know if anything I need to change in any steps or any issue in certificate creation.

cristipogacean, 2020-10-27:
@SandeepThombare ,

I think your issue is at step 5. I'm assuming you are providing a password when the pfx is generated. the OPC publisher is not aware about this password and cannot open the pfx. Can you try generating it without a password as follows:

openssl pkcs12 -export -keypbe NONE -certpbe NONE -name server -in server.der -inkey server.key -out server.pfx
and then press Enter twice when asked for password. 

I also changed one thing in the cert.ext file, but I think this should not make a difference:
extendedKeyUsage=clientAuth
this certificate is only used for OPC UA client purpose

SandeepThombare, 2020-10-28:
@cristipogacean thanks for reply.

I have created the certificate and pfx file without password and copied to respective folder locations.

With this certificates, module haven't created any certificate and it has used the custom certificated created using openssl. Which is good news for me.

But, when this certificate I am trying to trust on my OPC UA server(KEPserverEx 6) it is giving error "The certificate is signed by CA that is untrusted or invalid.....". attaching the screenshot of error.
[image]

So in short, the Module is using custom certificate and exchanging that with KEPserver but KEPserver is not trusting the same and because of this issue Edge and KEPServerEx is not communicating.

Not sure what is missing here. Please let me know if anything is missed here.

Attaching the certificate used in above test,
[cert.zip](https://github.com/Azure/Industrial-IoT/files/5453797/cert.zip)

If you can share the Certificate creation steps that will help me to get it work.

cristipogacean, 2020-10-28:
@SandeepThombare 

Indeed, I managed to replicate this issue with the kepware server. weird is that other servers accept it... 
Anyhow, I tuned the cert.ext as follows and this works for me. Would you be so kind and give it a try?

```
authorityKeyIdentifier=keyid,issuer
basicConstraints=CA:TRUE
keyUsage=digitalSignature, nonRepudiation, keyEncipherment, dataEncipherment, keyCertSign, cRLSign
extendedKeyUsage=clientAuth
subjectKeyIdentifier=hash
subjectAltName=@alt_names

[alt_names]
URI.1 = urn:xxx:Microsoft.Azure.IIoT:microsoft:
DNS.1 = xxx
IP.1 = xxx.xxx.xxx.xxx

```

cristipogacean, 2020-11-05:
Closing for now, please reopen if needed.

closed 2020-11-05 by cristipogacean