Azure/Industrial-IoT #759: Describe logs

Ayushi-shah1, 2020-10-13:
I am using the latest OPC twin module of image URI as mcr.microsoft.com/iotedge/opc-twin:latest in IoT edge which running in Windows VM. But the OPCTwin is unable to detect OPC UA server. I am getting repeatedly same logs.
Below is the log of OPC Twin.

[05:48:57 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Connecting all clients to 10.0.0.9 using MqttOverTcp.
[05:48:57 INF Root] Starting module OpcTwin version 2.7.170.10628.
[05:48:57 INF Root] Initiating prometheus at port 9701/metrics
[05:48:57 ERR Root] Unable to start metric server. For more info, please check troubleshooting guide for edge metrics collection
System.Net.HttpListenerException (5): **Access is denied.**
   at System.Net.HttpListener.AddPrefixCore(String registeredPrefix)
   at System.Net.HttpListener.AddAllPrefixes()
   at System.Net.HttpListener.Start()
   at Prometheus.MetricServer.StartServer(CancellationToken cancel)
   at Prometheus.MetricHandler.Start()
   at Microsoft.Azure.IIoT.Module.Framework.Client.MetricServerEx.StartWhenEnabled(IMetricServer server, IModuleConfig config, ILogger logger) in D:\a\1\s\common\src\Microsoft.Azure.IIoT.Hub.Module.Client\src\Extensions\MetricServerEx.cs:line 23
[05:48:57 WRN Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Bypassing certificate validation for client.
[05:48:57 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Running in iotedge context.
[05:49:00 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] 0: Module dt-23sep-edgedev_OPCTwin disconnected due to Retry_Expired - now Disconnected...
[05:54:06 ERR Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Module Host failed to start.
[05:54:06 ERR Root] Error during module execution - restarting!
System.AggregateException: One or more errors occurred. (MQTT channel open failed.)
 ---> Microsoft.Azure.Devices.Client.Exceptions.IotHubCommunicationException: MQTT channel open failed.
   at Microsoft.Azure.Devices.Client.Transport.Mqtt.MqttTransportHandler.<>c__DisplayClass93_0.<<CreateChannelFactory>b__0>d.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at Microsoft.Azure.Devices.Client.Transport.Mqtt.MqttTransportHandler.OpenAsyncInternal(CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.Transport.Mqtt.MqttTransportHandler.OpenAsync(CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.Transport.ProtocolRoutingDelegatingHandler.OpenAsync(CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.Transport.ErrorDelegatingHandler.<>c__DisplayClass23_0.<<ExecuteWithErrorHandlingAsync>b__0>d.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at Microsoft.Azure.Devices.Client.Transport.ErrorDelegatingHandler.ExecuteWithErrorHandlingAsync[T](Func`1 asyncOperation)
   at Microsoft.Azure.Devices.Client.Transport.RetryDelegatingHandler.<>c__DisplayClass33_0.<<OpenInternalAsync>b__0>d.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at Microsoft.Azure.Devices.Client.Transport.RetryDelegatingHandler.EnsureOpenedAsync(CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.InternalClient.OpenAsync()
   at Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory.ModuleClientAdapter.CreateAsync(String product, IotHubConnectionStringBuilder cs, String deviceId, String moduleId, ITransportSettings transportSetting, TimeSpan timeout, IRetryPolicy retry, Action onConnectionLost, ILogger logger) in D:\a\1\s\common\src\Microsoft.Azure.IIoT.Hub.Module.Client\src\Default\IoTSdkFactory.cs:line 267
   at Microsoft.Azure.IIoT.Utils.Try.Options[T](Func`1[] options)
   --- End of inner exception stack trace ---
   at Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost.StartAsync(String type, String siteId, String productInfo, String version, IProcessControl reset) in D:\a\1\s\common\src\Microsoft.Azure.IIoT.Hub.Module.Framework\src\Hosting\ModuleHost.cs:line 169
   at Microsoft.Azure.IIoT.Modules.OpcUa.Twin.ModuleProcess.RunAsync() in D:\a\1\s\modules\src\Microsoft.Azure.IIoT.Modules.OpcUa.Twin\src\ModuleProcess.cs:line 109
[05:54:06 INF Root] Stopped prometheus metric server
[05:54:06 INF Microsoft.Azure.IIoT.OpcUa.Edge.Supervisor.Services.SupervisorServices] All twins stopped - supervisor exiting...
[05:54:06 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Connecting all clients to 10.0.0.9 using MqttOverTcp.
[05:54:06 INF Root] Starting module OpcTwin version 2.7.170.10628.
[05:54:06 INF Root] Initiating prometheus at port 9701/metrics
[05:54:06 ERR Root] Unable to start metric server. For more info, please check troubleshooting guide for edge metrics collection
System.Net.HttpListenerException (5): Access is denied.
   at System.Net.HttpListener.AddPrefixCore(String registeredPrefix)
   at System.Net.HttpListener.AddAllPrefixes()
   at System.Net.HttpListener.Start()
   at Prometheus.MetricServer.StartServer(CancellationToken cancel)
   at Prometheus.MetricHandler.Start()
   at Microsoft.Azure.IIoT.Module.Framework.Client.MetricServerEx.StartWhenEnabled(IMetricServer server, IModuleConfig config, ILogger logger) in D:\a\1\s\common\src\Microsoft.Azure.IIoT.Hub.Module.Client\src\Extensions\MetricServerEx.cs:line 23

cristipogacean, 2020-10-13:
Try setting the "User": "ContainerAdministrator", in the twin module's container create options. 
E.g.
```
{
    "Hostname": "twin",
    "User": "ContainerAdministrator",
    "Cmd": [
        "PkiRootPath=/mount/pki"
    ],
    "HostConfig": {
        "Mounts": [
            {
                "Type": "bind",
                "Source": "C:\\mount",
                "Target": "C:\\mount"
            }
        ]
    }
}
```

marcschier, 2021-07-14:
Close due to no activity.

closed 2021-07-14 by marcschier