Azure/Industrial-IoT #774: OPC Publisher stops publishing tags when others are updated (scan rates)

gregeva, 2020-10-14:
**Describe the bug**
Changing the scan rate of certain Nodes/tags seems to create a condition where the IoT Edge OPC Publisher either stops scanning or publishing all the other nodes which were not updated.

This has been tested and seen on the same application and endpoint across two versions (2.7.170 and 2.7.182).

**To Reproduce**
Steps to reproduce the behavior:
1. Subscribe to a number of OPC UA tags with a scan rate of 1000 ms (or whatever you want)
2. Fetch published Nodes to ensure all have been added with correct scan rates
3. Monitor telemetry on IoT Hub endpoint to ensure subscriptions were processed and data is flowing for all tags
4. Change the scan rate on a couple of the NodeIds (tags) from 1000ms to 500ms (or whatever you want)
5. Fetch published Nodes to ensure update took and all have correct scan rates
6. Monitor telemetry on IoT Hub endpoint to ensure subscriptions were processed and data is flowing for all tags
7. Wait patiently until the end of time for updates from the tags which did not have their scan rates changed (in step 4)
8. Once patience timeout reached, reset scan rates to be the same for all tags/nodes and watch data start flowing

**Expected behavior**
I would expect that the different node subscriptions would be able to handle different publishing rates.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Industrial IoT Application (please complete the following information):**
 - OS: Azure ;-)
 - Version: 2.7.170, 2.7.182

dacolgit, 2020-10-19:
Hi @gregeva 

What do you mean for scan rate?
Is it the rate at which node values change in the server (a scan rate of 1000 ms means a node changing his value every 1000ms ) ?

dacolgit, 2020-11-03:
Closed as not able to reproduce it and had no feedback. Please reopen if it's still an issue

gregeva, 2020-12-09:
Yes, I validate still seeing this issue and it is quite easy to reproduce.  See recorded video as well as captures from the test.

https://youtu.be/1X2tPwXlCOA
https://drive.google.com/file/d/1r6yfWA4DPVdb33GN7B8CIwQxzxGdfHOr/view?usp=sharing

gregeva, 2021-02-17:
@barnstee Could you please follow up on this with your team? This capability is expected by our joint customer. FYI @chmagitt

barnstee, 2021-02-17:
@dacolgit , please have another look, Greg got back to us.

gregeva, 2021-03-19:
Any update with this @dacolgit or @barnstee .  Our customer will soon be looking for this, and our dev team is also seeing this issue.

ervoges, 2021-03-19:
Adding some additional information about this issue.

Seems like the dataSetWriterID is not getting populated properly for those of us running into this issue.  On a working set-up, in the endpoint JSON on our CosmosDB I see this when I publish:
```
"dataSetWriters": [
        {
                "dataSetWriterId": "uat08804e2aff7cb3a758defb3a50f5d4af2b0cfbf4_2000",
                "dataSet": {
                    "dataSetSource": {
                        "publishedVariables": {
                            "publishedData": [
                                {
                                    "publishedVariableNodeId": "nsu=ThingWorx Kepware Server;s=ATChannel.ATDevice.RampLong",
                                    "samplingInterval": "00:00:01",
                                    "queueSize": 1
                                }
                            ]
```

However when publishing on a different environment that is not able to get telemetry from all published nodes, I see this:
```
"dataSetWriters": [
           {
               "dataSetWriterId": "uat0854f26c50c78f57ee8791f7845136306902d6d0",
                "dataSet": {
                    "dataSetSource": {
                        "publishedVariables": {
                            "publishedData": [
                                {
                                    "publishedVariableNodeId": "nsu=ThingWorx Kepware Server;s=Simulation Examples.Functions.Ramp1",
                                    "samplingInterval": "00:00:01",
                                    "queueSize": 1
                                }
                            ]
```

Also if I publish a new node with a new publishing interval on the broken set-up, a new dataSetWriter is created but the dataSetWriterID stays as the same endpointID. Meaning I now have two different dataSetWriters using the same ID (an issue that seems to have been addressed in 2.7.199 release notes).

My publishing logs also tell me that the publisher removes any currently publishing nodes from their subscription before publishing a new node with a new publishing interval:
```
[18:44:43 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Subscription 'uat0854f26c50c78f57ee8791f7845136306902d6d0' registered/updated in session '25d991220e3155ebe835e739624851b6837263e5' in state Refresh
[18:44:43 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] Removed 1 monitored items from subscription uat0854f26c50c78f57ee8791f7845136306902d6d0
[18:44:43 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] Added 1 monitored items to subscription uat0854f26c50c78f57ee8791f7845136306902d6d0
[18:44:43 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] Now monitoring 1 nodes in subscription uat0854f26c50c78f57ee8791f7845136306902d6d0
[18:44:43 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Subscription 'uat0854f26c50c78f57ee8791f7845136306902d6d0' registered/updated in session '25d991220e3155ebe835e739624851b6837263e5' in state Refresh
[18:44:43 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] Set Monitoring to Reporting for 1 nodes in subscription uat0854f26c50c78f57ee8791f7845136306902d6d0 
```

I have noticed that when I go in an manually add the suffixes myself, I start getting telemetry from all my nodes again, but obviously I think the expectation is that the publisher should be updating the IDs properly on its own.

We are all using the same version--2.7.206 and so theoretically should have the same publisher. However it looks like some people's publishers are forgetting to add the publishing interval suffix on the end of the endpointID to create a unique dataSetWriterID. We've tried looking for other difference between the working set-ups and the broken ones but so far have not been able to find any meaningful differences.

marcschier, 2021-03-20:
Hi @gregeva and @ervoges we will look into it.  When you say setup, does this mean aks cluster?  Do you have more clusters connected to the same cosmos db database by chance?  Or just multiple competing clients updating?

ervoges, 2021-03-22:
Hi @marcschier , thanks so much for looking into it. So we have one CosmosDB per deployment and one namespace. Most of the namespaces on my team are in the same cluster, but I am not sure where Greg's deployment is. I don't think we have clients competing to update because as far as I know I am the only one using my endpoint and application.

gregeva, 2021-03-22:
My cluster, namespace, documentDB, and opinions are my own. :-)

I've tested and seen this behaviour on the two supported versions that I've worked with; initially 2.7.170 and more recently 2.7.182.  Unlike my cutting edge colleagues, I've yet to make the leap to 2.7.199 or 2.7.206.

marcschier, 2021-03-23:
Hi @ervoges, @gregeva, we will look into it.   However, if 2 namespaces (at different version levels) share the same cosmos db collection and documents, then it is possible that you get random behavior with regards to your document contents.   While a sequential upgrade scenario is supported, simultaneous operation of new and old versions on the same database is not.  Also, having more than a single namespace associated with a unique set of underlying Azure resources is also not supported.  I would recommend setting up a full resource group of resources for every namespace in the cluster or a full cluster per deployed resources group "environment".

ervoges, 2021-03-23:
Hi, @marcschier thanks for the info--I think I might have not done the best job explaining our environment, sorry about that! 
We all have our own namespaces, with our own resources, with our own CosmosDB. No two resource groups share the same CosmosDB or write to the same documents. No resource groups share any resources. Additionally, at least in my team, we are all on the same version: 2.7.206.
When I attempt to get telemetry from two nodes with different publishing intervals using my environment (personal resources and resource group that nobody else uses), I am not able to get telemetry from both nodes.

Hope that helps!

gregeva, 2021-03-23:
We've followed the documented approach by Microsoft and PTC.  We're not doing anything crazy here if that is what you are thinking.

karok2m, 2021-03-23:
@ervoges, #689 was specifically intended to fix the issue of duplicate `dataSetWriterId`s in the job definition. The fix was included already in `2.7.199` release. Are you still seeing different dataSetWriters being created with the same `dataSetWriterId` with `2.7.206` version ?
Just to clarify this a bit, after the #689 fix, `dataSetWriterId` is generated based on endpoint id and the publishing interval. If the publishing interval has been specified, then it will be appended to the endpoint id to get the `dataSetWriterId`. But if the publishing interval has not been specified, then nothing will be added after the endpoint id. So it is OK to see `dataSetWriterId` without a publishing interval suffix.

karok2m, 2021-03-23:
@ervoges, hm, this is quite puzzling. The behavior that you are describing is what I would expect from `2.7.182`. A few questions here:
1. Have you upgraded OPC Publisher edge module itself or also the back-end microservices to `2.7.206` version ? The job definitions are generated by Publisher back-end microservice. So if you have only upgraded the OPC Publisher edge module, then you would still see the issue.
2. Can you please provide logs of Publisher back-end microservice ? You can use `kubectl logs` to get the logs. But you would first need to get the pod name of Publisher back-end microservice. That would the the one with `-publisher-` in the name. Here the commands that I've used to get logs from my deployment, please adjust the commands with your namespace name:
  ```bash
  PS C:\dev\misc\e2etesting\20210323.e2etesting20210323343.Orchestrated> kubectl get pods -n azure-industrial-iot
  NAME                                                           READY   STATUS    RESTARTS   AGE
  azure-industrial-iot-edge-jobs-7b74f76657-nd6bf                1/1     Running   0          98m
  azure-industrial-iot-engineering-tool-6dc977fcb5-j57tv         1/1     Running   0          98m
  azure-industrial-iot-events-776f6dbffb-jt228                   1/1     Running   0          98m
  azure-industrial-iot-events-processor-57b7964966-xmwpt         1/1     Running   0          98m
  azure-industrial-iot-gateway-765bfc6f64-q7ldr                  1/1     Running   0          98m
  azure-industrial-iot-history-6fbc4d6df7-g9d4k                  1/1     Running   0          98m
  azure-industrial-iot-onboarding-549c56448b-mhmtc               1/1     Running   0          98m
  azure-industrial-iot-publisher-57cd69cd6f-skgmv                1/1     Running   0          98m
  azure-industrial-iot-registry-78c6b9cbb-pmxn9                  1/1     Running   0          98m
  azure-industrial-iot-sync-549fdbdbf8-bvpz5                     1/1     Running   0          98m
  azure-industrial-iot-telemetry-cdm-processor-d7d7b6dbb-vb2mj   1/1     Running   0          98m
  azure-industrial-iot-telemetry-processor-ddf9c9b9d-l2sp9       1/1     Running   0          98m
  azure-industrial-iot-tunnel-processor-68bf85f7cc-hdm2f         1/1     Running   0          98m
  azure-industrial-iot-twin-766d8684bc-tvlpj                     1/1     Running   0          98m
  PS C:\dev\misc\e2etesting\20210323.e2etesting20210323343.Orchestrated>
  PS C:\dev\misc\e2etesting\20210323.e2etesting20210323343.Orchestrated> kubectl logs azure-industrial-iot-publisher-57cd69cd6f-skgmv -n azure-industrial-iot
  ...
  ```
3. What API endpoints are you using to publish and unpublish nodes, single-node (`/v2/publish/{endpointId}/start` and `/v2/publish/{endpointId}/stop`) or bulk-publishing (`/v2/publish/{endpointId}/bulk`) endpoints ?

And regarding your logs: they actually look fine. If a new DataSetWriter was generated for your new node with the same `dataSetWriterId`, then having duplicate `dataSetWriterId`s could have resulted in OPC Publisher edge module only receiving new DataSetWriter and thus removing nodes that belonged to the old one. This should be resolved with distinct `dataSetWriterId`s.

ervoges, 2021-03-23:
@karok2m I listed the information you asked for below!

1) Yes, I have redeployed a few times just to make certain that I was on the correct version.  Although if there are other things I should check other than my pods and iotedge module please let me know!

2) Here is the result when I run `kubectl get pods` on my namespace:
```
NAME                                                            READY   STATUS    RESTARTS   AGE
azure-industrial-iot-edge-jobs-7955b7fdd7-xshr5                 1/1     Running   0          29h
azure-industrial-iot-engineering-tool-69755b6f9f-wwgpx          1/1     Running   0          29h
azure-industrial-iot-events-599b8797f4-p87v7                    1/1     Running   0          29h
azure-industrial-iot-events-processor-5b88957cc6-fqrrq          1/1     Running   0          29h
azure-industrial-iot-gateway-c4c547678-srrc2                    1/1     Running   0          29h
azure-industrial-iot-history-6b49f6c67f-7sjg8                   1/1     Running   0          29h
azure-industrial-iot-onboarding-7d764b4b9b-h54wm                1/1     Running   0          29h
azure-industrial-iot-publisher-6c776bd67f-vr7jq                 1/1     Running   0          29h
azure-industrial-iot-registry-998dcdbd4-8gkm9                   1/1     Running   0          29h
azure-industrial-iot-sync-549945995b-dmdxk                      1/1     Running   0          29h
azure-industrial-iot-telemetry-cdm-processor-6744bdf764-92m54   1/1     Running   0          29h
azure-industrial-iot-telemetry-processor-55dbbf7bdd-dc7vh       1/1     Running   0          29h
azure-industrial-iot-tunnel-processor-577655d7f9-zgkzh          1/1     Running   0          29h
azure-industrial-iot-twin-6479b7b9d5-xf2wm                      1/1     Running   0          29h
azure-industrial-iot-vault-cd67bf785-rf5sd                      1/1     Running   0          21d
PS C:\Users\evogelsperger\source\protocol-adapter-azure-iot> kubectl logs azure-industrial-iot-publisher-6c776bd67f-vr7jq -n pegasus-stratus-evogelsperger
[14:17:08 INF Microsoft.Azure.IIoT.OpcUa.Publisher.Deploy.IoTHubPublisherDeployment] Updating opc publisher module deployment with image mcr.microsoft.com/iotedge/opc-publisher:2.7.206
[14:17:10 INF Microsoft.Azure.IIoT.OpcUa.Publisher.Deploy.IoTHubPublisherDeployment] Updating opc publisher module deployment with image mcr.microsoft.com/iotedge/opc-publisher:2.7.206
[14:17:10 INF Microsoft.Azure.IIoT.Utils.HostAutoStart] All hosts started.
[14:17:10 INF Microsoft.Azure.IIoT.AspNetCore.Cors.CorsSetup] CORS is enabled
[14:17:10 INF Microsoft.Azure.IIoT.AspNetCore.Cors.CorsSetup] Allow all headers, origins and methods
[14:17:10 INF Root] Opc-Publisher-Service web service started with id bf321285-fed8-481d-82fb-f7a4d974646e
```

3) I am using `https://{{OPC-SERVICEURL}}/publisher/v2/publish/{endpointID}/start`

koepalex, 2021-03-24:
@ervoges just to make Sure, did you already tried to delete the Job (which removes the cosmosdb entry) and are able to reproduce with the instance above?

ervoges, 2021-03-24:
@koepalex by "delete the job" do you mean have I tried deleting the endpoint so that the endpoint file is gone from CosmosDB or manually unpublishing the node from CosmosDB? (or something else entirely :) )

And yes, I am able to reproduce with the instance I listed yesterday. I reproduced the bug and then copied and pasted the logs from my pods.

karok2m, 2021-03-24:
@ervoges, to delete a job you would either unpublish all nodes that are present in it or you can perform `DELETE` on `https://{{OPC-SERVICEURL}}/publisher/v2/jobs/{id}` endpoint to delete a job as a whole. Here the idea is to make sure that there would not be several dataSetWriters (with duplicate `dataSetWriterId`s) within the job definition that would have been created by a version prior to `2.7.199` before performing the tests.

ervoges, 2021-03-24:
@karok2m Got it--I have already been unpublishing all nodes and republishing many times as I've been testing and working, but I have not tried the API endpoint you just listed so let me see if that does anything different.

ervoges, 2021-03-24:
@koepalex @karok2m I deleted the entire job using `https://{{OPC-SERVICEURL}}/publisher/v2/jobs/{id}` and then republished two nodes with different publishing intervals and I still have two dataSetWriters with the same ID.

karok2m, 2021-04-01:
@gregeva, @ervoges, just to summarize our findings from today, `2.7.206` version fixes issue with duplicate `dataSetWriterId`s and as a result data from both dataSets is arriving to the cloud. Would you need any additional information regarding this issue ? Or can we close it ?

ervoges, 2021-04-01:
@karok2m Thank you so much again for the meeting today. After looking deeper into the issue after the meeting, turns out there were two kubernetes deployments running in two different namespaces pointing to the same service URL. One namespace was running 2.7.206 and the other namespace was running 2.7.170. I did not know the second namespace existed and so when I was checking my pods versions and updating them, I kept seeing 2.7.206 deployed on the first namespace.

Layered deployment was on and working and grabbed 2.7.206 from the first namespace for the modules, but the published nodes were talking to microservices from the second namespace running 2.7.170. Seems like the duel namespaces sharing a service URL was the cause of my problems.

I'm comfortable closing this now! Thank you. :)

gregeva, 2021-04-02:
Thanks for resolving this @karok2m and @koepalex .  As I understand this issue was resolved in 2.7.199, and so will be all good once we are able to get our environments and customer upgraded to 2.7.206.

closed 2021-04-02 by gregeva