Azure/Industrial-IoT #813: Perf/OPC: Fails when more 468 values are added to publisher

NadeemKhan81, 2020-10-19:
**Describe the bug**
Customer is using the OPC publisher and added new tags to the publishednodes.json file. He has found that as soon as he has 469 tags the code fails. Log is attached. Is this a known limitation?

This is discussed in Industrial Iot Teams channel. 
[OPC Logs.txt](https://github.com/Azure/Industrial-IoT/files/5403900/OPC.Logs.txt)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior*
OPC publisher should be able to add new tags (Unless it reach the limit if any)

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: IoT Edge device ID
 - OS: [e.g. iOS8.1]- centos
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]- iotedge 1.0.9 or higher

**Additional context**
Add any other context about the problem here.

barnstee, 2020-10-20:
Added missing attachment.
[OPC Logs.txt](https://github.com/Azure/Industrial-IoT/files/5408115/OPC.Logs.txt)

barnstee, 2020-10-20:
@NadeemKhan81, from the logs, it is only loading 468 monitored items in each case. Can you update the bug with the pn.json file, please? There seems to be an error in the last item added.

cristipogacean, 2020-10-20:
@barnstee, the pn.json is uploaded in the internal teams channel. I did not notice any syntax error in the file. If we want to upload the file we should anonymize it first.

cristipogacean, 2020-10-20:
added the actual error, and my previews feedback 

[Error.txt](https://github.com/Azure/Industrial-IoT/files/5408254/Error.txt)

My hunch here the issue is a CreateMonitoredItems failure due to too many items and I believe the extensions I proposed for the standalone publisher will fix this issue by providing the user the option to control the content of subscriptions. Nevertheless it would make sense to try to replicate this issue.

NadeemKhan81, 2020-10-21:
@barnstee, i hope you got the required pn.json from the teams channel. Please let me know if you need anything else.
Thanks @cristipogacean.

dacolgit, 2020-10-27:
@NadeemKhan81 I started looking into the issue and I have a couple of question.
Which publisher version are you using?
I checked the publishednodes.json uploaded in the teams channel and it has 671 OpcNodes. My expectation was a publishednodes.json with 469 nodes. Is it the .json file that causes the issue?

NadeemKhan81, 2020-10-27:
@dacolgit , Yes this json file is from customer. It works fine till 468 tags. and once it crosses the threshold of approx 468 tags, it seems to fail to establish a subscription

Regharding publisher version will check with customer and update you.

NadeemKhan81, 2020-10-27:
@dacolgit , User configuration is set to pull the following:  mcr.microsoft.com/iotedge/opc-publisher:latest

dacolgit, 2020-10-29:
I'm trying to reproduce the issue, but unsuccessfully.

I need all the informations about the publisher and the server.
Do you confirm that are you using a Kep Server?
If yes, can you export the device configuration of the server as csv file and upload it in the
internal teams channel?

Is the publisher running as an IoT edge module in Windows or Linux?
Could you send your 'Container Create Options' configuration for the publisher module?

Thanks,
Daniele

NadeemKhan81, 2020-10-29:
Hi Daniele,
If you want I can set up a meeting with customer to understand the issue better.
Please let me know your convenient time to discuss with customer.

Could you please tell me your MS alias?

Thanks & Regards,
Nadeem Khan

Escalation Engineer | AI Intelligent Devices | Office: +1 (425) 5388510 | nadkha@microsoft.com
Mon -Fri (9:00 AM - 6:00 PM PST)

[Microsoft]

dacolgit, 2020-11-04:
The issue was caused by a bug in the --ms=0 configuration parameter.
A fix is merged in master.

closed 2020-11-04 by dacolgit

NadeemKhan81, 2020-11-05:
@dacolgit , Thanks for your effort on this to fix this issue. Appreatiate that.