Azure/Industrial-IoT #816: New Discovery Module not appearing in Engineering Tool Discovery. Proper procedure?

sabsher, 2020-10-21:
I deployed the Azure Industrial Platform all-in-one using the guide found here: https://azure.github.io/Industrial-IoT/deploy/howto-deploy-all-in-one.html

All has worked well! I then deployed a new IoT Industrial gateway (on the same device where I deployed the Industrial Platform) using this guide: https://azure.github.io/Industrial-IoT/deploy/quickstart-gateway-installer.html

The deployment also completed without issues. However, the Discovery module is not appearing in the engineering tool. All other modules appear and are connected under the "More" menu and I have verified all modules have good status within the IoT Hub.

Is there a documented method for adding additional discovery modules and/or IoT Edge Devices to a hub created by the all-in-one script?

chmagitt, 2020-10-23:
Same issue for me ...

sabsher, 2020-10-23:
> Same issue for me ...

I have been unable to find any information about how to resolve. Hopefully someone is able to provide some insight!

aguywithcode, 2021-02-22:
I've been running through the process of getting it up and running. One issue I ran into was that the Disco module did not properly initialize. There are two ways to check if this is the issue:

1. Login to the edge vm (using the username and password provided by the deploy script) and run iotedge list
2. Look at the edge device in the azure portal (it'll be listed under the generated iothub)

Verify that the modules are running, in the portal it should be reporting back as running and on the vm it should show in the list of running modules.

The problem I ran into was that the module version being requested wasn't on the list of available tags. I changed the version request in the portal to latest and it did pull down the correct version even though it's not visible on the list of tags.

closed 2023-01-16 by marcschier