Azure/Industrial-IoT #818: how to use encrypted username and password in Publishernodes.json-OPC publisher version 2.7.199

SandeepThombare, 2020-10-22:
**Describe the solution you'd like**
I want to use encrypted username and password in Publishernodes.json - OPC publisher module version 2.7.199-Standalone

I am using OPC publisher module to inject data from KEPserver to IOT Edge and IOT Hub. I have configured the Publishernodes.json file with required details of KEPserver . 

I want to use encrypted username and password of KEPServer in Publishernodes.json.

**Additional context**
Part of my publishernodes.json
**"UseSecurity":true
"OpcAuthenticationMode":"UsernamePassword",
"EncrypetedAuthUsername":"Someusername",
"EncrypetedAuthPassword":"Somepassword",**

How can I use encrypted username and password in above file.

Thanks in advance.

SandeepThombare, 2020-10-28:
Hi,

I am looking for help on this topic.
Please look into this @monikavar

karok2m, 2020-10-28:
`2.7.199` version of OPC Publisher does not have a way to encrypt username and password in `published_nodes.json` file similar to `2.5.x` version, that used direct methods for setting those values. `2.7.199` does support providing username and password through `published_nodes.json` (via `OpcAuthenticationUsername` and `OpcAuthenticationPassword` fields), but they will be there in plain text. Here is a sample of `published_nodes.json` for `2.7.199` version with username and password:
 
```json
[
    {
        "DataSetWriterGroup": "SoftingDataFEED",
        "DataSetWriterId" : "SoftingDataFEED_DataSetFlow_1",
        "DataSetPublishingInterval": 5000,
        "EndpointUrl": "opc.tcp://192.168.1.17:51210/UA/DemoServer",
        "UseSecurity": true,
        "OpcAuthenticationMode": "UsernamePassword",
        "OpcAuthenticationUsername": "usr",
        "OpcAuthenticationPassword": "pwd",
        "OpcNodes": [
            {
                "Id": "ns=3;i=10851",
                "DataSetFieldId":"VariableId_1",
                "OpcSamplingInterval": 1000,
                "OpcPublishingInterval": 3000
            },
            {
                "Id": "ns=3;i=10852",
                "DataSetFieldId":"VariableId_2",
                "OpcSamplingInterval": 1000,
                "OpcPublishingInterval": 3000
            }
        ]
    }
]
```
 
However, if you are upgrading from `2.5.x` version to `2.7.199`, and you have a `published_nodes.json` file which contains encrypted username and password (via `EncryptedAuthUsername` and `EncryptedAuthPassword` fields), then `2.7.199` version is able to load this file and decrypt username and password.

karok2m, 2020-11-05:
@SandeepThombare, I am closing this issue. Please reopen it if you need more information.

closed 2020-11-05 by karok2m