Azure/Industrial-IoT #827: What version of OPC Publisher should I use?

Jan-Olof, 2020-10-26:
I want to use OPC Publisher in IoT Edge as a stand alone module. I am confused as to what version or even repository I should use? (https://github.com/Azure/iot-edge-opc-publisher or https://github.com/Azure/Industrial-IoT.)

Can you help me?

barnstee, 2020-10-26:
You should always use the Link provided in the Azure Marketplace, it points to the latest and greatest version:
https://azuremarketplace.microsoft.com/en-us/marketplace/apps/microsoft_iot.iotedge-opc-publisher?tab=Overview

You shouldn't use the GitHub repos at all, since we build and test the source code ourselves and provide Docker containers through the Microsoft Container Registry.

closed 2020-10-26 by barnstee

Jan-Olof, 2020-10-26:
Thanks a lot for your quick answer! Much appreciated!