Azure/Industrial-IoT #846: Stuck at Signing In

darin4brugh, 2020-11-02:
New to this type of programming so probably missing something obvious.

I clone the repo then .\deploy and it starts. It then installs Az and AzureAD. It then just says Signing in .... It has been on this screen for over two hours.

I am a global admin in AAD. I also added App Admin and Cloud App Admin //not sure that matters once I am global.

I have also checked Yes in tenant properties under access management for resource properties. 

Does this mean I have tenant wide consent for my application?

Is there something I am missing?

darin4brugh, 2020-11-03:
Let it run over nite. When I arrived today I had message 
Warning '***' contains more than one active subscription. First one will be selected for further use. 
Followed by 
Please choose a subscription from this list. then my two subscriptions.

Its running and building now.

closed 2020-11-03 by darin4brugh

saramonteiro, 2021-12-28:
I am facing the same issue.
I don't have multiple subscriptions.

saramonteiro, 2021-12-30:
How I solved:
Using `Connect-AzAccount -TenantId <my tenant id>` 
We also can use `az login`