Azure/Industrial-IoT #858: Simulators trying to pull non-existent images

MatthijsvdVeer, 2020-11-06:
**Describe the bug**
When deploying the full solution, the Edge VM's don't behave as expected. Not all modules are loaded. On closer inspection of the Linux agent, I noticed it's trying to pull version tag `:2.8.100` of the publisher, discovery and twin modules. The tag list for these modules don't include 2.8.100, but only the following:

    "2.8.100-alpha",
    "2.8.100-alpha-debug",
    "2.8.100-alpha-linux-amd64",
    "2.8.100-alpha-linux-amd64-debug",
    "2.8.100-alpha-linux-arm32v7",
    "2.8.100-alpha-linux-arm32v7-debug",
    "2.8.100-alpha-linux-arm64v8",
    "2.8.100-alpha-linux-arm64v8-debug",
    "2.8.100-alpha-nanoserver-amd64-1809",
    "2.8.100-alpha-nanoserver-amd64-1809-debug",
    "2.8.100-alpha-nanoserver-amd64-1909",
    "2.8.100-alpha-nanoserver-amd64-1909-debug",

**To Reproduce**
1. Deploy full solution
2. SSH to Linux VM
3. Run `iotedge logs -f edgeAgent`
4. See all the errors regarding the module image not being pulled
```
---> (Inner Exception #2) Microsoft.Azure.Devices.Edge.Agent.Edgelet.EdgeletCommunicationException- Message:Error calling Create module discovery: Could not create module discovery
        caused by: Could not pull image mcr.microsoft.com/iotedge/discovery:2.8.100
```

**Expected behavior**
Alle Edge VMs should have a valid configuration after deployment

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu 16.04

**Additional context**
N/A

closed 2020-11-11 by karok2m

mm-mbinder, 2020-11-18:
I'm still having this issue. Deployed the full solution and the VMs try to load 2.8.103 as version for the discovery, twin and publisher modules, which does not exist in the list of tags.  Specifying the version with the -version flag didn't work either.

karok2m, 2020-11-25:
@mm-mbinder, please use our latest release for deployments, which is `2.7.199`. To do that:
* Checkout `release/2.7.199` branch.
* Deploy the platform and specify the version to be deployed in command line:

  For windows:
  ```bash
  .\deploy -type all -version "2.7.199"
  ```

  For Linux
  ```bash
  ./deploy.sh -type all -version "2.7.199"
  ```