Azure/Industrial-IoT #861: Unable to register manually OPC endpoint

ma3goul, 2020-11-09:
Hi everyone ,

**Describe the bug**
I installed the iiot platform with it's microservices , I connected my raspberry as industrial iot gateway to an OPC UA server : all the needed modules were deployed on the raspberry .

When I try to add manually the OPC server endpoint using  postman (registry/v2/applications) , nothing appear when I try to list the applications .

I even try it on the microsoft opc-plc server but it's not working also .

**Expected behavior**
Register Manually an opcua endpoint

Thank you.

denizetkar, 2021-06-11:
I'm using [Helm 0.3.2 version](https://github.com/Azure/Industrial-IoT/tree/helm_0.3.2) and having the same problem. In my case, I deployed the IIoT cloud modules into my minikube k8s cluster and I deployed the edge modules into an Ubuntu VM using also the example  opc ua server `mcr.microsoft.com/iotedge/opc-plc`.
My OS is Windows 10 and I think this problem has to do with **onboarding module** not working. So, when I run `kubectl get pods -n azure-iiot-ns`:
```bash
NAME                                              READY   STATUS    RESTARTS   AGE
azure-iiot-edge-jobs-b98fd4bcb-tb2v8              1/1     Running   0          38s
azure-iiot-events-79848fbcbf-9g8sj                1/1     Running   0          38s
azure-iiot-events-processor-75c985f9b7-2xlc8      0/1     Error     1          38s
azure-iiot-gateway-756967d779-bsfmr               1/1     Running   0          38s
azure-iiot-history-8999f6796-glhzn                1/1     Running   0          38s
azure-iiot-onboarding-5db46cb77f-hhhc7            0/1     Error     1          38s
azure-iiot-publisher-55988b778d-b57qv             1/1     Running   0          38s
azure-iiot-registry-55b98d9f6c-j5pl6              1/1     Running   0          38s
azure-iiot-sync-5c4479cd49-dnxjv                  1/1     Running   0          38s
azure-iiot-telemetry-processor-5647594c59-cc6g6   0/1     Error     1          38s
azure-iiot-tunnel-processor-7954d89fb8-k2hph      0/1     Error     1          38s
azure-iiot-twin-7d5b98fd9d-fkpkf                  1/1     Running   0          38s
```

Also if I take a peek into the logs, `kubectl logs kubectl logs azure-iiot-onboarding-5db46cb77f-hhhc7 -n azure-iiot-ns`:
```bash
[18:09:12 INF Microsoft.Azure.IIoT.Hub.Processor.EventHub.EventProcessorHost] Using Consumer Group: "onboarding" 
[18:09:12 INF Microsoft.Azure.IIoT.AspNetCore.Diagnostics.Default.MetricServerHost] Started prometheus at 9501/metrics 
[18:09:12 INF Microsoft.Azure.IIoT.Messaging.Default.EventBusHost] Event bus host running.
[18:09:13 ERR Microsoft.Azure.IIoT.Hub.Processor.EventHub.EventProcessorHost] Error starting event processor host.
Microsoft.Azure.EventHubs.Processor.EventProcessorConfigurationException: Encountered error while fetching the list of EventHub PartitionIds
 ---> Microsoft.Azure.EventHubs.MessagingEntityNotFoundException: The messaging entity 'sb://ihsuprodamres112dednamespace.servicebus.windows.net/iot-hub-materialfluss' could not be found. To know more visit https://aka.ms/sbResourceMgrExceptions.
   at Microsoft.Azure.EventHubs.Amqp.Management.AmqpServiceClient.GetRuntimeInformationAsync()
   at Microsoft.Azure.EventHubs.EventHubClient.GetRuntimeInformationAsync()
   at Microsoft.Azure.EventHubs.Processor.PartitionManager.GetPartitionIdsAsync()
   --- End of inner exception stack trace ---
   at Microsoft.Azure.EventHubs.Processor.PartitionManager.GetPartitionIdsAsync()
   at Microsoft.Azure.EventHubs.Processor.PartitionManager.GetPartitionIdsAsync()
   at Microsoft.Azure.EventHubs.Processor.PartitionManager.InitializeStoresAsync()
   at Microsoft.Azure.EventHubs.Processor.PartitionManager.StartAsync()
   at Microsoft.Azure.EventHubs.Processor.EventProcessorHost.RegisterEventProcessorFactoryAsync(IEventProcessorFactory factory, EventProcessorOptions processorOptions)
   at Microsoft.Azure.IIoT.Hub.Processor.EventHub.EventProcessorHost.StartAsync() in D:\a\1\s\common\src\Microsoft.Azure.IIoT.Hub.Processor\src\EventHub\EventProcessorHost.cs:line 101
[18:09:13 ERR Microsoft.Azure.IIoT.Services.Processor.Onboarding.HostStarterService] Failed to start some hosts.
Microsoft.Azure.EventHubs.Processor.EventProcessorConfigurationException: Encountered error while fetching the list of EventHub PartitionIds
 ---> Microsoft.Azure.EventHubs.MessagingEntityNotFoundException: The messaging entity 'sb://ihsuprodamres112dednamespace.servicebus.windows.net/iot-hub-materialfluss' could not be found. To know more visit https://aka.ms/sbResourceMgrExceptions.
   at Microsoft.Azure.EventHubs.Amqp.Management.AmqpServiceClient.GetRuntimeInformationAsync()
   at Microsoft.Azure.EventHubs.EventHubClient.GetRuntimeInformationAsync()
   at Microsoft.Azure.EventHubs.Processor.PartitionManager.GetPartitionIdsAsync()
   --- End of inner exception stack trace ---
   at Microsoft.Azure.IIoT.Hub.Processor.EventHub.EventProcessorHost.StartAsync() in D:\a\1\s\common\src\Microsoft.Azure.IIoT.Hub.Processor\src\EventHub\EventProcessorHost.cs:line 106
   at Microsoft.Azure.IIoT.Services.Processor.Onboarding.HostStarterService.StartAsync(CancellationToken cancellationToken) in D:\a\1\s\services\src\Microsoft.Azure.IIoT.Services.Processor.Onboarding\src\HostStarterService.cs:line 79
Unhandled exception. Microsoft.Azure.EventHubs.Processor.EventProcessorConfigurationException: Encountered error while fetching the list of EventHub PartitionIds
 ---> Microsoft.Azure.EventHubs.MessagingEntityNotFoundException: The messaging entity 'sb://ihsuprodamres112dednamespace.servicebus.windows.net/iot-hub-materialfluss' could not be found. To know more visit https://aka.ms/sbResourceMgrExceptions.
   at Microsoft.Azure.EventHubs.Amqp.Management.AmqpServiceClient.GetRuntimeInformationAsync()
   at Microsoft.Azure.EventHubs.EventHubClient.GetRuntimeInformationAsync()
   at Microsoft.Azure.EventHubs.Processor.PartitionManager.GetPartitionIdsAsync()
   --- End of inner exception stack trace ---
[... cut ...]
   at Microsoft.Extensions.Hosting.HostingAbstractionsHostExtensions.RunAsync(IHost host, CancellationToken token)
   at Microsoft.Extensions.Hosting.HostingAbstractionsHostExtensions.RunAsync(IHost host, CancellationToken token)
   at Microsoft.Extensions.Hosting.HostingAbstractionsHostExtensions.Run(IHost host)
   at Microsoft.Azure.IIoT.Services.Processor.Onboarding.Program.Main(String[] args) in D:\a\1\s\services\src\Microsoft.Azure.IIoT.Services.Processor.Onboarding\src\Program.cs:line 47
```

Can someone help me to get the onboarding module running, please?

karok2m, 2022-03-07:
@ma3goul, please check this [comment](https://github.com/Azure/Industrial-IoT/issues/1483#issuecomment-1015477258) for step-by-step instructions on how to register a server manually.

closed 2023-01-16 by marcschier