Azure/Industrial-IoT #862: Unable to select version on deployment script

lucarv, 2020-11-09:
**Describe the bug**
I can’t use the version switch with the deploy script anymore. See below
$ ./deploy.sh -version "2.7.183"
Using preview version...
deploy.ps1: Cannot bind parameter because parameter 'version' is specified more than once. To provide multiple values to parameters that can accept multiple values, use the array syntax. For example, "-parameter value1,value2,value3".
**To Reproduce**

Steps to reproduce the behavior:
1. Clone repo
2. Run command with version switch

**Expected behavior**
Correct version of modules are deployed

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Ubuntu 18.04

**Additional context**
We were able to run the script to create the IIoT Framework by doing ./deploy.sh “-version 2.7.183”
This however completely ignored the version switch and deployed 2.1.802 which is useless as the modules never start on IoT edge (as previously acknowledged)

koepalex, 2020-11-10:
Please try to switch branch to "release/2.7.180" and start deployment "2.7.183" script again

karok2m, 2020-11-10:
@lucarv, thanks for reporting this bug. Please use the following command as a workaround in the meantime:

```bash
$ ./deploy.sh -type all -version "2.7.183"
```

`all` is the default value for `-type`, so this will produce the same deployment as in your case, without the `-type` parameter.

karok2m, 2020-11-12:
The fix is merged into `master`.

closed 2020-11-12 by karok2m