Azure/Industrial-IoT #910: Integration of a third party OPC UA Publisher

matthi97, 2020-11-16:
**Describe the bug**
I wrote my own OPC UA PubSub Implementation according to the current specification supporting PubSub over MQTT with JSON encoding. I wanted to connect my subscriber to the azure ecosystem and visualize my data via TSI.
Steps I took:
1.	Execute the full-deployment script (.\deploy) of Azure IIoT 2.7.199
2.	Connected my OPC UA Publisher to the IoTHub following this instruction: https://docs.microsoft.com/de-de/azure/iot-hub/iot-hub-mqtt-support#using-the-mqtt-protocol-directly-as-a-device (with success)
3.	Published example data from my OPC UA Publisher 
4.	Launch TSI explorer
In step 4, the data is “recognized” but with a null identifier and with no content. (see attached image)

**Expected behavior**
Published data should be recognized & visualized by TSI.

**Screenshots**
Attached you can find messages recorded at different points in the architecture.
[Issue.zip](https://github.com/Azure/Industrial-IoT/files/5547256/Issue.zip)

The issue was first recognized with version 2.7.180 and was verified to be still present in 2.7.199.

barnstee, 2020-11-17:
TSI doesn't currently support decoding OPC UA JSON PubSub messages. Why did you write your own OPC Publisher, rather than using ours?

cristipogacean, 2020-11-17:
This issue seem to be similar with #626.
A workaround would be to recreate the TSI environment using the correct properties(they are case sensitive).

barnstee, 2020-11-18:
Thanks Cristi. @matthi97 , why did you build your own Publisher?

matthi97, 2020-11-18:
Thanks for the fast reply!
@barnstee , I built my own Publisher in another context and for another purpose. I was searching for a counter part on the subscriber side to test my compliance to the OPC UA PubSub standard.
I thought the decoding of the incoming PubSub messages is already done before the last EventHub inside a docker container or am I wrong with this assumption?
For my clarification: Is the format of my messages correct and therefore the only problem is related to #626?
Many thanks in advance!

barnstee, 2020-11-18:
Correct, the OPC UA PubSub JSON message is in fact decoded by the event processor, I wasn't sure if you were using that. If your message format is 100% OPC UA PubSub spec compliant I can't tell you of course, but the fact that the event processor can decode it and you end up with messages in TSI is a good sign. If you follow the recommendation in #626 and you get the full fidelity messages, you're at least compatible to our interpretation of the spec! :-)

closed 2021-02-26 by marcschier