Azure/Industrial-IoT #912: Opcpublisher stops sending, after running for few days. Requires restart to resume telemetry

anipuramattom, 2020-11-18:
**Describe the bug**
Opcpublisher stops sending data to cloud, after running for few days. In the log, I can see 'msg send failures' keeps increasing on every send interval. After I restart opcpublisher, data starts to flow again.

**To Reproduce**
This issue we have seen in multiple VMs. When configured high amount of tags (above 10K), and run continuously for more than 2 weeks, message send failure occurs. From there on, no messages are received on the cloud. Please refer below log. 
The telemetry resumes after opcpublisher restart.

**Screenshots**
==========================================================================
[07:02:00 INF] OpcPublisher status @ 11/16/2020 07:02:00 (started @ 10/28/2020 09:50:32)
[07:02:00 INF] ---------------------------------
[07:02:00 INF] OPC sessions (configured/connected): 1/1
[07:02:00 INF] OPC subscriptions (configured/connected): 1/1
[07:02:00 INF] OPC monitored items (configured/monitored/to remove): 49445/49445/0
[07:02:00 INF] ---------------------------------
[07:02:00 INF] monitored items queue bounded capacity: 24000
[07:02:00 INF] monitored items queue current items: 0
[07:02:00 INF] monitored item notifications enqueued: 425815601
[07:02:00 INF] monitored item notifications enqueue failure: 13881
[07:02:00 INF] ---------------------------------
[07:02:00 INF] messages sent to IoTHub: 666352
[07:02:00 INF] last successful msg sent @: 11/15/2020 02:24:35
[07:02:00 INF] bytes sent to IoTHub: 136843849560
[07:02:00 INF] avg msg size: 205362
[07:02:00 INF] msg send failures: 37316
[07:02:00 INF] messages too large to sent to IoTHub: 0
[07:02:00 INF] times we missed send interval: 1619613
[07:02:00 INF] number of events: 425801720
[07:02:00 INF] ---------------------------------
[07:02:00 INF] current working set in MB: 346
[07:02:00 INF] --si setting: 10
[07:02:00 INF] --ms setting: 225000
[07:02:00 INF] --ih setting: Amqp_Tcp_Only
[07:02:00 INF] ==========================================================================

**Desktop (please complete the following information):**
 - OS: Ubuntu 
 - Version: 18.04LTS
 - Opcpublisher : v2.5.3

barnstee, 2020-11-19:
This happens when the call to the IoT Hub client SDK SendEventAsync() fails. If this happens after 2 weeks, it usually indicates an issue with the Internet connection to the cloud or perhaps it is possible that you ran out of IoT Hub quota?

barnstee, 2020-11-24:
Closing due to no activity. Please reopen if this is still an issue.

anipuramattom, 2020-11-27:
Barnstee, this happened again on other VM recently.. I don't see any Hub quota issue. We were receiving data from other servers. 
Regarding internet connection issue, we had confirmed that there was internet connection at the time of above log generated. i.e @ 11/16/2020 07:02:00 where as last successful msg was sent @ 11/15/2020 02:24:35.
The message again started flowing soon after we restarted opcpublisher. 
I have even checked the system memory log, which also was in good state.

anipuramattom, 2020-11-27:
the point to be noted that **The message again started flowing soon after we restarted opcpublisher.**

barnstee, 2020-11-27:
ok, I will try to repro locally.

anipuramattom, 2020-12-07:
Is there any update on this issue. 
We are running our production service in 5 VMs and often getting issues from one or other VMs.

barnstee, 2020-12-08:
Still working on a repro. Next time it happens, please collect the edge logs via the "iotedge support-bundle" command and send them to me, please.

goetae, 2020-12-10:
Hello
I'm testing various Azure features as well and I have experienced the same behavior - the module was running fine and at some point stopped transporting data without any error. 

I used the Marketplace Version: 2.5.4
I configured about 10 variables that get fetched every second and sent to iot-hub every 10 seconds. 

The last diagnostic entry, before I restarted the service. 

```
[23:59:00 INF] OpcPublisher status @ 12/08/2020 23:59:00 (started @ 11/15/2020 21:05:41)
[23:59:00 INF] ---------------------------------
[23:59:00 INF] OPC sessions (configured/connected): 1/1
[23:59:00 INF] OPC subscriptions (configured/connected): 1/1
[23:59:00 INF] OPC monitored items (configured/monitored/to remove): 10/10/0
[23:59:00 INF] ---------------------------------
[23:59:00 INF] monitored items queue bounded capacity: 8192
[23:59:00 INF] monitored items queue current items: 0
[23:59:00 INF] monitored item notifications enqueued: 3540562
[23:59:00 INF] monitored item notifications enqueue failure: 0
[23:59:00 INF] ---------------------------------
[23:59:00 INF] messages sent to IoTHub: 163171
[23:59:00 INF] last successful msg sent @: 12/04/2020 06:40:41
[23:59:00 INF] bytes sent to IoTHub: 674396358
[23:59:00 INF] avg msg size: 4133
[23:59:00 INF] msg send failures: 40803
[23:59:00 INF] messages too large to sent to IoTHub: 0
[23:59:00 INF] times we missed send interval: 167073
[23:59:00 INF] number of events: 3540562
[23:59:00 INF] ---------------------------------
[23:59:00 INF] current working set in MB: 103
[23:59:00 INF] --si setting: 10
[23:59:00 INF] --ms setting: 262144
[23:59:00 INF] --ih setting: Amqp_Tcp_Only
```

Before service restart I tried to find the root cause without success. As I do not have setup an end-to-end solution, I did not noticed the issue in time. The logfiles are overwritten after 2 days and therefore no further details.
I did some code review an

barnstee, 2020-12-17:
Right on the money, @goetae , I also noticed the lack of exception log output when an IoT Hub or Edge Hub exception occurs. I've added this to a new 2.5.4 release branch, see check-in here: https://github.com/Azure/Industrial-IoT/commit/1592af31d407ec928562a7aaedb22b1a05ccc00e
The latest version of OPC Publisher is always marked by the "latest" tag, right now this points to 2.7.199. However, the latest version doesn't have the IoT hub methods interface, that some customers need. We will fix this in version 3.0. @anipuramattom is on 2.5.3 (probably for this reason) and hence I fixed the log output and am in the process of releasing version 2.5.4.

barnstee, 2020-12-17:
OK, the release is done, please try out 2.5.4: mcr.microsoft.com/iotedge/opc-publisher:2.5.4

anipuramattom, 2020-12-29:
@barnstee , thank you. We have updated to v2.5.4. Will update you on this. It's working fine for the last one week.

barnstee, 2021-01-15:
Closing due to inactivity. If it happens again, just reopen it.

anipuramattom, 2021-02-11:
> OK, the release is done, please try out 2.5.4: mcr.microsoft.com/iotedge/opc-publisher:2.5.4

@barnstee , **with reference to the previous issue, after deploying v2.5.4** , after 5-6 weeks, now we are getting below error in opcpublisher

[09:25:46 INF] messages sent to IoTHub: 252650
[09:25:46 INF] last successful msg sent @: 02/10/2021 19:33:04
[09:25:46 INF] bytes sent to IoTHub: 32602899123
[09:25:46 INF] avg msg size: 129043
[09:25:46 INF] msg send failures: 8374
[09:25:46 INF] messages too large to sent to IoTHub: 0
[09:25:46 INF] times we missed send interval: 638898
[09:25:46 INF] number of events: 99756044
[09:25:46 INF] ---------------------------------
[09:25:46 INF] current working set in MB: 184
[09:25:46 INF] --si setting: 10
[09:25:46 INF] --ms setting: 225000
[09:25:46 INF] --ih setting: Amqp_Tcp_Only
[09:25:46 INF] ==========================================================================
[09:25:54 ERR] Error while sending message. Dropping...
System.ObjectDisposedException: The semaphore has been disposed.
   at System.Threading.SemaphoreSlim.CheckDispose()
   at System.Threading.SemaphoreSlim.WaitAsync(Int32 millisecondsTimeout, CancellationToken cancellationToken)
   at System.Threading.SemaphoreSlim.WaitAsync(TimeSpan timeout, CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.Transport.Amqp.AmqpConnectionHolder.EnsureConnectionAsync(TimeSpan timeout)
   at Microsoft.Azure.Devices.Client.Transport.Amqp.AmqpConnectionHolder.OpenSessionAsync(DeviceIdentity deviceIdentity, TimeSpan timeout)
   at Microsoft.Azure.Devices.Client.Transport.AmqpIoT.AmqpUnit.EnsureSessionAsync(TimeSpan timeout)
   at Microsoft.Azure.Devices.Client.Transport.AmqpIoT.AmqpUnit.OpenAsync(TimeSpan timeout)
   at Microsoft.Azure.Devices.Client.Transport.Amqp.AmqpTransportHandler.OpenAsync(CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.Transport.ProtocolRoutingDelegatingHandler.OpenAsync(CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.Transport.ErrorDelegatingHandler.<>c__DisplayClass23_0.<<ExecuteWithErrorHandlingAsync>b__0>d.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at Microsoft.Azure.Devices.Client.Transport.ErrorDelegatingHandler.ExecuteWithErrorHandlingAsync[T](Func`1 asyncOperation)
   at Microsoft.Azure.Devices.Client.Transport.RetryDelegatingHandler.<>c__DisplayClass33_0.<<OpenAsyncInternal>b__0>d.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at Microsoft.Azure.Devices.Client.Transport.RetryDelegatingHandler.EnsureOpenedAsync(CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.Transport.RetryDelegatingHandler.<>c__DisplayClass14_0.<<SendEventAsync>b__0>d.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at Microsoft.Azure.Devices.Client.Transport.RetryDelegatingHandler.SendEventAsync(Message message, CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.InternalClient.SendEventAsync(Message message)
   at OpcPublisher.HubCommunicationBase.MonitoredItemsProcessorAsync(CancellationToken ct) in D:\a\1\s\modules\src\Microsoft.Azure.IIoT.Modules.OpcUa.PublisheommunicationBase.cs:line 1762
[09:25:56 INF] ==========================================================================

This now happened continuously for 2 VMs within 3 days gap.
Currently we are running opcpublisher in 6 VMs, all have below versions:
iotedge- v1.0.9.3
opc-publisher- v2.5.4

I can share you separately  support_bundle logs in erichb@microsoft.com, if required.

barnstee, 2021-02-11:
Reopening and assigning to @varunpuranik to have a look.

barnstee, 2021-02-11:
Please send me the support bundle via email.

anipuramattom, 2021-02-23:
@varunpuranik , @barnstee ,
Is there any update on the issue. We faced similar issue again on another VM

koepalex, 2021-03-31:
@anipuramattom there is a also a version of OPC Publisher as part of Industrial IoT Platform, that can be used as standalone. Which is the currently maintained version `mcr.microsoft.com/iotedge/opc-publisher:2.7.206` in our long-running tests it is running for more than 50 days.

onderyildirim, 2021-05-26:
are there any updates on this? 
is there a release on 2.5 branch where this bug is resolved? 
is it resolved in 2.7 branch?

whakey, 2021-06-10:
I ran into the exact same issue few times. I'm currently using 2.5.4, because our configuration pipeline relies on direct methods. I was waiting for the direct methods to be merged back into the newest versions.

If someone can confirm this is fixed in 2.7 branch, might need to consider moving to that if direct methods are not coming soon

anipuramattom, 2021-07-09:
@koepalex , I tried with new version v2.7.206, but ran into new issues...

I am connecting to Ignition OPC server with large number of taglist. When I subscribe to 50K tags, everything works perfectly fine. However, when I am subscribing to 125K tags, the behaviour changes:
 1. It takes around 15 - 20 min to complete the subscription
 2. After reading the initial screen shot values of tags, no subscription happens
 3. Ignition server network get disconnect and CPU usage become 100%
 4. Finally my Ignition and TagBrowser crashes
 
With the same amount of 125K tags, it works fine with old OPC version 2.5.4

**System Configuration:**

- OpcPublisher Version – 2.7.206
- Edge Configuration: 4core, 16GB RAM
- Ignition Startup Memory  allocated– 2048 MB
- Ignition Max Memory allocated – 6144 MB
- No. of Tags configured: 1,25K
- No. of Tags simulated / sec = 10K

**Log:**

[09:57:24 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Connecting all clients to opctestclient1 using MqttOverTcp. 
[09:57:24 INF Root] Starting module OpcPublisher version 2.7.206.28401. 
[09:57:24 INF Root] Initiating prometheus at port 9702/metrics 
[09:57:24 INF Root] Prometheus metric server started. 
[09:57:24 WRN Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Bypassing certificate validation for client. 
[09:57:24 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Running in iotedge context. 
[09:57:25 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] 0: Module apmttestOPC_opcpublisher reconnected due to Connection_Ok. 
[09:57:25 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Initialize device twin for apmttestOPC - opcpublisher 
[09:57:25 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Applying initial desired state. 
[09:57:25 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Reporting currently initial state. 
[09:57:25 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Module Host started. 
[09:57:25 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.WorkerSupervisor] Creating new worker... 
[09:57:25 INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.LegacyJobOrchestrator] File /appdata/pn.json has changed, reloading... 
[09:57:27 INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Models.PublishedNodesJobConverter] Read 1 items from published nodes file in 00:00:01.2655035 
[10:08:49 INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Models.PublishedNodesJobConverter] Converted items to jobs in 00:11:22.2205269 
[10:08:52 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.WorkerSupervisor] Starting worker 'StandalonePublisher_0'... 
[10:08:52 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker] Worker StandalonePublisher_0: {} 
[10:08:52 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker] Worker: StandalonePublisher_0 processing job: Standalone_apmttestOPC_opcpublisher, mode: Active 
[10:08:54 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Subscription 'opc.tcp://10.1.0.4:4096/discovery_f3e927fc11d4907f79465b2378dbfadad593dd8d' registered/updated in session 'ab9c3ef401e7cdb4570c0edd9e9cbb49067f5e06' in state Init 
......
......
......
[10:08:54 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Subscription 'opc.tcp://10.1.0.4:4096/discovery_5677312f7467609f9b0a388ae3dadf73fbf9036d' registered/updated in session 'ab9c3ef401e7cdb4570c0edd9e9cbb49067f5e06' in state Init 
[10:08:54 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Creating session 'ab9c3ef401e7cdb4570c0edd9e9cbb49067f5e06' for endpoint 'opc.tcp://10.1.0.4:4096/discovery'... 
[10:08:54 WRN Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Accepting untrusted peer certificate D16932066CD4896B74900405C27614191C7040DD, 'C=US, S=CA, L=Folsom, OU="", O=Inductive Automation, CN=Ignition OPC UA Server' due to AutoAccept(UntrustedCertificates) set! 
[10:08:54 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Session 'ab9c3ef401e7cdb4570c0edd9e9cbb49067f5e06' created, loading complex type system ...  
[10:08:55 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Session 'ab9c3ef401e7cdb4570c0edd9e9cbb49067f5e06' complex type system loaded 
[10:08:55 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Azure IIoT ab9c3ef401e7cdb4570c0edd9e9cbb49067f5e06 took 00:00:00.4860881. 
[10:08:55 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Connected to 'opc.tcp://10.1.0.4:4096/discovery' 
[10:08:55 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] Added 1000 monitored items to subscription opc.tcp://10.1.0.4:4096/discovery_e0e81dd377bfad99c31e58d44259175ed96d6fe1 
[10:08:55 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] Now monitoring 1000 nodes in subscription opc.tcp://10.1.0.4:4096/discovery_e0e81dd377bfad99c31e58d44259175ed96d6fe1 
.............
............
.............
[10:09:20 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] DataChange for subscription: opc.tcp://10.1.0.4:4096/discovery_8772036e66a0021d27610fe8ed7096b40581df86 having no monitored items yet 
[10:09:20 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] Now monitoring 1000 nodes in subscription opc.tcp://10.1.0.4:4096/discovery_8772036e66a0021d27610fe8ed7096b40581df86 
[... cut ...]
[10:12:04 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] DataChange for subscription: opc.tcp://10.1.0.4:4096/discovery_43c06186d7620c604763444b8fd699737fde735c having no monitored items yet 
[10:12:13 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] DataChange for subscription: opc.tcp://10.1.0.4:4096/discovery_0c7642609967e8309f35a9c2f7c795df782aec2f having no monitored items yet 
[10:12:14 INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.DataFlowProcessingEngine] 
..................
..................

koepalex, 2021-08-14:
I will take over this issue next week and try to sort out all issues, but there will be no bug fixing for `2.5.X` 

@anipuramattom, please check if you used the `--fd=true` flag, this could explain large startup delays.

koepalex, 2021-08-30:
Started long running test to analyze the issue with linked fixes for diagnostic log. After 5 days of running, no decrease in sending telemetry is visible.

anipuramattom, 2021-08-30:
@koepalex , good to hear that.  Thank you for the update.
Waiting for the release....

anipuramattom, 2021-09-08:
@koepalex , is there any update on the latest release with fixes. As mentioned earlier, we are in production deployment and is delayed due to the mentioned issues.

koepalex, 2021-09-09:
After 15 days of running, no decrease in sending telemetry is visible

koepalex, 2021-09-17:
After 22 days of running, no decrease in sending telemetry is visible

marcschier, 2022-04-01:
Our long running tests (14 days) do not experience this issue.  If there is an update on this issue when using 2.8.2 or later release, please re-open.

closed 2022-04-01 by marcschier