Azure/Industrial-IoT #920: Deployment issue - deploy.ps formatting

arfontai, 2020-11-20:
I've cloned the repo and get this error when I try to execute the deploy script.
It seems their is a formatting issue in the deploy.ps file

PS C:\dev\Industrial-IoT> .\deploy
Az installed.
AzureAD installed.
At C:\dev\Industrial-IoT\deploy\scripts\deploy.ps1:1162 char:27
+ Write-Host "Signing in ..."
+                           ~
The string is missing the terminator: ".
At C:\dev\Industrial-IoT\deploy\scripts\deploy.ps1:1175 char:1
+
Missing closing ')' in expression.
At C:\dev\Industrial-IoT\deploy\scripts\deploy.ps1:1001 char:21
+                 try {
+                     ~
Missing closing '}' in statement block or type definition.
At C:\dev\Industrial-IoT\deploy\scripts\deploy.ps1:1175 char:1
+
The Try statement is missing its Catch or Finally block.
At C:\dev\Industrial-IoT\deploy\scripts\deploy.ps1:996 char:35
+             if ($aadAddReplyUrls) {
+                                   ~
Missing closing '}' in statement block or type definition.
At C:\dev\Industrial-IoT\deploy\scripts\deploy.ps1:913 char:13
+         try {
+             ~
Missing closing '}' in statement block or type definition.
At C:\dev\Industrial-IoT\deploy\scripts\deploy.ps1:912 char:19
+     while ($true) {
+                   ~
Missing closing '}' in statement block or type definition.
At C:\dev\Industrial-IoT\deploy\scripts\deploy.ps1:656 char:27
+ Function New-Deployment() {
+                           ~
Missing closing '}' in statement block or type definition.
    + CategoryInfo          : ParserError: (:) [], ParseException
    + FullyQualifiedErrorId : TerminatorExpectedAtEndOfString

nainesh72, 2020-11-24:
Hi
I have Same Problem 
PS C:\Industrial-IoT> .\deploy
Az installed.
AzureAD installed.
At C:\Industrial-IoT\deploy\scripts\deploy.ps1:1162 char:27
+ Write-Host "Signing in ..."
+                           ~
The string is missing the terminator: ".
At C:\Industrial-IoT\deploy\scripts\deploy.ps1:1175 char:1
+
Missing closing ')' in expression.
At C:\Industrial-IoT\deploy\scripts\deploy.ps1:1001 char:21
+                 try {
+                     ~
Missing closing '}' in statement block or type definition.
At C:\Industrial-IoT\deploy\scripts\deploy.ps1:1175 char:1
+
The Try statement is missing its Catch or Finally block.
At C:\Industrial-IoT\deploy\scripts\deploy.ps1:996 char:35
+             if ($aadAddReplyUrls) {
+                                   ~
Missing closing '}' in statement block or type definition.
At C:\Industrial-IoT\deploy\scripts\deploy.ps1:913 char:13
+         try {

mitkox, 2020-11-26:
Same issue here with 2.7.199. Have try to checkout the release and deploy from the 2.7.199 directory again without success. 

1.	clone the repo
2.	checkout release/2.7.199 branch if you want to deploy 2.7.199 version
3.	run the following command to deploy:
.\deploy -type all -version 2.7.199
 
Doesn't work.

marcschier, 2021-07-14:
Please try deploying from main with --version 2.8.0

closed 2021-07-14 by marcschier