Azure/Industrial-IoT #933: opc-publisher:2.8.2 not found: manifest unknown: manifest tagged by "2.8.2" is not found

gregeva, 2020-11-25:
**Describe the bug**
I am trying to deploy another IoT Edge in order to test Azure Industrial IoT with realistic emulated data from a series of production lines; however when I connect up a new IoT Edge I am getting this error from edgeAgent saying that it cannot find the opc-publisher module to deploy.

I have not touched my installation otherwise, and the deployment manifest is OOTB from the documented installation procedure.

**To Reproduce**
Steps to reproduce the behavior:
1. Create an Ubuntu 18.04 LTS VM
2. Connect to it via SSH
3. Install Azure IoT Edge on it
4. Add IoT Edge device in IoT Hub and add the iiot device tag as well as Linux OS
5. Add the Primary Connection String to /etc/iotedge/config.yaml
6. Restart IoT Edge and wait patiently
7. Grab logs from edgeAgent and get the error that the opc-publisher module stated in the deployment manifest cannot be found

**edgeAgent container logs**
`d.cs:line 127
   at Microsoft.Azure.Devices.Edge.Agent.Edgelet.Version_2019_11_05.ModuleManagementHttpClient.CreateModuleAsync(ModuleSpec moduleSpec) in /home/vsts/work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Edgelet/version_2019_11_05/ModuleManagementHttpClient.cs:line 99
   at Microsoft.Azure.Devices.Edge.Agent.Core.LoggingCommandFactory.LoggingCommand.ExecuteAsync(CancellationToken token) in /home/vsts/work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Core/LoggingCommandFactory.cs:line 60
   at Microsoft.Azure.Devices.Edge.Agent.Core.Commands.GroupCommand.ExecuteAsync(CancellationToken token) in /home/vsts/work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Core/commands/GroupCommand.cs:line 35
   at Microsoft.Azure.Devices.Edge.Agent.Core.LoggingCommandFactory.LoggingCommand.ExecuteAsync(CancellationToken token) in /home/vsts/work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Core/LoggingCommandFactory.cs:line 60
   at Microsoft.Azure.Devices.Edge.Agent.Core.PlanRunners.OrderedRetryPlanRunner.ExecuteAsync(Int64 deploymentId, Plan plan, CancellationToken token) in /home/vsts/work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Core/planrunners/OrdererdRetryPlanRunner.cs:line 87<---
<4> 2020-11-25 18:37:35.468 +00:00 [WRN] - Reconcile failed because of the an exception
System.AggregateException: One or more errors occurred. (Error calling Create module publisher: Could not create module publisher
	caused by: Could not pull image mcr.microsoft.com/iotedge/opc-publisher:2.8.2
	caused by: manifest for mcr.microsoft.com/iotedge/opc-publisher:2.8.2 not found: manifest unknown: manifest tagged by "2.8.2" is not found) ---> Microsoft.Azure.Devices.Edge.Agent.Edgelet.EdgeletCommunicationException: Error calling Create module publisher: Could not create module publisher
	caused by: Could not pull image mcr.microsoft.com/iotedge/opc-publisher:2.8.2
	caused by: manifest for mcr.microsoft.com/iotedge/opc-publisher:2.8.2 not found: manifest unknown: manifest tagged by "2.8.2" is not found
   at Microsoft.Azure.Devices.Edge.Agent.Edgelet.Version_2019_11_05.ModuleManagementHttpClient.HandleException(Exception exception, String operation) in /home/vsts/work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Edgelet/version_2019_11_05/ModuleManagementHttpClient.cs:line 210
   at Microsoft.Azure.Devices.Edge.Agent.Edgelet.Versioning.ModuleManagementHttpClientVersioned.Execute[T](Func`1 func, String operation) in /home/vsts/work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Edgelet/versioning/ModuleManagementHttpClientVersioned.cs:line 127
   at Microsoft.Azure.Devices.Edge.Agent.Edgelet.Version_2019_11_05.ModuleManagementHttpClient.CreateModuleAsync(ModuleSpec moduleSpec) in /home/vsts/work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Edgelet/version_2019_11_05/ModuleManagementHttpClient.cs:line 99
   at Microsoft.Azure.Devices.Edge.Agent.Core.LoggingCommandFactory.LoggingCommand.ExecuteAsync(CancellationToken token) in /home/vsts/work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Core/LoggingCommandFactory.cs:line 60
   at Microsoft.Azure.Devices.Edge.Agent.Core.Commands.GroupCommand.ExecuteAsync(CancellationToken token) in /home/vsts/work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Core/commands/GroupCommand.cs:line 35
   at Microsoft.Azure.Devices.Edge.Agent.Core.LoggingCommandFactory.LoggingCommand.ExecuteAsync(CancellationToken token) in /home/vsts/work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Core/LoggingCommandFactory.cs:line 60
   at Microsoft.Azure.Devices.Edge.Agent.Core.PlanRunners.OrderedRetryPlanRunner.ExecuteAsync(Int64 deploymentId, Plan plan, CancellationToken token) in /home/vsts/work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Core/planrunners/OrdererdRetryPlanRunner.cs:line 87
   --- End of inner exception stack trace ---
   at Microsoft.Azure.Devices.Edge.Agent.Core.PlanRunners.OrderedRetryPlanRunner.<>c.<ExecuteAsync>b__7_0(List`1 f) in /home/vsts/work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Core/planrunners/OrdererdRetryPlanRunner.cs:line 115
   at Microsoft.Azure.Devices.Edge.Agent.Core.PlanRunners.OrderedRetryPlanRunner.ExecuteAsync(Int64 deploymentId, Plan plan, CancellationToken token) in /home/vsts/work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Core/planrunners/OrdererdRetryPlanRunner.cs:line 116
   at Microsoft.Azure.Devices.Edge.Agent.Core.Agent.ReconcileAsync(CancellationToken token) in /home/vsts/work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Core/Agent.cs:line 146
   at Microsoft.Azure.Devices.Edge.Agent.Core.Agent.ReconcileAsync(CancellationToken token) in /home/vsts/work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Core/Agent.cs:line 157
---> (Inner Exception #0) Microsoft.Azure.Devices.Edge.Agent.Edgelet.EdgeletCommunicationException- Message:Error calling Create module publisher: Could not create module publisher
	caused by: Could not pull image mcr.microsoft.com/iotedge/opc-publisher:2.8.2
	caused by: manifest for mcr.microsoft.com/iotedge/opc-publisher:2.8.2 not found: manifest unknown: manifest tagged by "2.8.2" is not found, StatusCode:404, at:   at Microsoft.Azure.Devices.Edge.Agent.Edgelet.Version_2019_11_05.ModuleManagementHttpClient.HandleException(Exception exception, String operation) in /home/vsts/work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Edgelet/version_2019_11_05/ModuleManagementHttpClient.cs:line 210
   at Microsoft.Azure.Devices.Edge.Agent.Edgelet.Versioning.ModuleManagementHttpClientVersioned.Execute[T](Func`1 func, String operation) in /home/vsts/work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Edgelet/versioning/ModuleManagementHttpClientVersioned.cs:line 127
   at Microsoft.Azure.Devices.Edge.Agent.Edgelet.Version_2019_11_05.ModuleManagementHttpClient.CreateModuleAsync(ModuleSpec moduleSpec) in /home/vsts/work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Edgelet/version_2019_11_05/ModuleManagementHttpClient.cs:line 99
   at Microsoft.Azure.Devices.Edge.Agent.Core.LoggingCommandFactory.LoggingCommand.ExecuteAsync(CancellationToken token) in /home/vsts/work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Core/LoggingCommandFactory.cs:line 60
   at Microsoft.Azure.Devices.Edge.Agent.Core.Commands.GroupCommand.ExecuteAsync(CancellationToken token) in /home/vsts/work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Core/commands/GroupCommand.cs:line 35
   at Microsoft.Azure.Devices.Edge.Agent.Core.LoggingCommandFactory.LoggingCommand.ExecuteAsync(CancellationToken token) in /home/vsts/work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Core/LoggingCommandFactory.cs:line 60
   at Microsoft.Azure.Devices.Edge.Agent.Core.PlanRunners.OrderedRetryPlanRunner.ExecuteAsync(Int64 deploymentId, Plan plan, CancellationToken token) in /home/vsts/work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Core/planrunners/OrdererdRetryPlanRunner.cs:line 87<---
<6> 2020-11-25 18:37:35.681 +00:00 [INF] - Updated reported properties
<6> 2020-11-25 18:37:40.935 +00:00 [INF] - Plan execution started for deployment 2
<6> 2020-11-25 18:37:40.935 +00:00 [INF] - Plan execution ended for deployment 2
<6> 2020-11-25 18:37:41.163 +00:00 [INF] - Updated reported properties
<6> 2020-11-25 18:37:46.528 +00:00 [INF] - Plan execution started for deployment 2
<6> 2020-11-25 18:37:46.528 +00:00 [INF] - Plan execution ended for deployment 2
<6> 2020-11-25 18:37:51.815 +00:00 [INF] - Plan execution started for deployment 2
[... cut ...]
<6> 2020-11-25 18:38:34.227 +00:00 [INF] - Plan execution ended for deployment 2
<6> 2020-11-25 18:38:39.561 +00:00 [INF] - Plan execution started for deployment 2
<6> 2020-11-25 18:38:39.561 +00:00 [INF] - Plan execution ended for deployment 2
<6> 2020-11-25 18:38:44.859 +00:00 [INF] - Plan execution started for deployment 2
<6> 2020-11-25 18:38:44.860 +00:00 [INF] - Plan execution ended for deployment 2
`

**Application (please complete the following information):**
 - Azure Industrial IoT application version running on AKS: 2.7.170

barnstee, 2020-11-27:
OPC Publisher version 2.8.2 has not been officially released yet and was uploaded to the MCR by accident. We removed it again. The latest released version is 2.7.199.

gregeva, 2020-11-27:
I understand what happened then, but the next question is how to fix it?

The layered deployment coming from the application has the deployment version at 2.8.2; so I would imagine that this deployment manifest needs to be rolled back as well, no?  (attached)

The problem is that all of the IoT Edge agents are being told to use a version of the OPC Publisher that is not there.

External email from: noreply@github.com

Closed #933<https://nam04.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2FAzure%2FIndustrial-IoT%2Fissues%2F933&data=04%7C01%7Cgeva%40ptc.com%7C3a14c8025e8f4321962908d892c3644e%7Cb9921086ff774d0d828acb3381f678e2%7C0%7C0%7C637420715188969625%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=BWRKNaV1iKj%2BRHGeW0eYLH%2BVc6ERdjaqVKi%2BucRn7lQ%3D&reserved=0>.

barnstee, 2020-11-27:
Right, we fixed this in master last week, please clone and try deployment again, specifying the -version parameter set to 2.7.199.

gregeva, 2020-11-30:
I installed version 2.7.170 when I did my installation a few months ago.

Now when I check I have version 2.8.2 running everywhere - except this new IoT Edge that I tried to add last week.

I didn't see anything about an auto-upgrade, but this looks to be what happened.  Can you confirm?  Can you point me to documentation?

barnstee, 2020-11-30:
if you have "latest" specified in your container deploy options, you will always get the latest version, yes. The latest version should have been set to 2.7.199.

karok2m, 2020-12-07:
@gregeva, when deploying a specific version please make sure to be on a release branch corresponding to that version before you run the deployment. That is, if you want to deploy `2.7.199` version, you need to deploy it from `release/2.7.199` branch. Please note that you will still need to specify the version in command line:

```bash
.\deploy -version "2.7.199"
```

gregeva, 2020-12-07:
This is very interesting and important information @karok2m .  This is not what our development/product management team have understood - meaning that our installation documentation is wrong as we have understood that applying the -version tag is all that was required.

http://support.ptc.com/help/thingworx/azure_connector_scm/en/#page/thingworx_scm_azure%2Fopc_ua%2Fc_setup_azure_infrastructure.html%23  (see step 3)

Could you confirm that we need to change our DevOps process as well as documented approach?

barnstee, 2020-12-07:
Reopening as there is no info (a video) now available and a separate internal bug was opened

barnstee, 2020-12-07:
@gregeva , I watched the video, thanks for providing. The issue is that the new test deployment setup is from master branch instead of from our release branch. We use master for integration, NOT as a latest release. Please always deploy from one of our release branches. Our documentation also calls this out on the main readme "Releases of the Platform".

To fix the deployment you have (using HELM):
Prepare machine install:
https://helm.sh/docs/intro/install/

Use Azure CLI:
az acr helm install-cli --client-version 

The commands below are required to update the version:
az aks get-credentials -a -g <resource-group> -n <aks-cluster-name>
helm get values azure-industrial-iot -n azure-industrial-iot > values.yaml
edit values.yaml to set image:tag to 2.7.199
helm repo add azure-iiot https://azureiiot.blob.core.windows.net/helm
helm repo update
helm upgrade azure-industrial-iot azure-iiot/azure-industrial-iot -n azure-industrial-iot -f .\values.yaml

This will restart the services and will correct the layered deplyoment.

barnstee, 2020-12-08:
And to answer your question, @gregeva , yes, please update your docs accordingly. I am closing this issue. Please reopen if there is still questions in relation to this.

closed 2020-12-08 by barnstee

karok2m, 2020-12-08:
@gregeva, I've been going over the documentation that you have and I finally understand how version `2.8.2` got deployed into your AKS cluster. So, actually, all microservice that have been deployed to AKS had version `2.8.2` in your case. And you can see this in your AKS deployment as well, you checked (in the video) tags of `aiiot-event-processor` deployment and you saw `app.kubernetes.io/version: 2.8.2`.

So the root cause of this issue is in this documentation: [Step 3. Deploying the Azure Industrial IoT (IIoT) Microservices on an Azure Kubernetes Cluster](http://support.ptc.com/help/thingworx/azure_connector_scm/en/index.html#page/thingworx_scm_azure%2Fopc_ua%2Fc_deploy_microservices.html%23)

There are a few problems:

#### 1. Azure IIoT platform version is not specified.

You are not specifying the version of microservices that should be deployed to your Kubernetes cluster.

 **Deploying the Azure Industrial IoT (IIoT) Microservices** section in the doc above walks you through creation of `aiiot.yaml` file where it points you to [Using configuration in Azure Key Vault](https://github.com/Azure/Industrial-IoT/blob/master/docs/deploy/howto-add-aks-to-ps1.md#using-configuration-in-azure-key-vault) and asks you to copy the content from the sample. If you look closely to the sample, you will see that no `image:tag` is specified there. This means that you will use whatever version is specified in the `values.yaml` file of the chart in our GitHub repository. And in our `master` branch that points to `2.8.2`. So this is how you ended up with `2.8.2` version of our platform. To fix this, your documentation should mention that version of the platform should be **explicitly** specified in `aiiot.yaml` file. So the sample that you were using should have the following two lines in it:

```yaml
image:
  tag: 2.7.182
```

Please use whatever version is desired instead of `2.7.182` here. But please only use one of our official release versions.

#### 2. Helm chart is consumed from GitHub repository

You are deploying microservices using helm chart from within the GitHub repository instead of the official Helm repositories that we publish to.

The deployment command that is used in your documentation is the following:

```bash
helm install aiiot --namespace aiiot .\deploy\helm\azure-industrial-iot\ -f aiiot.yaml
```

Here you can see that `helm` is going to consume the chart from `.\deploy\helm\azure-industrial-iot\` directory. This is fine for development purposes, but I would not recommended this method for production deployments. We are actually publishing our `azure-industrial-iot` chart to several Helm chart repositories and you should use one of them. [Here](https://github.com/Azure/Industrial-IoT/blob/master/docs/deploy/howto-deploy-helm.md#helm-repositories) you can find the details of both Helm chart repositories that we publish to. I would suggest to change your documentation to use one of them, as shown here:

```bash
helm repo add microsoft https://microsoft.github.io/charts/repo
helm repo update
helm install aiiot microsoft/azure-industrial-iot --namespace aiiot --version 0.3.1 -f aiiot.yaml
```

Please note that I am explicitly specifying `0.3.1` version of the chart here, as `0.3.1` version should be used for deployments of our `2.7.x` releases (`2.7.170`, `2.7.182`, `2.7.183` or `2.7.199`).

To be honest, if you explicitly specify `image:tag` in `aiiot.yaml`, then deploying from GitHub repo will also work as long as you are on one of our `release/2.7.x` release branches (or tags). The codes of the chart in `master` branch are intended for use with our upcoming release, so it should not be used for deployment of `2.7.x` version.

@gregeva, please let me know if you need any additional information regarding those two points.

gregeva, 2020-12-08:
This is really great clarification!  Thanks a lot @barnstee and @karok2m .  I will implement your suggestions tomorrow to get back to where I should be, and then will work with our product management and documentation teams to get our documentation and training material updated.

barnstee, 2020-12-08:
Awesome, good luck! 👍

gregeva, 2020-12-09:
I've added the image tag with the version to the aiiot.yaml and did an upgrade of the application and confirm that it took the change and put the appropriate versions in place as well as got the correct deployment manifests.

`helm upgrade aiiot --namespace aiiot .\deploy\helm\azure-industrial-iot\ -f aiiot.yaml`

Considering that I want to get this working to your production standard, I then uninstalled the application, added the Microsoft repo, and installed using the Helm chart with specified version from the repo.

`
helm uninstall aiiot --namespace aiiot
kubectl get all --namespace aiiot
helm install aiiot --namespace aiiot .\deploy\helm\azure-industrial-iot\ -f aiiot.yaml
helm repo add microsoft https://microsoft.github.io/charts/repo
helm repo update
helm install aiiot microsoft/azure-industrial-iot --namespace aiiot --version 0.3.1 -f aiiot.yaml
`

Despite having the image tag with application version specified in aiiot.yaml, the 0.3.1 version Helm chart is loading 2.7.170 - however the OPC Publisher and other modules still report running 2.7.182 as specified in the image version tag.

Other than Batched Telemetry no longer working, everything seems to be working.  I'm going to try to manually change the Batched Telemetry settings as I wanted to customise them anyway.

barnstee, 2020-12-10:
OK, good, LMK how it goes. Batching config changed between 170 and 183, please see release notes.

gregeva, 2020-12-10:
+@chmagitt

karok2m, 2020-12-10:
@gregeva, a small clarification: value for `app.kubernetes.io/version: 2.8.2` tag (and `helm.sh/chart: azure-industrial-iot-0.4.0`) actually comes from metadata of the chart and not the value of `image:tag`. But since you have not specified an `image:tag`, they would have had the same value, which is `2.8.2`. The best way to get image version would actually be by using `kubectl describe` for one of the deployments. The command would look something like this:

```bash
kubectl describe deployment aiiot-events-processor -n aiiot
```

It will output the whole definition of the deployment, which will also contain full definition of the image, including the tag/version. Please use this and see if you still see `2.7.170` there or `2.7.183`.

Regarding your helm commands, I was suggesting that you use those commands:

```bash
helm repo add microsoft https://microsoft.github.io/charts/repo
helm repo update
helm install aiiot microsoft/azure-industrial-iot --namespace aiiot --version 0.3.1 -f aiiot.yaml
```

Instead of this one:

```bash
helm install aiiot --namespace aiiot .\deploy\helm\azure-industrial-iot\ -f aiiot.yaml
```

So you don't need to run `helm install` twice. The second one should actually conflict with the first one and complain that `aiiot` release already exists.

gregeva, 2020-12-11:
Hi @karok2m .  I did actually run the commands as you've stated here; I just made a mistake when pasting them into the comments here.  I actually had to reinstall using the Git repo Helm chart deployment as the application was no longer showing up in Helm for upgrade/uninstall.... this way I was able to uninstall it.. and then reinstall it using these 3 helm commands that you mentioned.  I'm going back now to correct what I did so that my friends who will be repairing other deployments can follow along exactly.