Azure/Industrial-IoT #967: Info Required - How to setup custom names for the IIOT components per organization standard instead of automatically generated names

achu152, 2020-12-09:


karok2m, 2020-12-11:
@achu152, are you taking about components of IIoT platform that are deployed using `azure-industrial-iot` Helm chart ? Or are you using some other deployment method ?

achu152, 2020-12-15:
@karok2m  :  Trying to deploy IIOT components via windows command line [.\deploy -version <version>] . The name of the components (ex: IOT Hub ,Keyvault etc)  created currently is as per a certain logic . Where exactly can i make changes to the script so as to create custom names? 
Tried giving a custom name as below in azuredeploy.minimum.json file . However , it does not work and still taking the name of the components as per the logic . Do we need to change it anywhere else?
 "iotHubName": {
            "type": "string",
            "defaultValue": "USETIOTHUB",
            "metadata": {
                "description": "The name of Azure IoT Hub created as part of the deployment."
            }

closed 2021-02-11 by achu152